Sevenstax/FreeV2G #206: Question:GETTING FreeV2G issue

JIN-MIRAE, 2023-07-27:
**Your Question**
Ask your questions here.

Hello,
I'm trying simulation test by using FreeV2G readme , but I have some problem.

First is that "source .venv/bin/activate" command dosen't practiced.
(error : 'soruce' isn't internal or external command, program for practice, or bat.file)

Second is that when I input command, "pip install python-libpcap", error happened as below image.
 
[image]
[image]

Thank you
Mirae

nststx, 2023-07-27:
Hi @JIN-MIRAE,

Thanks for submitting your issue. The `source` command is a command specific to UNIX systems. On windows you just execute the *.bat* activation script to activate the virtual environment. If you run `python -m venv .venv` the activation script should be located under *.venv/Scripts/activate.bat*.
On Windows we also do not use the **python-libpcap** package. Instead please try to install the **scapy** package.

If you have any more questions, feel free to ask!

Best regards,
Niklas Stoffers

JIN-MIRAE, 2023-07-28:
Hi, I tried again according to your comment, and it's successful.

And then I confirmed output, "Wait untill an EV connects", as below image.
[image]

But, I have another problem.
After seeing the output, I have to connect the PLC interface of the EV to the EVSE physically?
I don't know that how to connect EV & EVSE physically.
So, could you provide any guide?

Thank you
Mirae

adirschl, 2023-07-28:
Hi Mirae,

I think you have two WHITEbeet Eval Boards. WHITEbeet-EI for EVSE side and WHITEbeet-PI for EV side. You physically connect the CP and PE pins of the boards, which means CP of the EI board to CP of the PI board and PE of the EI board to PE of the PI board. According to ISO15118-3 the PLC communication between EVSE and EV takes place over ControlPilot wire (CP) and PE wire. 

BR,
Andreas
CODICO

JIN-MIRAE, 2023-07-28:
Yes. I did connected CP of PI & EI and PE of PI & EI.
[image]

But, after connecting CP and PE, any additional output was not take placed.
Please let me know that what should I do.

Thank you

adirschl, 2023-07-28:
Hi Mirae,
can you please measure the voltage between CP and PE on EVSE side before connecting and after connecting?
BR,
Andreas
CODICO

JIN-MIRAE, 2023-08-07:
Hello Andreas,

Test condition is that:
1. EVSE was connected
[image]
2. CP of EVSE is positive(+) and PE of EVSE is negative(-)
[image]

A result is that:
Connected : AC 0.33V / DC -355mV
NOT Conncected : AC 0.73V / DC -0.6V

Is there a problem of voltage measurment?
If so, let me know that.

Thank you

JIN-MIRAE, 2023-08-07:
I found a problem of voltage measurment, so I inform you.
The problem is that measured values are changing continuously.
Even if when I tried measurment directly on pin as below image, same problem happened.
[image]

lho-stx, 2023-08-07:
Hey @JIN-MIRAE,

you should see a constant +12V when no EV is connected (actually it is a PWM signal with 1kHz and -12V to +12V peak to peak swing and 100% duty cycle). When an EV is connected, the voltage should drop to +9V (again,  18V peak to peak without a DC bias). Now there are three possibilities:

- the measurement was not done right (I do not think this is a problem here)
- you multimeter is faulty (I also do not think this is a problem)
- the whitebeet is faulty (what I think is the most likely)

Just to be sure:
If you can, connect the CP to a oscilloscope and see, if there is a PWM signal at all an what peak to peak voltage it has. To do this please change `self.whitebeet.controlPilotSetDutyCycle(100)` in Evse.py to `self.whitebeet.controlPilotSetDutyCycle(50)` and do the measurement. You should see a 1 kHz PWM with 50 % duty cycle, centered around 0V and with 18V peak to peak voltage.

Please feel free to share your oscilloscope measurement!

If there is no PWM at all, please get in touch with CODICO and please close this issue, as this is a problem not related to the Whitebeet firmware or FreeV2G software.

Thank you for contributing,
lho

adirschl, 2023-08-08:
Hey @JIN-MIRAE,

as this seems to be not a software related problem please open a ticket at CODICO by writing an email to plc@codico.com.
please do following measurements:

1) voltage between CP and PE on the WHITEbeet-EI (in the unconnected state, **not connected** with WHITEbeet-PI)
-> you should measure +12V, if not here is a hardware problem on the WHITEbeet board.

2) all other measurements should be done by oscilloscope, as there will be a 1kHz PWM signal with different voltages and dutycycles present.
-> after connecting the WHITEbeet boards there should be a PWM with +9V and -12V (only the positive halfwave is reduced in voltage due to a serial diode in the EV circuitry), dutycycle dependant on the software settings.

BR,
Andreas Dirschl,
CODICO

lho-stx, 2023-08-08:
Closing this issue, because it is not software related.

closed 2023-08-08 by lho-stx