Sevenstax/FreeV2G #211: WB-EI (EVSE) to WB-PI (EV) unstable setup.

TESolution, 2023-08-02:
Platform: EVSE
White beet carrier board WB-CARRIER-BOARD-EI
Firmware Version: V02_00_01
Host Controller Interface: Ethernet

Platform: PEV
White beet carrier board WB-CARRIER-BOARD-PI
Firmware Version: V01_00_07
Host Controller Interface: Ethernet

I prepared development setup using two White beet boards. I am able to set a communication between them once per a few attempts. 

Steps to reproduce the behaviour.
[image]

1. Connect boards according to setup presented on the picture above. 
2. Run EV application 
3. Run EVSE application 
// The order in which the application is launched for each side doesn't matter in my experiments.

I would expect behaviour presented on _Development WB-EI to WB-PI - Succeed.jpg_ each time when I run application
[image], while usually it is not possible to set a communication between boards. 
_Development WB-EI to WB-PI - Failed_1.jpg_
_Development WB-EI to WB-PI - Failed_2.jpg_

[image]
[image]

lho-stx, 2023-08-03:
Hey @TESolution,

thank you for using the issue template and giving detailed information about your setup!

I will try to reproduce your issue and come back to you as soon as possible.

Best regards,
lho

lho-stx, 2023-08-08:
Hey @TESolution,

today I ran sessions for 10 minutes straight. I used the latest version of the master branch, choosed an energy amount so the charging lasted several seconds (changed this between sessions). I even stopped some session by pressing `ctrl+c`, let FreeV2G timeout, etc.

On the EV side, the error is that the session started notification was send unexpectedly.

I was able to start the charging session every single time. I am using a Linux machine as a host. We had issues with timing on windows before, but this was sorted out by using a new python library and some users worked around this issue by elevating the FreeV2G process priority.

I kindly ask you to create wireshark logs of a failing session for EV and EVSE. You need to enable the portmirror by passing `--portmirror` to the application.

Best regards,
lho

TESolution, 2023-08-08:
Hello @lho-stx, 

Could you advise me what are recommended Python Packages versions?

In general, I am trying to follow User Manual, but finally I am working with following versions:
[image]

Indeed, I am using Windows. When I tried virtualized Linux I had problem with ethernet network cards. I could switch bench setup to RPi instead of Windows, which is my goal anyway. EVSE White beet should be control through SPI. The exercise I am trying to do now, using two WBs under Windows is just to proof the concept and better understanding data exchange. 

And some more approaches with changed charging and battery properties: 
[image]

I also tried to run application on the both boards with disconnected CP. It's been connected on application request. 
[image]
[image]

Best regards, 
TES

lho-stx, 2023-08-09:
Hey @TESolution,

on Linux `pip list` in the virtualenv gives me:

```bash
Package           Version
----------------- -------
asttokens         2.2.1
backcall          0.2.0
Cython            3.0.0
decorator         5.1.1
executing         1.2.0
ipython           8.14.0
jedi              0.19.0
matplotlib-inline 0.1.6
parso             0.8.3
pexpect           4.8.0
pickleshare       0.7.5
pip               23.2.1
prompt-toolkit    3.0.39
ptyprocess        0.7.0
pure-eval         0.2.2
Pygments          2.16.1
python-libpcap    0.4.1
scapy             2.5.0
setuptools        68.0.0
six               1.16.0
stack-data        0.6.2
traitlets         5.9.0
typing_extensions 4.7.1
wcwidth           0.2.6
wheel             0.41.0
```

Regarding a virtualized linux environment:

You need to use an actual virtual machine. You need to pass through the network interface directly (in VirtualBox the option is called `Bridged Adapter` and you may also need to enable promiscuos mode). WSL is not supported.

We are improving the documentation at the moment and this is still a work in progress, but maybe this [wiki entry](https://github.com/Sevenstax/FreeV2G/wiki/guide-quick-start#use-case-spi-raspberrypi-example) helps you setting up RaspberryPi and Whitebeet!

Thank you for trying the CP disconnect/reconnect, although it did not change anything... I will further try to reproduce your problem.

Best regards,
lho

TESolution, 2023-08-10:
Hello @lho-stx , 

May I ask what is your Linux distribution as well as Python version for desktop development? 
I am facing issue with "python-libpcap" package installation due to incompatibility with Python 3.10. Trying to downgrade to 3.7... 

In general I have the same problem on RPi, also with "scapy" package. 

Best regards,
TES

TESolution, 2023-08-15:
I will be grateful for checking my RPi logs:
[RPi_log.txt](https://github.com/Sevenstax/FreeV2G/files/12353322/RPi_log.txt)

I am using _raspbian-plc-8gb.img.gz_ image. 

Unfortunately I have not progressed since last time on Linux neither. I installed the same packages as you provided, but I have still some problems with "scape" - the same error as you can see in the log from RPi, and another problem with "python-libpcap", which was solved by creating virtual environment under Linux, but "pylibpcap" issue still exists what causes script crash due to lack of package installed.

lho-stx, 2023-08-16:
Hello @TESolution,

I checked your logs. Can you please do me a favor and show me the output of `ls /dev/`? It seems the name of the spi device is off...

Regarding the Python version:
on my Pi it uses 3.9.2

Thank you,
lho

TESolution, 2023-08-17:
Hello @lho-stx,

In general I am trying to run two setups:

1. WB-PI + WB-EI, Windows PC + Virtual Linux Ubuntu 
[image]

Comment: May I ask what is your Linux distribution as well as Python version for **desktop** development?
I am facing issue with "python-libpcap" package installation due to incompatibility with Python 3.10. Trying to downgrade to 3.7...

2. WB-PI (Windows PC) + WB-EI (RPi) 
[image]

Please see attached output:
[ls_dev.txt](https://github.com/Sevenstax/FreeV2G/files/12369158/ls_dev.txt)

you can also see the result of running application using spidev0.1. 

Best regards, 
TES

lho-stx, 2023-09-04:
Hello @TESolution,

I see no problem in your setup, but can you try omitting the `-m` parameter? So just call 

```bash
python3 Application.py spi -i spidev0.1 -r EVSE
``` 

Are you sure you connected your Whitebeet RxReady pin to the correct GPIO on the Pi? Can you take a picture of your setup where I can see the hardware connection between the boards? In the meantime you can also double check the GPIO pins again:

Signal | Whitebeet | Whitebeet Carrier Board | Raspberry Pi 4 (name / wPi / BCM)
-- | -- | -- | --
CLK | PAD 24 (PD3) | J8 SCK | SPI0 SCLK / 14 / 11
MOSI | PAD 36 (PB15) | J8 MOSI | SPI0 MOSI / 12 / 10
MISO | PAD 35 (PB14) | J8 MISO | SPI0 MISO / 13 / 9
NSS | PAD 77 (PB9) | J8 NSS | GPIO.5 / 5 / 24
RxReady | PAD 37 (PD4) | J1 PD4 | GPIO.3 / 3 / 22
TxPending | PAD 38 (PD11) | J1 PD11 | GPIO.2 / 2 / 27

Also keep the host selection pins for SPI in mind:

Whitebeet | Whitebeet Carrier Board | Level
-- | -- | --
PAD 85 (PC2) | J4 PC2 | GND
PAD 84 (PA4) | J4 PA4 | VCC

Thank you,
lho

PS:
We are working on a get started guide. It is still a draft, so inforamtion may missing, but you can already [have a look](https://github.com/Sevenstax/FreeV2G/wiki/guide-quick-start#use-case-spi-raspberrypi-example).

TESolution, 2023-09-10:
Hello @lho-stx, 

There is no different when I call "python3 Application.py spi -i spidev0.1 -r EVSE". 

I am using Raspberry Pi 3 model B v1.2. Pinout is a bit different: 

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/Pawel/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/Pawel/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

Signal | Whitebeet | Whitebeet Carrier Board | Raspberry Pi 3 B
-- | -- | -- | --
CLK | PAD   24 (PD3) | J8   SCK | SPI0   SCLK / 23
MOSI | PAD   36 (PB15) | J8   MOSI | SPI0   MOSI / 19
MISO | PAD   35 (PB14) | J8   MISO | SPI0   MISO / 21
NSS | PAD   77 (PB9) | J8   NSS | GPIO.8   / 24
RxReady | PAD   37 (PD4) | J1   PD4 |  
TxPending | PAD   38 (PD11) | J1   PD11 |  

</body>

</html>

Regarding get started guide - could you provide following files:
- requierements_spi.txt
- requierements_eth.txt

Here you can see my SPI connection:
[image]
[image]
[image]

There is worth to add that I tried many combinations of:
- CS on RPi (GPIO.8 CS0 / GPIO.7 CS1)
- WB PC2 left floating or GND
- WB PA4 left floating or VCC (3.3V)

lho-stx, 2023-09-11:
Hey @TESolution,

The pinout on all Raspberry Pi Models are the same. I think I see the problem: The table I posted sues `wiring Pi` and `BCM` pin numbers, not physical pin numbers. I apologize for the confusion, try the following wiring:

```
                +-----+-----+---------+------+---+---Pi 4B--+---+------+---------+-----+-----+
                | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
                +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
                |     |     |    3.3v |      |   |  1 || 2  |   |      | 5v      |     |     |
                |   2 |   8 |   SDA.1 | ALT0 | 1 |  3 || 4  |   |      | 5v      |     |     |
                |   3 |   9 |   SCL.1 | ALT0 | 1 |  5 || 6  |   |      | 0v      |     |     |
                |   4 |   7 | GPIO. 7 |   IN | 1 |  7 || 8  | 1 | IN   | TxD     | 15  | 14  |
                |     |     |      0v |      |   |  9 || 10 | 1 | IN   | RxD     | 16  | 15  |
                |  17 |   0 | GPIO. 0 |   IN | 0 | 11 || 12 | 0 | IN   | GPIO. 1 | 1   | 18  |
TxPending/PD11->|  27 |   2 | GPIO. 2 |   IN | 1 | 13 || 14 |   |      | 0v      |     |     |
RxReady/PD4---->|  22 |   3 | GPIO. 3 |   IN | 1 | 15 || 16 | 0 | IN   | GPIO. 4 | 4   | 23  |
                |     |     |    3.3v |      |   | 17 || 18 | 1 | IN   | GPIO. 5 | 5   | 24  |<-NSS/PB9
MOSI/PB15------>|  10 |  12 |    MOSI | ALT0 | 0 | 19 || 20 |   |      | 0v      |     |     |
MISO/PB14------>|   9 |  13 |    MISO | ALT0 | 0 | 21 || 22 | 0 | IN   | GPIO. 6 | 6   | 25  |
SCLK/PD3------->|  11 |  14 |    SCLK | ALT0 | 0 | 23 || 24 | 1 | OUT  | CE0     | 10  | 8   |
                |     |     |      0v |      |   | 25 || 26 | 1 | OUT  | CE1     | 11  | 7   |
                |   0 |  30 |   SDA.0 |   IN | 1 | 27 || 28 | 1 | IN   | SCL.0   | 31  | 1   |
                |   5 |  21 | GPIO.21 |   IN | 1 | 29 || 30 |   |      | 0v      |     |     |
                |   6 |  22 | GPIO.22 |   IN | 1 | 31 || 32 | 0 | IN   | GPIO.26 | 26  | 12  |
                |  13 |  23 | GPIO.23 |   IN | 0 | 33 || 34 |   |      | 0v      |     |     |
                |  19 |  24 | GPIO.24 |   IN | 0 | 35 || 36 | 0 | IN   | GPIO.27 | 27  | 16  |
                |  26 |  25 | GPIO.25 |   IN | 0 | 37 || 38 | 0 | IN   | GPIO.28 | 28  | 20  |
                |     |     |      0v |      |   | 39 || 40 | 0 | IN   | GPIO.29 | 29  | 21  |
                +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
                | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
                +-----+-----+---------+------+---+---Pi 4B--+---+------+---------+-----+-----+
```

With best regards,
lho

TESolution, 2023-09-11:
Hi @lho-stx, 

I understand you was wondering the CS was wrongly connected to RPi GPIO8(CS0) or GPIO7(CS1) instead of GPIO24 (RPi Physical 18).  

[image]

I connected WB CS (NSS) to GPIO24 (RPi Physical 18) and it does not help. 

Best regards, 
TES

lho-stx, 2023-09-12:
Hey @TESolution,

Regarding the pin out:

the script uses the BCM pin numbering, therefore connecting NSS to phyiscal pin 18 is correct.

Regarding the software versions:

You always can use the latest firmware with the latest master branch (this is true for both, EV and EVSE). If you are using Whitebeet version V02_00_01 you can use the latest tag for this version:

https://github.com/Sevenstax/FreeV2G/tree/EVSE_v2.0.1_4

(but you could also use the latest master branch, because the major versions are backwards compatible).

I have copied the setup, but I only have access to the Raspberry Pi 3B (without +). Nevertheless, here is my hardware setup:

[image]

I also set the bootstrap pins:

Whitebeet | Whitebeet Carrier Board | Level
-- | -- | --
PAD 85 (PC2) | J4 PC2 | VCC
PAD 84 (PA4) | J4 PA4 | GND

I then pulled the latest master from FreeV2G.

This setup worked for me!

Here are the log files:

[whitebeet_raspi_3B_freev2g.txt](https://github.com/Sevenstax/FreeV2G/files/12583554/whitebeet_raspi_3B_freev2g.txt)
[whitebeet_raspi_3B_whitebeet.txt](https://github.com/Sevenstax/FreeV2G/files/12583555/whitebeet_raspi_3B_whitebeet.txt)

I also created a SPI log with [Saleae Logic2](https://www.saleae.com/de/downloads/). It is free of charge and you can download it to view the log file. I needed to zip it due to github:

[whitebeet_raspi_3B_spi.zip](https://github.com/Sevenstax/FreeV2G/files/12583607/whitebeet_raspi_3B_spi.zip)

Please double check your cabling and compare it to my setup.

Thank you,
lho

PS:
Just for my sanity: You did connect the GND from Whitebeet to GND of the RaspberryPi? I can not see it well on the posted picture.

TESolution, 2023-09-12:
Hello @lho-stx,

Yes, I had GND connected between RPi (Physical 25) and WB (J8 GND). 

I decided to use clean copy of RPi OS instead of raspbian-plc-8gb.img.gz. I installed everything from the beginning based on provided pip list and finally I run EVSE through SPI. Now, I am trying to run stable configuration of EV side. I have WB-PI connected to Windows PC via ethernet and results are more or less the same as at the beginning of investigation. Script sometimes starts and crashes during running or even does not start. I am trying to connect WB-PI to the same RPi board (to get rid of Windows host) but using ethernet connection. 
Unfortunately following commands give me errors
python3 Application.py eth -i "eth0" -m C4:93:00:34:CF:CF -r EV
python3 Application.py eth -i eth0 -m C4:93:00:34:CF:CF -r EV

I also tried combinations of near MAC addresses: from *:CF to *:D1. 

[RPi_ETH_EV.txt](https://github.com/Sevenstax/FreeV2G/files/12590165/RPi_ETH_EV.txt)

1. What do I need to run application on RPi using ethernet connection? 
2. My EV board (WB-PI) uses firmware V01_00_07, I was not able to find any .fwu file on codico download. Is V01_00_07 the latest? 
3. Which Linux distribution do you use in your VirtualBox host? Debian only? 
4. Using Ubuntu 64bit (bridged network and enabled promiscuous mode) no success. 
[ubuntu_ETH_EV.txt](https://github.com/Sevenstax/FreeV2G/files/12590529/ubuntu_ETH_EV.txt)
5. Could you share requierements_eth.txt mentioned in the guide-quick-start? 

Best regards, 
TES

lho-stx, 2023-09-18:
Hey @TESolution,

1. You need to run the application with elevated rights. The get started guide is done now, [please follow the instructions in the guide](https://github.com/Sevenstax/FreeV2G/wiki/guide-quick-start#linux-debian) especially the "Usage FreeV2G"
2. 1.0.7 is the latest
3. I run Linux. I use Debian 12.1 with Python 3.11 (recently switched from Debian 11, Python 3.7).
4. You are missing some python packages. Please refer to the [get started guide](https://github.com/Sevenstax/FreeV2G/wiki/guide-quick-start#linux-debian)
5. The master branch will be updated soon, the files are in there. You can also get them from the [pull request](https://github.com/lho-stx/FreeV2G/tree/add_evse_cert)

Best regards,
lho

TESolution, 2023-09-18:
Hello @lho-stx, 

I tried following the instruction... No progress on neither RPi nor Debian. 
[debian_userguide.txt](https://github.com/Sevenstax/FreeV2G/files/12652390/debian_userguide.txt)
[RPi_ETH_EV_2.txt](https://github.com/Sevenstax/FreeV2G/files/12652391/RPi_ETH_EV_2.txt)

Debian with Python 3.11.2
RPi with Python 3.9.2

Can you see 
TES

lho-stx, 2023-09-19:
Hey @TESolution,

may I ask you which version of Debian you have installed? In my Debian 12.1 installation and in the Debian 11 installation I can install the `python3-venv` and `libpcap-dev`

It looks like `libpcap-dev` is missing on you Pi, did you try to install it?

BR,
lho

TESolution, 2023-09-19:
Hello @lho-stx,

I have created 3 separated virtual machines using official releases: 

debian-12.1.0-amd64-DVD-1.iso
debian-12.1.0-amd64-netinst.iso
debian-12.1.0-i386-DVD-1.iso

The same effect on all of them. 

Yes, I installed libpcap-dev, but I still got "No module named 'pylibpcap'" error. 
[RPi_ETH_EV_libpcap.txt](https://github.com/Sevenstax/FreeV2G/files/12657082/RPi_ETH_EV_libpcap.txt)

Best regards, 
TES

lho-stx, 2023-09-20:
Hey @TESolution,

I also installed Debian 12.1 from `debian-12.1.0-amd64-netinst.iso`. I am sorry, but it is really hard to debug your VM setup, because these problems are not related to FreeV2G itself.

Did you run

```
apt update
```

as root before installing the packages? Maybe the repo database is out of date?

For me I can install the packages from the repositories:

```
lho@stx-pc-lho:~$ apt list | grep python3-venv
python3-venv/stable,now 3.11.2-1+b1 amd64 [installed]
lho@stx-pc-lho:~$ apt list | grep libpcap-dev
libpcap-dev/stable,now 1.10.3-1 amd64 [installed]
```
Best regards,
lho

TESolution, 2023-09-20:
Hello @lho-stx,

Yes, I did run "apt update"
[image]

Repo has been cloned by command:
[image]

I solved previous problem with missing packages and problem to run commands. 

- I use "enp0s8" as an interface on virtual machine, it it correct? 
[image]

Please see the log:
[vboxuser DebianNet Eth.txt](https://github.com/Sevenstax/FreeV2G/files/12677911/vboxuser.DebianNet.Eth.txt)

Best regards, 
TES

lho-stx, 2023-09-29:
Hey @TESolution,

yes you would need to use the `enp0s8` interface in your vm. But you have multiple interfaces enabled in the VM, are you sure you used the correct one? I the log it seems you tried different interfaces. Can you please create a wireshark log of your failing session? Also you are using sudo as a root user, it is not necessary, because you are already root. 

Best regards,
lho

TESolution, 2023-10-17:
Hi @lho-stx, 

Just to not leave it without an answer... I found a solution, which is under testing now. 
I had an issues with Linux virtual machine. 

Yes, the mentioned interface (enp0s8) was correct. In my case sudo is necessary despite the operating under root. 

Thank you for support. 

Best regards, 
TES

lho-stx, 2023-10-27:
Hey @TESolution,

thank you for your feedback! If you have anymore questions, please let me know!

Best regards,
lho

closed 2023-10-27 by lho-stx