Sevenstax/FreeV2G #212: How to set an environment to use White beet together with Raspberry Pi through SPI?

TESolution, 2023-08-02:
Among available documentation _AppNote_norm_Raspberry_Pi_for_QCA7005_rev.1.0.pdf_ seems to be insufficient. 
I already prepared Raspberry board using raspbian-plc-8gb.img.gz image. 

I would like to start the communication between boards using any interface ETH or SPI. The goal I would like to achieve is to run EVSE application on Raspberry Pi + WB-EI and setup a communication with any host (Raspberry Pi or PC) + WB-PI usingany interface (ETH or SPI).

lho-stx, 2023-08-03:
Hey @TESolution,

you are referring to documentation not provided by SEVENSTAX, therefore I can not give you any information about the mentioned document. 

But we are currently working on improving documentation and creating quick start guides for users to follow along. This will include setting up FreeV2G on Raspberry Pi using SPI. Please be patient, the guides will be found soon in the wiki section of this repository. There will be an announcement when the guides are uploaded!

Best regards,
lho

closed 2023-08-03 by lho-stx

adirschl, 2023-08-03:
Hey @TESolution,

the purpose of the mentioned AppNote is not, to describe how to use Raspberry Pi with WHITEbeet, it is ment for communicating with REDbeet.
For support request regarding REDbeet or the mentioned AppNote please open a ticket by writing an email to plc@codico.com.

BR,

Andreas Dirschl
FAE 
CODICO

TESolution, 2023-08-03:
Hello @lho-stx , @adirschl , 

There is only one document on PLC Documentation Area, which referes to RPi, that's why I mentioned it. 

Does it mean that at this moment there is no documentation how to use RPi with WB? 
What does raspbian-plc-8gb.img.gz contain of and isn't it a pre-configured environment? If I wrongly adressed my questions, I can switch to Codico support. 

Best regards

adirschl, 2023-08-04:
Hello @TEsolution,

as mentioned above, for support requests that do not relate to WHITEbeet software please contact CODICO support (plc@codico.com).

WHITEbeet software related topics are documented in the WHITEbeet software user manual (Sevenstax_WHITEBEET_Software_usermanual_r2.17.pdf) and https://github.com/Sevenstax/FreeV2G 

There is documented how to use FreeV2G with a linux system (like you usually have on a RaspberryPi) in the "Getting FreeV2G" and "Getting Started" sections of FreeV2G

raspbian-plc-8gb.img.gz is part of the REDbeet documentation. It contains a preconfigured raspbian for RaspberryPi3 to communicate with QCA7005 on REDbeet (what is described in the AppNote_norm_Raspberry_Pi_for_QCA7005_rev.1.0.pdf).

BR,
Andreas Dirschl
FAE
CODICO