Sevenstax/FreeV2G #214: Question: EV is stopping after pre-chrage. Why is EV not able to send the "power Delivery Request to EVSE"? Do we need a seperate EV sub id in whitebeet.py

ArijitChatterjee-BlinkCharging, 2023-08-07:
Setup-
Platform- EV
Firmware version - V01_00_06
Host Controller interface - Ethernet

code- master 

Description of the question-
We are using WB-CARRIER-BOARD EV and running freeV2G. The EV sends the pre-charge requset and the EVSE responds the pre-charge. After that the EV goes in the charging ready state but cannot send the power Delivery request. After that the EV self.state = session stopped is sent and the EV stops the 
[CodicoEV_ApplicationLOG.txt](https://github.com/Sevenstax/FreeV2G/files/12272679/CodicoEV_ApplicationLOG.txt)
[CodicoEV_UART_DebugLog.txt](https://github.com/Sevenstax/FreeV2G/files/12272682/CodicoEV_UART_DebugLog.txt)
process.
We have checked the logs of EVSE where the pre-charge is PASS an
[CodicoEV_ApplicationLOG.txt](https://github.com/Sevenstax/FreeV2G/files/12272684/CodicoEV_ApplicationLOG.txt)
[CodicoEV_UART_DebugLog.txt](https://github.com/Sevenstax/FreeV2G/files/12272685/CodicoEV_UART_DebugLog.txt)
d waiting for the Power Delivery Request.

I have attached the Application logs and the logs.
Please give me a solution.

nststx, 2023-08-07:
Hi @ArijitChatterjee-BlinkCharging,

Thanks for submitting your issue. I've taken a look at your logs and it seems like the EV isn't able to serialize it's PowerDeliveryReq. This is most likely due to an incorrect charging profile. To investigate this issue further I will need a Wireshark log of the V2G communication. To capture such a log please enable the port mirror on the EV Whitebeet, connect it directly to your PC and capture a Wireshark log on the corresponding interface. If you're on the newest FreeV2G version you can enable the port mirror with the `--portmirror` command line option.

If you need any further help in capturing the log, please let me know!

Best regards,
Niklas Stoffers

ArijitChatterjee-BlinkCharging, 2023-08-07:
> Hi @ArijitChatterjee-BlinkCharging,
>
> Thanks for submitting your issue. I've taken a look at your logs and it seems like the EV isn't able to serialize it's PowerDeliveryReq. This is most likely due [...]

Here is the wireshark logs you asked for.

ArijitChatterjee-BlinkCharging, 2023-08-07:
Here is the wireshark logs you asked for

[214.zip](https://github.com/Sevenstax/FreeV2G/files/12274010/214.zip)

nststx, 2023-08-07:
Hi @ArijitChatterjee-BlinkCharging,

Thanks for coming back with the logs. There is indeed an issue with the ChargingProfile. However it's not in the way it's set but rather in the way that the EV firmware is trying to serialize it. Unfortunately we don't have a new firmware version yet that fixes this issue. This issue only persists with DIN protocol and we will fix it with an upcoming firmware release. 
In the meantime you can do a workaround by simply not setting the charging profile. You can try using the EV_v1.0.6_1 tag in which the charging profile is not being set.

Please let me know if you are able to proceed any further with the suggested workaround!

Best regards,
Niklas Stoffers

ArijitChatterjee-BlinkCharging, 2023-08-08:
Hi @nststx ,
I tried running it from the EV_v1.0.6_1 tag. Its the same logs which is getting generated.
Best Regards
Arijit

nststx, 2023-08-09:
Hi @ArijitChatterjee-BlinkCharging,

Thanks for your response. I looked into this again and I was actually wrong about the EV_v1.0.6_1 tag not setting the charging profile. I'm sorry for that. Thus to disable setting the charging profile please checkout the EV_v1.0.6_1 tag again and remove or comment out the `self.whitebeet.v2gSetChargingProfile(self.schedule)` in *Ev.py*. If you haven't made any modifications this should be in line 543.

Please let me know if you were able to proceed any further with the above suggestion!

Best regards,
Niklas Stoffers

ArijitChatterjee-BlinkCharging, 2023-08-10:
Hello @nststx ,
I commented out "self.whitebeet.v2gSetChargingProfile(self.schedule)" in Ev.py. Now I am getting a different output where the "Session Started" is getting received but after that it is not going to the further processes but the program is not exiting from the loop and the program is not stopping it is just freezing after "Session started". 
I am mentioning the Application logs below. The "Session stopped" is coming when I am disconnecting the EV from the EVSE. Otherwise it is stuck in "Session started".
Best Regards
Arijit

Welcome to Codico Whitebeet EV reference implementation
Initiating framing interface
iface: ETH, name: enp2s0, mac: c4:93:00:34:d0:02
WHITE-beet-PI firmware version: V01_00_06
Set the CP mode to EV
Start the CP service
Set the CP state to State B
Set SLAC to EV mode
Start SLAC
EVSE connected
Start SLAC matching
SLAC matching successful
Set V2G mode to EV
Set V2G configuration
Set DC charging parameters
Create new charging session
ControlPilot duty cycle: 5.5
"Session started" received
	Protocol: 0
	Session ID: d2052b757f78d464
	EVSE ID: 051a18412a
	Payment method: 0
	Energy transfer mode: 1
"Session Stopped" received
Change State to State B
EV loop finished
Goodbye!

ArijitChatterjee-BlinkCharging, 2023-08-10:
Hello @nststx,
Here are the UART logs and the wireshark logs you asked for.
Best Regards
@ArijitChatterjee-BlinkCharging 

[CodicoEV_UART_DebugLog.txt](https://github.com/Sevenstax/FreeV2G/files/12311247/CodicoEV_UART_DebugLog.txt)
[EV.zip](https://github.com/Sevenstax/FreeV2G/files/12311263/EV.zip)

nststx, 2023-08-10:
Hi @ArijitChatterjee-BlinkCharging,

Thanks for coming back with the logs. The session stops because the EVSE is still processing its authorization. I don't know which charging station you're using but you'd have to take a look how your charging station requests for authorization. Maybe it comes with an app or something where you authorize the charging or there is some setting that you need to change. This is specific to your charger.

Best regards,
Niklas Stoffers

ArijitChatterjee-BlinkCharging, 2023-08-10:
Hi @nststx,
EV is not going to the authorization request. EVSE is still waiting to respond to "Session Setup request". 
Any suggestions on that.
Best Regards
@ArijitChatterjee-BlinkCharging

nststx, 2023-08-10:
Hi @ArijitChatterjee-BlinkCharging,

The log you've sent contains a full session up to the point of the failing authorization. Thus I dont understand why the EVSE is signaling that it's still waiting for session setup because the session setup did run successfully (EV sent a request and the EVSE also responds). This is likely to be a bug of your charger. You can try restoring your charger to factory settings if possible.

Best regards,
Niklas Stoffers

closed 2023-09-01 by github-actions[bot]