Sevenstax/FreeV2G #216: Cannot start pre charge status on Benz EQE

FivaTagin, 2023-08-15:
**Your Setup**
Platform: EVSE
Firmware Version: i.e. V01_00_05
Host Controller Interface: Ethernet 
Host: FreeV2G (EV_v1.0.4_0) 

**Describe the bug**
We are not able to start the pre charge process when we using white-beet board with FreeV2G on Benz EQE
The process will directly terminate itself after get into the pre-charge process, according to the design guide, the EV shall wait certain amount of time for charger raising the voltage. however, the application looks get something wrong and directly end itself. 

[image]

[image]

[image]

**To Reproduce**
We use sudo .venv/bin/python3 Application.py eth -i eth0 -m c4:93:xx:xx:xx:xx -r EVSE to trigger the application.

**Expected behavior**
the application shall react the same behaviour like this repo shows.

**Logs**
[20230811_143705_.log](https://github.com/Sevenstax/FreeV2G/files/12341773/20230811_143705_.log)

REFERENCE APPLICATION:
  Name:      ISO15118 (EVSE)
  Version:   V02_01_00
  Target:    WHITE-BEET
  CPU:       STM32F745

 Hostcontroller Interface: ETHERNET!

APPLICATION RUNNING ...
[    0.010] [NOTICE] SLAC: stxSLAC_Init()
[    0.011] [NOTICE] APL-ETH:   MAC of NIC[0]:  c4:93:00:48:b0:99
[    3.054] [NOTICE] APL-ETH:   MAC of NIC[1]:  c4:93:00:48:b0:98
[    3.054] [NOTICE] APL-MAIN: BOOT-CFG: Boot QCA7005 firmware from host
[    3.055] [NOTICE] APL-MAIN: BOOT-CFG: FW2
[    3.056] [NOTICE] APL-MAIN: BOOT-CFG: Module configured to EVSE
[    3.057] [NOTICE] PLCV-HL:  Boot from host selected
[    3.059] [NOTICE] V2GCPS: stxV2GCPS_RegisterCallback()
[    3.069] [NOTICE] APL-ETH: Got link:
[    3.069] [NOTICE] APL-ETH:     Interface: eth1
[    3.070] [NOTICE] APL-ETH:     MAC:       C4:93:00:48:B0:98
[    3.367] [WARN  ] PLC-HL: stxPLC_HL_Send() - IF not linked!
[    3.368] [NOTICE] PLCV-HL: Reset QCA700x
[    3.369] [WARN  ] PLC-HL: stxPLC_HL_Send() - IF not linked!
[    3.542] [WARN  ] PLC-HL: stxPLC_HL_Send() - IF not linked!
[    3.849] [WARN  ] PLC-HL: stxPLC_HL_Send() - IF not linked!
[    4.834] [NOTICE] APL-MAIN:
[    4.834] [NOTICE] APL-MAIN: =============================
[    4.835] [NOTICE] APL-MAIN: NEW LAN CONFIGURATION (IPv6):
[    4.836] [NOTICE] APL-MAIN: for IP Config handle [fe01]
[    4.836] [NOTICE] APL-MAIN: IPv6 Addresses:
[    4.837] [NOTICE] APL-MAIN:   Link-Local:  fe80:0:0:0 : c693:ff:fe48:b098
[    4.838] [NOTICE] APL-MAIN:   Local: 0:0:0:0 : 0:0:0:0
[    4.838] [NOTICE] APL-MAIN:   DNS1:  2001:4860:4860:0 : 0:0:0:8888
[    4.839] [NOTICE] APL-MAIN:   DNS2:  fd00:0:0:0 : be05:43ff:fe44:5280
[    4.840] [NOTICE] APL-MAIN: =============================
[    4.841] [NOTICE] APL-MAIN:
[    4.876] [WARN  ] PLC-HL: stxPLC_HL_Send() - IF not linked!
[    4.899] [WARN  ] PLC-HL: stxPLC_HL_Send() - IF not linked!
[    5.184] [WARN  ] PLC-HL: stxPLC_HL_Send() - IF not linked!
[    5.490] [WARN  ] PLC-HL: stxPLC_HL_Send() - IF not linked!
[    6.175] [WARN  ] PLC-HL: stxPLC_HL_Send() - IF not linked!
[    6.282] [NOTICE] PLCV-HL:  =======================================
[    6.282] [NOTICE] PLCV-HL:    FW and PIB was loaded successful
[    6.797] [NOTICE] PLCV-HL:  =======================================
[    6.798] [NOTICE] PLCV-HL:    BOOT FROM HOST READY
[    6.949] [NOTICE] PLC-HL:  Own PLC MAC = C4:93:00:48:B0:9A
[    7.344] [WARN  ] PLC-HL: stxPLC_HL_Send() - IF not linked!
[    8.056] [WARN  ] PLC-HL: stxPLC_HL_Send() - IF not linked!
[    8.127] [WARN  ] PLC-HL: stxPLC_HL_Send() - IF not linked!
[    8.608] [NOTICE] APL-ETH: Got link:
[    8.609] [NOTICE] APL-ETH:     Interface: plc0
[    8.609] [NOTICE] APL-ETH:     MAC:       C4:93:00:48:B0:99
[    9.689] [NOTICE] APL-MAIN:
[    9.689] [NOTICE] APL-MAIN: =============================
[    9.690] [NOTICE] APL-MAIN: NEW LAN CONFIGURATION (IPv6):
[    9.690] [NOTICE] APL-MAIN: for IP Config handle [fe00]
[    9.691] [NOTICE] APL-MAIN: IPv6 Addresses:
[    9.692] [NOTICE] APL-MAIN:   Link-Local:  fe80:0:0:0 : c693:ff:fe48:b099
[    9.693] [NOTICE] APL-MAIN:   Local: 0:0:0:0 : 0:0:0:0
[    9.693] [NOTICE] APL-MAIN:   DNS1:  2001:4860:4860:0 : 0:0:0:8888
[    9.694] [NOTICE] APL-MAIN:   DNS2:  fd00:0:0:0 : be05:43ff:fe44:5280
[    9.695] [NOTICE] APL-MAIN: =============================
[    9.696] [NOTICE] APL-MAIN:
[   33.257] [NOTICE] SLAC: stxSLAC_Stop()
[   33.263] [WARN  ] V2GCPS: stxV2GCPS_Stop() - Failed
[   33.302] [NOTICE] SLAC: stxSLAC_Start()
[   35.303] [NOTICE] PLCV-HL: PIB File was updated => Download PIB without reset
[   86.112] [NOTICE] SLAC: stxSLAC_StartSlacMatchingProcess()
[   86.153] [NOTICE] V2GCPS: Change State from A to B
[   86.153] [NOTICE] V2GCPS: stxXBAR_V2GCPS_NotifyHandler()
[   86.154] [NOTICE] V2GCPS: # V2GCPS_INFOTYPE_CP_STATE
[   86.154] [NOTICE] V2GCPS: STATE B
[   89.920] [NOTICE] SLACSERV: stxSLACSERV_NotifyCB_Started()
[   90.200] [WARN  ] SLAC:   Unexpected message!
[   90.240] [WARN  ] SLAC:   Unexpected message!
[   91.326] [NOTICE] SLAC: SLAC matching was successful!
[   91.326] [NOTICE] SLACSERV: stxSLACSERV_NotifyCB_Success()
[   91.332] [NOTICE] V2GSVC: Activate external payment option
[   91.333] [NOTICE] V2GSVC: Deactivate contract payment option
[   91.334] [NOTICE] V2GAPP: Successfully set configuration
[   91.335] [NOTICE] V2GAPP:   Protocols: DIN, ISO
[   91.335] [NOTICE] V2GAPP:   Energy modes: DC_Core, DC_Extended, DC_Combo,
[   91.336] [NOTICE] V2GAPP:   Payment options: External
[   91.337] [NOTICE] V2GAPP:   Certificate installation: not supported
[   91.337] [NOTICE] V2GAPP:   Certificate update: not supported
[   91.450] [NOTICE] V2GSVC: Starting in EVSE mode.
[   91.450] [NOTICE] V2GAPP: Listen was started
[   91.465] [NOTICE] V2GSVC: TCP: Listen on port 58110 for non-TLS
[   91.466] [NOTICE] V2GSVC: TCP: Listen on port 58111 for TLS
[   92.248] [NOTICE] PLCV-HL: PIB File was updated => Download PIB without reset
[   93.120] [NOTICE] V2GSVC: TCP: A client connected on the TCP port
[   93.126] [NOTICE] V2GDIN12: EV requested to start session.
[   93.187] [NOTICE] V2GAPP: Received event 'Session started'
[   93.188] [NOTICE] V2GAPP: Received event 'Payment selected'
[   93.206] [NOTICE] V2GDIN12: EV requested to start authorization.
[   93.207] [NOTICE] V2GAPP: Received event 'Authorization Status requested'
[   94.750] [NOTICE] V2GDIN12: Message performance timeout of 1550ms reached.
[   94.751] [NOTICE] V2GDIN12: EVSE is still processing authorization.
[   95.776] [NOTICE] V2GDIN12: EV requested to start authorization.
[   96.793] [NOTICE] V2GDIN12: EVSE finished authorization
[   96.812] [NOTICE] V2GDIN12: EV requested to start charge parameter discovery.
[   96.812] [NOTICE] V2GAPP: Received event 'Schedules requested'
[   96.813] [NOTICE] V2GAPP: Received event 'Energy transfer mode selected'
[   96.814] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[   96.815] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[   96.816] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[   96.816] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[   96.817] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[   96.835] [NOTICE] V2GDIN12: EVSE finished charge parameter discovery
[   98.095] [NOTICE] V2GCPS: Change State from B to C
[   98.095] [NOTICE] V2GCPS: stxXBAR_V2GCPS_NotifyHandler()
[   98.096] [NOTICE] V2GCPS: # V2GCPS_INFOTYPE_CP_STATE
[   98.096] [NOTICE] V2GCPS: STATE C
[   98.301] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[   98.302] [NOTICE] V2GDIN12: EV requested to start cable check.
[   98.303] [NOTICE] V2GAPP: Received event 'Cable check requested'
[   98.314] [NOTICE] V2GDIN12: EVSE finished cable check
[   98.331] [NOTICE] V2GDIN12: EV requested to start pre charging.
[   98.332] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[   98.333] [NOTICE] V2GAPP: Received event 'DC charge parameters changed'
[   98.334] [NOTICE] V2GAPP: Received event 'Pre charging started'
[   98.335] [NOTICE] V2GAPP: stxXBAR_V2G_SendXBARStatus_EmptyMsg() - SubAdd = 0x88
[   98.452] [NOTICE] V2GCPS: Change State from C to B
[   98.452] [NOTICE] V2GCPS: stxXBAR_V2GCPS_NotifyHandler()
[   98.453] [NOTICE] V2GCPS: # V2GCPS_INFOTYPE_CP_STATE
[   98.454] [NOTICE] V2GCPS: STATE B
[   98.566] [NOTICE] V2GSVC: TCP: Client disconnected or connection closed by server
[   98.567] [NOTICE] V2GAPP: Received event 'Session stopped'
[   98.582] [NOTICE] V2GSVC: TCP: Listen on port 58110 for non-TLS
[   98.583] [NOTICE] V2GSVC: TCP: Listen on port 58111 for TLS
[   98.587] [NOTICE] V2GSVC: TCP: Client disconnected or connection closed by server
[   98.595] [NOTICE] SLAC: stxSLAC_Stop()
[  100.382] [NOTICE] PLCV-HL:  =======================================
[  100.383] [NOTICE] PLCV-HL:    FW and PIB was loaded successful
[  100.898] [NOTICE] PLCV-HL:  =======================================
[  100.899] [NOTICE] PLCV-HL:    BOOT FROM HOST READY

nststx, 2023-08-16:
Hi @FivaTagin,

Thanks for submitting your issue. To further investigate your issue I will need a Wireshark log of the V2G communication. To capture such a log please connect your Whitebeet directly to your PC (no switch inbetween) and enable the port mirror on your EVSE Whitebeet. To enable the port mirror please put `evse.whitebeet.networkConfigSetPortMirrorState(1)` in the *Application.py* file before `evse.loop()`. 
I'm also confused about the version numbers you're using. You said you are using an EVSE but with Firmware Version V01_00_05 and FreeV2G version EV_v1.0.4_0. However we don't have an EVSE firmware version V01_00_05 and your log reports the version V02_01_00. I'm also pretty sure that the EV_v1.0.4_0 FreeV2G version is not compatible with your EVSE version. Are you sure you're using FreeV2G version EV_v1.0.4?

If you need further help capturing the log or checking version numbers, please let me know!

Best regards,
Niklas Stoffers

FivaTagin, 2023-08-17:
Hi there, 
thanks for the reply, the correct EVSE version is :
[image]

We will upload the wireshark log ASAP on BENZ EQE,

FivaTagin, 2023-08-18:
Hi @nststx, 

the attachments are the failure log we got from the link between white-beet and our OS 

[E.zip](https://github.com/Sevenstax/FreeV2G/files/12376182/E.zip)

We notice this car will directly reject the pre-charge, but still not sure what makes this. 

Please feel free to advise any question or information if you have any concerns.

Best

Josh.

nststx, 2023-08-21:
Hi @FivaTagin,

Thanks for coming back with the logs. I've taken a look at them and what's probably causing this issue is that the PreChargeResponse still contains an invalid isolation state. You'd need to update your DC charging parameters respectively, setting the isolation level to valid. I also noticed that you don't seem to execute a cable check as it immediately finished once started. Note that executing a cable check is a mandatory security measure for charging a vehicle. Setting your isolation level to valid without executing a cable check is a violation of IEC 61851-23.

Best regards,
Niklas Stoffers

FivaTagin, 2023-08-21:
Hi @nststx Niklas, 

Thanks for the replying and point out the missing part of my experiment. 

Is this behaviour caused by different types of cars? 

Because I saw you mentioned this source code can work on Zoe on your site. 

I would appreciate it if you can advise what kind of changes I might need to put on this source code.

Much appreciated

Josh

nststx, 2023-08-22:
Hi @FivaTagin,

In general its up to the car to decide whether it wants to proceed charging with an invalid isolation level. However according to the IEC 61851-23 safety measures charging with an invalid isolation level is prohibited. Thus the charger should always carry out an isolation check and the car should always reject charging if the EVSE didn't perform an isolation check (indicated by an invalid isolation level). From our experience many cars do not fully comply to those standards, thus some cars may still continue with precharge and power delivery even if the isolation level is invalid. To set the Whitebeet's isolation level to valid you need to update the charging parameters accordingly. In the FreeV2G reference application you can do this by setting the `'isolation_level'` to 1 in the `charging_parameters` in the EVSE's charging loop. If you're on the master branch you should find this in line *140* in *Evse.py*

https://github.com/Sevenstax/FreeV2G/blob/f7039015eebd6d4cbca82fc463a31e62aa808135/Evse.py#L140

If you're currently only testing high-level communication you don't need to change anything for the cable check procedure. However if you're currently developing a charger you should always carry out the cable check according to IEC 61851-23 before applying any significant voltage and current to the DC line.

Please let me know if you're able to proceed any further with your session with the suggested changes.

Best regards,
Niklas Stoffers

closed 2023-09-13 by github-actions[bot]