Sevenstax/FreeV2G #217: SLAC Matching Process Problem

ShubhamKhapekar-Eaton, 2023-08-17:
**Your Setup**
Platform: PEV
Firmware Version: i.e. V01_00_07
Host Controller Interface: Ethernet
Host: FreeV2G

Platform: EVSE
Firmware Version: i.e. V01_01_07
Host Controller Interface: SPI
Host: Own Implementation

**Describe the bug**
Unexpected response code (0x10) to Start SLAC matching Service, while WB running as EVSE
(refer Log - WhiteBeetModule_Start_SLAC_Mtaching_Error_0x10.sal)

**To Reproduce**
On EVSE side, Codico module (EI) simulated as EVSE – using an SPI interface, the host is performing following commands in sequence
a.	Set the CP mode to EVSE
b.	Set the CP duty cycle to 100%
c.	Start the CP service
d.	Set SLAC to EVSE mode
e.	Get CP state, Check for state B
f.	Start SLAC matching
g.	Set CP duty cycle to 5%
h.	Await for SLAC matching status

On EV side, FreeV2G stack running on PC. - Python based Host MPU Application for the use with WHITE-BEET Embedded ISO15118 Module: Python script  command line output is
Set the CP mode to EV
Start the CP service
Set the CP state to State B
Set SLAC to EV mode
Start SLAC
EVSE connected
Start SLAC matching
**SLAC matching failed**
EV loop finished
Goodbye!

**Expected behavior**
The test is performed with 2 Eval Boards EI, and PI for EVSE & EV side respectively, the expected response in this case is 0x00 according to according to user manual, referring to section 16.4.5 Start SLAC Matching Process followed by Generic result code; see section 10.1.5. OR Any value between 0x00 - 0x06 indicating SLAC status but not 0x10

**Logs**
[White_Beet_SPI_logs.zip](https://github.com/Sevenstax/FreeV2G/files/12370885/White_Beet_SPI_logs.zip)

**Additional context**
1. Sometime It has been observed that white beet module goes out of sync, spilling over the data bytes to next frame. 
Example: In data frame Identifier 0x55 is observed 3 times instead of 2 bytes, Thus the EOF data byte (0xC1) gets shifted and appeared to be sent while next SPI transfer session of Size identifier frame like 0xC1 0xAA 0xAA 0x00 ... Again 1 remaining byte of length from this fame gets spilled over to next data frame like 0x0A 0x55 0x55 0x00 0x00 ... and continues (refer log - Problem_WB_Module_Lagging_By_1Byte.sal)

2. Sometimes it is observed that SPI_TX_PENDING remained HIGH forever (refer log WhiteBeetModule_SPI_TX_PENDING_Pin_Error_HIGH_All_Time.sal)

nststx, 2023-08-21:
Hi @ShubhamKhapekar-Eaton,

Thanks for submitting your issue. I've taken a look at your SPI logs. You're getting the 0x10 response code from Whitebeet for SLAC Start Matching because the SLAC module has not been started yet. You said that you set the SLAC mode to EVSE (step d). However I'm not able to see the corresponding frame in your SPI log. Prior to starting SLAC matching you need to start the SLAC module itself (mod: 0x28, sub: 0x42). Please make sure that your host implementation sends this frame and let me know if that fixed your issue. Regarding the additional issues you're facing, I will need some more time to look into them.

Best regards,
Niklas Stoffers

ShubhamKhapekar-Eaton, 2023-08-24:
Thanks for the response. I have modified the host implementation to set the SLAC mode to EVSE. This helped solved this issue, SLAC matching went successful.

closed 2023-08-24 by ShubhamKhapekar-Eaton