Sevenstax/FreeV2G #219: V2G EVSE - Set configuration Service (0x27 0x60) | Responded Malformed Data (0x04)

ShubhamKhapekar-Eaton, 2023-08-24:
**Your Setup**
Platform: PEV
Firmware Version: i.e. V01_00_07
Host Controller Interface: Ethernet
Host: FreeV2G

Platform: EVSE
Firmware Version: i.e. V01_01_07
Host Controller Interface: SPI
Host: Own Implementation

**Describe the bug**
On EVSE side, Generic result code result received as Malformed Data (0x04) for V2G EVSE Set Configuration Service 
(refer Log - ChargingSeq_SLAC_Pass_V2G_Set_Config_Fail_21Aug_Record4.sal)

To Reproduce
On EVSE side, Codico module (EI) simulated as EVSE – using an SPI interface, the host is performing following commands in sequence
a. Set the CP mode to EVSE
b. Set the CP duty cycle to 100%
c. Start the CP service
d. Set Interface
e. Set SLAC to EVSE mode
f. Get CP state, Check for state B
g Start SLAC matching
h. Set CP duty cycle to 5%
i. Await for SLAC matching status, After SLAC matching is successful
j. Set V2G mode as EVSE
k. V2G EVSE - Set configuration -----> **Error observed here Malformed data (0x40)**
l. V2G EVSE - Set DC Charging Parameters

**Expected behavior**
The test is performed with 2 Eval Boards EI, and PI for EVSE & EV side respectively, the expected response in this case is 0x00 according to according to user manual, referring to section 18.10.1 Set Configuration followed by Generic result code; see section 10.1.5.

The data sent in **V2G EVSE - Set configuration** service is exactly same as example provided in white beet user manual for 18.10.1 Set Configuration Service.
C0 27 60 00 00 2A (ReqID: 0; Payload: 42 byte)
0F 2b 34 39 2a 31 32 33 2a 34 35 36 2a 37 38 39
(EVSE ID DIN: +49*123*456*789)
0F 44 45 2a 41 32 33 2a 45 34 35 42 2a 37 38 43
(EVSE ID ISO: DE*A23*E45B*78C)
02 00 01 (Protocols DIN70121 and ISO15118 supported)
02 00 01 (Payment method EIM and PNC)
01 01 (Energy transfer mode DC extended)
00 (Certificate Installation Supported)
00 (Certificate Update Supported)
00 C1

**Logs**
[ChargingSeq_SLAC_Pass_V2G_Set_Config_Fail_21Aug_Record4.zip](https://github.com/Sevenstax/FreeV2G/files/12428595/ChargingSeq_SLAC_Pass_V2G_Set_Config_Fail_21Aug_Record4.zip)

nststx, 2023-08-24:
Hi @ShubhamKhapekar-Eaton,

Thanks for submitting your issue. I think you're using the wrong version of the Whitebeet User Manual. For EVSE boards with major firmware version 1 the command `mod:0x27 sub:0x60` is not the **Set configuration** command but rather the **Set supported protocols** command. That also explains why it's rejected with a `0x04`. Please note that between major version 2 and 1 the V2G Framing API changed quite a lot. You can find the right version of the User Manual under the approriate FreeV2G tag (*EVSE_v1.1.7_1* in this case).

Best regards,
Niklas Stoffers

nststx, 2023-08-29:
Hi @ShubhamKhapekar-Eaton,

We suggest using the latest firmware as it always contains the newest features and latest bugfixes. Updating the firmware itself is a fairly simple process if you have Ethernet access to the board as you can simply use the firmware update tool provided by us. If you only have SPI and want to integrate the update process in your host controller application please take a look at the firmware update module in the User Manual. The update procedure itself is relatively straightforward. If you encounter any errors or issues during implementation of the update or the update itself we will of course provide help and guidance on the necessary steps to fix these issues. You can obtain the firmware update files from the CODICO download area. This file can be used for every whitebeet. You don't need a separate update file per module. The update itself is a quick process and should only take a minute or two at most.

I hope I could clarify your questions. If you have any further questions, feel free to ask!

Best regards,
Niklas Stoffers

adirschl, 2023-08-29:
Hi [ShubhamKhapekar-Eaton](https://github.com/ShubhamKhapekar-Eaton),

please contact me under plc@codico.com including your full name and company email adress.
I then will create login credentials for ther CODICO download area.

BR,
Andreas Dirschl
CODICO

ChaiyotMix, 2023-09-12:
Hi @ShubhamKhapekar-Eaton and @nststx 
I encountered the same problem as you.
My module didn't accept command 27:60 and return code: 4
I'd like to know if you've been able to solve the problem yet.

My setup:
EVSE
Firmware Version: i.e. V01_01_07
Host Controller Interface: Ethernet

PEV
Firmware Version: i.e. V00_01_04
Host Controller Interface: Ethernet

and I run all program by Raspberry Pi 4 Model B 

Best regards,
Chaiyot

closed 2023-10-04 by github-actions[bot]