Sevenstax/FreeV2G #222: Question:

Maxbl74, 2023-08-30:
Hey there, 

I am getting the following issue when trying to communicate with a White Beet Pl 1.1: 
[image]

PC is connected to board via ethernet connection and board is powered up at 5V. I have successfully installed the python packages --pre scapy[basic] and Cython but the python-libpcap returned an error. Nonetheless, it seems that this package is not needed when running on Windows.

I also wanted to read the log files via the UART port but whenever I plug/unplug an usb cable, no COM port appears/disappears in the device manager.

Is that something normal? Am I missing something somewhere?

Thanks for your help.

Maxime

lho-stx, 2023-09-01:
Hey @Maxbl74,

yes you are right, on Windows python-libpcap is not needed.

Regarding your other questions:
It seems you have no winpcap nor npcap installed. Please try installing one of these and try again! Please keep me updated about your issue!

Best regards,
lho

Maxbl74, 2023-09-01:
Hello lho,

First of all, I was able to get the log info from the UART (see image below), it was a problem with the USB cable.

[image]

Now that this is working, I was able to validate the FW version running on the White Beet (V01_00_04) and the mac address (c4:93:00:34:d0:77).

Then I downloaded the code and followed the instructions given at https://github.com/Sevenstax/FreeV2G/tree/EV_v1.0.4_0.

However, I am still seeing the same issue:
 
[image]

I have tried with different MAC address (c4:93:00:34:d0:78 or c4:93:00:34:d0:79) but still get to the same point. Tried also to change "eth0" to "eth1" or "eth" but same outcome. 

I have also tried your suggestions, to install winpcap or npcap, but I get the following error when trying to install:

[image]

I was originally running with Python 3.8 and did an upgrade to 3.11 but get the same error in both versions.

Any idea(s) what I could try next?

Thank you.

lho-stx, 2023-09-04:
Hey @Maxbl74,

first of all I need to apologize for some confusion: WinPCAP/Npcap is a third party library for Windows that allows Windows users to record network packets. It is not available as Python package, but you can download it [here](https://npcap.com/).

Your issues should resolve by  installing Npcap.

Is there a reason why you are using an outdated Whitebeet firmware, or are just getting started? Please update to the latest Whitebeet firmware to get full support in the future.

Best regards,
lho

Maxbl74, 2023-09-04:
> You don't often get email from ***@***.*** Learn why this is important<https://aka.ms/LearnAboutSenderIdentification>
>
> [EXTERNAL EMAIL WARNING]
> This email was sent from outside of the organization.
> Be careful when opening links and attachments.

Hey Iho,

Thanks for your suggestions!

I will try to install Npcap and see if I get a better outcome.

Regarding the FW version, we received the board like that. Where do you get the latest revision and what tool do I need to flash the board?

Thanks!

Envoyé à partir de Outlook pour iOS<https://aka.ms/o0ukef>

closed 2023-09-26 by github-actions[bot]