Sevenstax/FreeV2G #225: Question: Error opening adapter

JIN-MIRAE, 2023-09-04:
**Your Question**
Ask your questions here.

Hello,

A error occured like image.
[image]

I did remade virtual environment again and again according to "GitHub readme".
But, the error happened continuously.

Ethernet is connected well, power is supplied well.
What is the problem?

Thank you
Mirae

lho-stx, 2023-09-04:
Hello @JIN-MIRAE,

You have initial communication between Whitebeet and FreeV2G, your virtual environment seems good. The warning

>Module did not accept command 27:a0, return code: 17

Means that the Whitebeet is not in the state of receiving the V2G set configuration command. Can you try to powercycle the Whitebeet starting FreeV2G. It seems the board is in state after a session failed ungracefully and the Whitebeet configuration is in a bad state. 

If the issue is not resolved, please follow [this guide](https://github.com/Sevenstax/FreeV2G/wiki/guide-creating-log-files) and create wireshark and UART logs. If you have questions regarding the guide please let me know, it is a work in progress but should contain all information to create the requested log files.

Thank you,
lho

JIN-MIRAE, 2023-09-05:
Fourtunately, the problem has been resloved. thank you.

By the way, I have another question.
I want to set the values of test environment, but GitHub guide only provide a comment "You can set the configuration via a configuration file in json format."
However, in FreeV2G SDK, only 'ev.json' file exist and when I used ev.json file, I could change  parameter of EV.

I want to setting other parameter also, like EVCC ID or environment of EVSE...
Is there another file of json format?
Please let me know that how can i do set the values of test environment.

Thank you

lho-stx, 2023-09-05:
Hey @JIN-MIRAE,

that are good news! May I ask what the problem was?

Regarding the configuration files:

There will be a new FreeV2G version in the near future (we do final testing right now). It will also include the possibility to use configuration files with EVSE. Until then you can change the parameters in the `Evse.py`file.

Thank you and best regards,
lho

closed 2023-09-05 by lho-stx

JIN-MIRAE, 2023-09-05:
That is really really good news!
Thank you for providing that news.

So, if new version of FreeV2G is updated, can I do set DC values & ID?
[image]

The log of DC values & ID values is printed automatically, but I hope to change them also at test.
Is it possible at new FreeV2G?

Best regards