Sevenstax/FreeV2G #227: Question: Setting values of WB_PI & EI board test environment

JIN-MIRAE, 2023-09-05:
**Your Question**
Ask your questions here.

Hi,

I got the news 'FreeV2G' will be updated in the near future.

So, if new version of FreeV2G is updated, can I do set DC values & ID?
[image]

The log of DC values & ID values is printed automatically, but I hope to change the values also at test.
Is it possible at new FreeV2G?

Best regards

lho-stx, 2023-09-06:
Hey @JIN-MIRAE,

the EVSE ID can be set with the current version of FreeV2G too! You need to modify the [`evse.config` dict in Evse.py](https://github.com/Sevenstax/FreeV2G/blob/1cd42442c5c29a68b1845e73c3972e4ab7f3b60b/Evse.py#L102).

Unfortunately there is no way to set the session ID at the moment with the current Whitebeet firmware. I will discuss internally if this is a new feature we can implement, but this needs a Whitebeet fimrware update.

Best regards,
lho

lho-stx, 2023-09-06:
Hey @JIN-MIRAE,

so when you click the lick in my last answer you will be directed to the line in the `Evse.py` file where it says:

```python
        self.evse_config = {
            "evse_id_DIN": '+49*123*456*789',
            "evse_id_ISO": 'DE*A23*E45B*78C',
            "protocol": [0, 1], 
            "payment_method": [0],
            "energy_transfer_mode": [0, 1, 2, 3],
            "certificate_installation_support": False,
            "certificate_update_support": False,
        }
```

There you can edit the EVSE ID to your liking (keep the correct format in mind, please see the Whitebeet manual chapter "18.10.1 Set Configuration").

Best regards,
lho

JIN-MIRAE, 2023-09-07:
You said uart communication isn't supported in setting parameter.

Then, could you tell me what means this?
[image]
How can use that way?

Thank you

lho-stx, 2023-09-07:
Hello @JIN-MIRAE,

There is a EVSE ID for DIN and for ISO protocol. The characters of the EVSE ID are transmitted as ascii to the whitebeet, so e.g.

|ascii|hex|
|---|---|
|D|0x44|
|E|0x45|
|*|0x2a|
|A|0x41|
|2|0x32|
|3|0x33|
.... and so on

Because the EVSE ID is a string of arbitrary size there is a byte (0x0F) that tells the size of the incoming string.

The Whitebeet uses a protocol called "framing protocol". It is a sevenstax own protocol, encapsuled in ethernet frames that is used to communicate with the Whitebeet. It can be used over ethernet and over SPI, but when using SPI it needs to be wrapped in a specific SPI protocol. Please see the Whitebeet manual for details about the protocols used:

[https://github.com/Sevenstax/FreeV2G/blob/master/SEVENSTAX_WhiteBeet_UserManual.pdf](https://github.com/Sevenstax/FreeV2G/blob/master/SEVENSTAX_WhiteBeet_UserManual.pdf)

The FreeV2G application demonstrates how to use the framing protocol with the whitebeet. Please have a look into the README.md of this repository in order to get started.

You can also find information how to use the protocol there. In chapter ` 9 WHITE beet – Module Interfaces` you can read about ho to use the framing protocol. There are also examples of how the message workflow needs to be in order to create a succesfull charging session.

Best regards,
lho

JIN-MIRAE, 2023-09-08:
Hi,

I understood that how to modify code thanks to your answer.

Though, I don't know hou to use that 'Example' still.
[image]

Most of all, 'C0' of the 'Example' isn't ASC|| code.
What is that 'Example'? 
Is there any tool for using that 'Example'?

Thank you

lho-stx, 2023-09-08:
Hey @JIN-MIRAE,

If you are using FreeV2G you are using the framing protocol under the hood. If you want to implement it by your own (e.g. for a custom host MCU) then I refer to the Whitebeet manual chapter `9 WHITE beet – Module Interfaces` and chapter `10 Framing Protocol`. It is explained how to use the two interfaces of the Whitebeet (ETH or SPI) are used and how you create a valid framing message in there (e.g. the 0xC0 marks the start and 0xC1 marks the end of a framing protocol message).

The example is just an example to show how the framing protocol payload for this specific message can look lit. If you want to use this specific example, you need to create valid framing protocol message and wrap it into an ethernet packet. Then you can:

1. send it via ethernet to the Whitebeet. You need to select ETH as host controller interface and the destination MAC address is the Whitebeet MAC address.
2. send it via SPI to the Whitebeet. Then you need to wrap it in the SPI communication protocol. Please see Whitebeet manual chapter `9.2 SPI Interface` for more information.

You can also use FreeV2G as a starting point for a custom user application.

Best regards,
lho

JIN-MIRAE, 2023-09-11:
Hello,

According to the readme of Sevenstax GitHub, when setting a configuration, I can use file of format of 'json'.
So, 'ev.json' can be used for setting battery parameter.
However, besides this method, I can set the configuration of battery by using 'battery.py'.

What is the difference of ev.json and battery.py?
Also is there any different rusult when using ev.json or battery.py?

Thank you

lho-stx, 2023-09-11:
Hello @JIN-MIRAE,

FreeV2G will overwrite the default values in `Battery.py` with the values from the `ev.json`. There is no difference in setting the values in `Battery.py` or `ev.json`, but I encourage you to use the `ev.json`, so you can create different configurations.

Best regards,
lho

closed 2023-09-11 by lho-stx

lho-stx, 2023-09-11:
Hey,

You can only set the parameters present in the `ev.json`. The e.g. `EvseMaxVoltage` parameter is set nad used in the `Evse.py`

Unfortunately the current `master` branch does not include the  possibility to use the `.json` file for EVSE yet. You need to set your values in [Application.py](https://github.com/Sevenstax/FreeV2G/blob/a81d3bc181e13fa1c2e5ba0f7773312ae196bba1/Application.py#L58).

There will be a FreeV2G update in the future that enables the config file for EVSE too!

Best regards,
lho