Sevenstax/FreeV2G #229: Question: no doc for module id 0xFF

zhanglongqi, 2023-09-07:
I received a frame but no document for this module id (0xFF)

[image]

[image]

lho-stx, 2023-09-07:
Hey @zhanglongqi,

modId 0xFF is tied to the framing module itself. SubId 0xF3 is "Status Missing End Of Frame". This is send when the end of frame byte in the last message to the Whitebeet is missing. Please check if the last message you send to the Whitebeet is malformed!

I will discuss internally if the information about this module will be added to the Whitebeet manual.

Best regards,
lho

closed 2023-09-07 by lho-stx

zhanglongqi, 2023-09-08:
Thanks, found the bug already.

This information is very important for the application development. Hiding this will only drag our process.

lho-stx, 2023-09-08:
Good to hear you made progress!

I understand your point and will communicate it internally!