Sevenstax/FreeV2G #235: Question: unit for time

zhanglongqi, 2023-09-13:
what is the unit for the time mentioned in the document?

[image]

nststx, 2023-09-14:
Hi @zhanglongqi,

Thanks for submitting your question. The unit for the remaining time for both full and bulk SOC is always seconds. We may need an additional note for that.

I hope I could clarify your question. If you have any more questions, feel free to ask! If you consider this resolved, please close this issue.

Best regards,
Niklas Stoffers

closed 2023-09-15 by zhanglongqi

LeeoMariani, 2023-11-07:
Hi @zhanglongqi, was the time remaining to complete the SOC shown on the packet correct for you?
Thx

zhanglongqi, 2023-11-08:
not always. Sometimes it’s negative value.