Sevenstax/FreeV2G #237: request ID of status message should be 0xFF

zhanglongqi, 2023-09-14:
**Your Setup**
Platform: EVSE
Firmware Version: V02_00_01
Host Controller Interface: SPI

**Describe the bug**
The request ID of status message(`session started`) should be `0xFF` but what I received is `0x00`.

[image]

[image]

nststx, 2023-09-14:
Hi @zhanglongqi,

Thanks for submitting your question. The request id for status messages for the V2G module was 0x00 in the past. As this was pretty inconvenient we fixed it with our newest V02_01_00 firmware release. Now all modules consistently use 0xFF as request id for status messages. Please try updating your EVSE to the newest firmware version and let me know if that resolved your issue.

If you have any more questions, feel free to ask!

Best regards,
Niklas Stoffers

zhanglongqi, 2023-09-15:
Thanks. I will try to update.

closed 2023-09-15 by zhanglongqi