Sevenstax/FreeV2G #239: sdp request

Fatih738, 2023-09-14:
I have received this messages from wb-carrier-board-ei with freev2g. How can I pass the sdp request section. How can I fix this warning. terminal picture is output of pev side. Pev side is my code. Just I want to use freev2g as evse side to simulate

[image]
[image]
thanks

nststx, 2023-09-14:
Hi @Fatih738,

The warning comes from the board not accepting the configuration command. This is most likely due to mismatching FreeV2G and Whitebeet Firmware versions. I see that your EVSE Firmware version is V01_01_07. Do you use the corresponding FreeV2G tag for that version? If not please try with the corresponding FreeV2G version or consider updating your EVSE board. Updating the firmware of the board will also provide you with the newest features and bugfixes.

Please let me know if you're able to fix your issue with the suggested solution. If you have any questions left, feel free to ask!

Best regards,
Niklas Stoffers

Fatih738, 2023-09-14:
git clone https://github.com/Sevenstax/FreeV2G/tree/EVSE_v1.1.7_0
cd FreeV2G-EVSE_v1.1.7_0 
python3 -m venv .venv
source .venv/bin/activate
pip install --pre scapy[basic]
pip install Cython
pip install python-libpcap

and I have tried these two section
sudo .venv/bin/python3 Application.py eth -i eth0 -m C4:93:00:34:A6:AD -r EVSE
python3 Application.py "eth0" -m C4:93:00:34:A6:AD -r EVSE

it didnt work my output is
[image]

Fatih738, 2023-09-15:
How can I run the FreeV2G python code while using wireshark on windows? Won't I have canceled the EVSE side?
(put evse.whitebeet.networkConfigSetPortMirrorState(1) before evse.loop()
When I put it, how the test setup wil be? 
Now, Im using wb-carrier-board ei with raspberry pi that running freev2g. Also Im using red-carrier-board as pev side. And cp,pe pins are connected to each other. Originally I have see EVSE id after EVCC id. 

thanks for your support.

lho-stx, 2023-09-15:
Hey @Fatih738,

we are in the process of creating user guides. This is still a draft, but you can find the guide for creating log files here:

[https://github.com/Sevenstax/FreeV2G/wiki/guide-creating-log-files](https://github.com/Sevenstax/FreeV2G/wiki/guide-creating-log-files)

To get the firmware update tool and the firmware files please contact your distributor. If you use the latest firmware you can always use the latest `master` branch.

Best regards,
lho

Fatih738, 2023-09-15:
[output_2023-09-15_14-20-15.log](https://github.com/Sevenstax/FreeV2G/files/12618770/output_2023-09-15_14-20-15.log)

Here is the log file..

I hope  it helps

thanks

lho-stx, 2023-09-18:
Hey @Fatih738,

like stated in the wiki entry, please create the wireshark log file. Even if you are using SPI please log the network traffic on the Whitebeet ethernet conector. Also please create SPI log files. The UART alone is not sufficient, I only see that the EV starts to request charging and then there is any error. To inspect the error further, please provide the log files.

Wireshark is only listening on the interface it does not interfere with the functionality of the rest of the applications using this interface. Of cause you can not run the FreeV2G PEV on the same interface when connected to the EVSE Whitebeet, please use a seperate ethernet interface for this.

Best regards,
lho

Fatih738, 2023-09-18:
just I have one ethernet interface. Im using rj45 ethernet cable between raspberry pi and wb-carrier-board-ei. I dont use FreeV2G PEV. I have own code that has red-carrier-board-p and stm32. Im using FreeV2G evse code on raspberry pi to simulate charging station.

lho-stx, 2023-09-18:
Hey @Fatih738,

I misunderstood your setup, I am sorry. 

To reliably create log files you need to connect directly to your wb-carrier-board-ei with a seperate interface. You may use a network switch, but I highly recommend to use a separate interface (we had issue with network switches not transmitting whole traffic in the past), so please use a dedicated ethernet interface (e.g. USB to ethernet adapter).

With best regards,
lho

Fatih738, 2023-09-18:
I wil use spi to run FreeV2G. I will record log over wireshark. 

When I run this 
sudo .venv/bin/python3 Application.py spi -i spidev0.0 -m 00:01:01:63:77:33 -r EVSE

I changed spi from spidev0.0 to spidev0.1

 Im getting this output from raspberry pi terminal
usage: Application.py [-h] [-m MAC] [-r ROLE] [-c [CONFIG]] interface
Application.py: error: unrecognized arguments: -i spidev0.1

How can I solve this. So I can get log record

Fatih738, 2023-09-20:
Im getting this error. How can be solved

C:\Users\Desktop>"D:\EVRY\codico\documentation_area\Sevenstax\Sevenstax_WHITEBEET_tools__2023_05_10\Windows\StxFwUpdater_FWUtool\FWUtool.exe" -f "D:\EVRY\codico\documentation_area\Sevenstax\WHITEBEET_ISO15118_EVSE_PNC_V02_01_00\WHITEBEET_ISO15118_EVSE_V02_01_00.fwu" -t C4:93:00:34:A6:AD -i “Ethernet”
Welcome to the SEVENSTAX FWU Tool 2.6

Initiating framing interface
Traceback (most recent call last):
  File "scapy\arch\windows\__init__.py", line 693, in dev_from_name
StopIteration

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "scapy\arch\windows\__init__.py", line 772, in pcapname
  File "scapy\arch\windows\__init__.py", line 696, in dev_from_name
ValueError: Unknown network interface '“Ethernet”'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "scapy\arch\windows\__init__.py", line 701, in dev_from_pcapname
StopIteration

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "FWUTool.py", line 318, in <module>
  File "FramingInterface.py", line 66, in initialize_framing
  File "EthernetAdapter.py", line 128, in start
  File "scapy\arch\pcapdnet.py", line 332, in __init__
  File "scapy\arch\windows\__init__.py", line 799, in open_pcap
  File "scapy\arch\windows\__init__.py", line 774, in pcapname
  File "scapy\arch\windows\__init__.py", line 704, in dev_from_pcapname
ValueError: Unknown pypcap network interface '“Ethernet”'
[18476] Failed to execute script 'FWUTool' due to unhandled exception!

lho-stx, 2023-09-20:
Hey @Fatih738,

The output tells you:

`ValueError: Unknown network interface '“Ethernet”'`
The interface name can be obtained by using the ipconfig command:

```
ipconfig

Windows-IP-Konfiguration
Ethernet-Adapter Ethernet 5:

   Verbindungsspezifisches DNS-Suffix: 
   Verbindungslokale IPv6-Adresse  . : fe80::xxxx:xxxx:xxxx:xxxx%xx
   IPv4-Adresse  . . . . . . . . . . : XXX.XXX.XXX.XXX
   Subnetzmaske  . . . . . . . . . . : 255.255.255.0
   Standardgateway . . . . . . . . . : XXX.XXX.XXX.XXX
```

Here it is "Ethernet 5" as an example.

Please check the interface name and make sure the name is wrapped in parantheses `"`.

Best regards,
lho

Fatih738, 2023-09-21:
Is this a true  charging process? did you see any problem

[image]

lho-stx, 2023-09-21:
Hey @Fatih738,

yes this looks like a successful charging session, although the remaining time is not changing...

Does this solve your issue? If yes please close this issue as solved.

Best regards,
lho

Fatih738, 2023-09-21:
I just changed the soc. 

thanks

Fatih738, 2023-09-28:
I want to ask that. Im not use S2 switch to drop 6V cp pin. cp pin is staying 9V. I asked before and send photo to you. This is a true charging process.  If there is a power  flow request from car, will the station give power to the car according to this picture?
[image]

lho-stx, 2023-09-29:
Hey @Fatih738,

from a ISO 15118 point of view it looks like a successful charging session, but there is more to do a charging session. According to IEC 61851 the EV needs to change the state to state C (+6V) or state D (+3V) when charging is active. The host application needs to check if the EV is in state C/D before applying power.

FreeV2G does not handle the state checks needed according to IEC 61851. As the IEC 61851 is related to functional safety, it is out of scope of FreeV2G.

Best regards,
lho

closed 2023-09-29 by lho-stx