Sevenstax/FreeV2G #248: EVSE does not work with Hyundai KONA

zhanglongqi, 2023-09-28:
**Your Setup**
Platform: EVSE
Firmware Version: V02_00_01
Host Controller Interface: Ethernet
Host: FreeV2G (EVSE) EVSE_v2.1.0_1)
 
 Host controller Interface: ETHERNET!

**Describe the bug**
The same FREEV2G program and same hardware setup, it only works with IONIC 5 but not with KONA.
The Kona I use works well with [SP 100 kw charger](https://carbuyer.com.sg/sp-group-charge-your-ev-singapore/). 
[image]

**Logs**
[kona-iso15118-2023-10-27.zip](https://github.com/Sevenstax/FreeV2G/files/12744934/kona-iso15118-2023-10-27.zip)

lho-stx, 2023-09-29:
Hey @zhanglongqi,

you configured your Whitebeet to support ISO only:

[image]

The Hyundai Kona only offers DIN:

[image]

Try to enable DIN for your Whitebeet too. This should fix your issue!

I want to ask you for a favor:

The next time you are testing the Ionic 5, can you create a wireshark log of the working charging session? We had issue with some Ionic 5 in the past and try to figure out why some are failing the charging session for no good reason and why the most do not.

Best regards,
lho

closed 2023-09-29 by lho-stx

zhanglongqi, 2023-09-30:
I tried  DIN only but it did not work and the error is showed in the screenshot marked as "Kona, DIN only"

zhanglongqi, 2023-09-30:
I have it before. 
It's 7z file actually. github do not allow 7z file uploading.
[ionic-5-no-DIN.7z.txt](https://github.com/Sevenstax/FreeV2G/files/12774075/ionic-5-no-DIN.7z.txt)