Sevenstax/FreeV2G #249: Question: what is the proper way to stop charging from EVSE side?

zhanglongqi, 2023-10-05:
This command `18.10.16 Stop Charging` has to be used to set the stop charging status after it was
requested by the status message `Stop Charging Requested`.
The  status message `Stop Charging Requested` is sent when the EV requested to stop charging.

Does it mean there is no proper way to stop the charging from EVSE side?

[image]

lho-stx, 2023-10-05:
Hey @zhanglongqi,

gracefully stopping a charging session is only initiated by the EV per ISO15118. 

May I ask you what you want to achieve? 

You can try to use the charging schedules to limit the amount of energy/time used for charging.

Best regards,
lho

zhanglongqi, 2023-10-07:
Thanks I will try the schedule.

For [this kind of application](https://www.toutiao.com/article/7286730013465149987/?app=news_article&timestamp=1696580930&use_new_style=1&req_id=20231006162849B68894214D575E7A5F77&group_id=7286730013465149987&share_token=6C1E3A94-83BD-4197-8B3B-19DA439ABF0B&tt_from=weixin&wxshare_count=2&source=m_redirect), it's better to control from the EVSE side.

ColinCGC, 2023-10-19:
I think this is what you're looking for - as per ISO15118 the EVSE can use the notification feature to let EV know it would like to stop charging, then the EV should initiate a graceful stop:

[image]

Works for me.

GovindMore, 2023-10-20:
> I think this is what you're looking for - as per ISO15118 the EVSE can use the notification feature to let EV know it would like to stop charging, then the EV should initiate a graceful stop:
> 
> [image]
> 
> Works for me.

Hi @ColinCGC ,
I am also trying to stop charging from EVSE side, so I sent this command in charging loop after some condition, but it throws warning "Warning: Module did not accept command 27:75, return code: 4". Could you please give some idea where should I put this and when should it be sent to EV?

lho-stx, 2024-01-02:
Hey @GovindMore,

the return code 4 indicates a malformed packet, please double check your packet payload.

Best regards,
lho

closed 2024-02-05 by github-actions[bot]