Sevenstax/FreeV2G #250: Charging Sequence Termination and Compatibility Problems with DIN70121-2:2012 and ISO Standards in EV Charging Systems.

Rahul-P5, 2023-10-09:
**EV Setup:**
Platform: PEV
Firmware version: V01_00_06
Host Controller Interface: Ethernet
Host: PC (Ubuntu)

**EVSE Setup1:**
Platform: EVSE
Firmware version: V02_00_01
Host Controller Interface: Ethernet
Host: PC (Ubuntu)

**EVSE setup2:**
platform: vSECC(Vector)
Firmware Version: V2.6.1

When we set the EV profile to DIN70121-2:2012, the charging sequence gets terminated after the precharge request/response.
However, when we try it with ISO on the Codico EVSE and PEV module, the charging sequence works fine.

For the DIN70121-2:2012 protocol, charging is terminated after the pre-charge with both EVSEs (Codico and vSECC).
However, with vSECC, ISO is not working at all (vSECC has ISO15118-2 and ISO15118-20 configurations).

Logs:
[logfiles](https://github.com/Sevenstax/FreeV2G/files/12845235/logfiles.zip)

ColinCGC, 2023-10-19:
It might not be your problem, but we did run into a similar sounding issue - it turns out one of the differences between DIN and ISO is that DIN has a much shorter timeout for the PowerDeliveryReq response - I think around 1.5s, compared to the 5s for ISO.

This means if the EVSE does not enable its output and send a response quickly enough after precharge then the sequence will terminate.

nststx, 2024-02-07:
Hi @Rahul-P5,

Thanks for submitting your issue. I currently suspect that the EV is failing to serialize it's PowerDeliveryReq. However UART output from the EV would be helpful here. I couldn't find that within the zip file. Also there seems to be something wrong with the formatting of the UART log of the EVSE. Could you please capture them again?

One more thing I noticed is that you're trying to charge with energy transfer mode DC core. This mode is pretty much not used at all and may be the reason why ISO isn't working against your vSECC. You may try to change it to DC extended. Otherwise I will need logs of WB against vSECC too.

Best regards,
Niklas Stoffers

closed 2024-02-29 by github-actions[bot]