Sevenstax/FreeV2G #254: Question: Wrong remaining time on the vehicle

LeeoMariani, 2023-10-31:
Hi there,

We are having problems with the "Remaining time to full SOC" and "Remaining time to bulk SOC" parameters (0x85 - DC Charge Parameters Changed). The value displayed in the vehicle does not represent the real remaining time.

For example, if the vehicle is charging at 63A, the remaining time shown is 12h and 35min, but the real remaining charge should be about 2 hours.

This is the packet received by the car (0x85), showing this example
55 55 00 00 C0 27 85 00 00 20 10 18 FF 06 72 FF 01 13 88 01 01 00 42 10 18 FF 02 A1 FF 00 00 00 01 00 01 12 84 01 01 60 18 00 50 C1

This is the packet sent to the car (0x63)
55 55 00 00 C0 27 63 04 00 14 01 01 7E 00 00 43 00 01 03 E8 00 01 00 01 02 01 00 3C 03 00 E5 C1

We would like to know what parameters we need to send to the CODICO Whiteboard to adjust the remaining time, or what we might be doing wrong?

We have done the same test with other chargers (ABB and WEG) and the remaining time is correct, observing the power and current during the charging process.

Codico FW version: V02_00_01
White Beet - User Manual Revision Revision No: 2.17
Vehicle Renault ZOE (2022/2023)

Thanks!

lho-stx, 2023-11-08:
Hey @LeeoMariani,

I am investigating this issue, please allow some more time!

Best regards,
lho

nikunjp26, 2023-12-14:
Dear @lho-stx 

We also facing same issue so can you please update on this? what is the conclusion of this?

best regards,
nikunj patel

lho-stx, 2024-01-11:
Hey @nikunjp26 and @LeeoMariani,

I checked the issue and I can not reproduce the problem on my side. The remaining time is calculated by the PEV and just send from the Whitebeet to the host. I double checked that this is working, but I am confused why this is only happening when using Whitebeet. Can you please create a wireshark log of such a session and enable port mirror? I want to compare the V2G communication to the framing communication. 

 Also, do you use FreeV2G or a custom host application?

Best regards,
lho

closed 2024-02-02 by github-actions[bot]