Sevenstax/FreeV2G #255: Consecutive session is not working with Mahendra XUV400 Car

nikunjp26, 2023-11-01:
Platform: EVSE
Firmware Version: i.e. V02_01_00
Host Controller Interface: Ethernet
Host: Own Implementation

We are using whitebeet module with **02_01_00** in our charger. We are trying to charge the mahendra xuv400 car but 2nd session was always goes in error and not able to charge the consucutive session with our charger.  we are doing the hard reset the whitebeet module after 1st session so it working fine. 

Here we attached the logs of two session in which we got the error so please help us to know why the consecutive session is not working properly? based on that we can improved our software 

[xuv400.zip](https://github.com/Sevenstax/FreeV2G/files/13227818/xuv400.zip)

let us know if need more details.

nikunjp26, 2023-11-03:
Dear Team,

Any update on this as we are waiting for this as this issue we are facing on public charger?

best regards,
Nikunj Patel

nikunjp26, 2023-11-03:
Dear team,

We oserved the V2G session Logs and some doughts at below points so can you please explain more on that

1) As per packet No. 7223 and 7224 car has already initiated the SECC discovery request so why PLC module does not reply quickly? why it wait for 2nd request ?

2) We found two broadcast Packet No. 7227 and 7228 is arrived after the SECC request. does it proper?

[image]

Let us know if need more details..

Best regards,
Nikunj Patel

lho-stx, 2023-11-06:
Hey @nikunjp26,

we inspected the logs and it seems, that you do not reset the SLAC module after each charging session.  In the next SLAC process the same credentials from the old charging session are used. This leads to the EVSE being in a different network from the PEV network. This behaviour is intended in order to support SLAC pause.  Please stop the SLAC module after each charging session, this should resolve your issues. This will most likely fix #256 too.

Best regards,
lho

nikunjp26, 2023-11-08:
Dear Iho,

Right now we don't have XUV400 car so will update you as quickly as we charge this car again.

Best Regards,
Nikunj Patel

nikunjp26, 2023-11-20:
Dear Iho,

We tested STOP SLAC change on each charging session with XUV400. Here I attached Wireshark Log. Sometimes, CAR was not charging.

[XUV_ConsecutiveSession_StopSlac.zip…]()

Best Regards,
Nikunj Patel

lho-stx, 2023-11-20:
Hey @nikunjp26,

unfortunately the link is not a log file zip but a link to this issue. Please upload your log file!

Best regards,
lho

nikunjp26, 2023-11-21:
Dear Iho,
Find the attached log file.
[Mahindra_Xuv.zip](https://github.com/Sevenstax/FreeV2G/files/13421734/Mahindra_Xuv.zip)

Best regards,
Nikunj Patel

lho-stx, 2023-11-29:
Hey @nikunjp26,

the EVSE maximum current in the ChargeParameterDiscoveryResponse of the failing session is set to 0A (even so minimum current is 5A). It is set to 0A by the user application after the SLAC matching. Maybe this is a problem for the PEV, because the PEV sends a SessionStopReq (that is not allowed at this point in the protocol)? In the working charging sessions the EVSE maximum current is set to an actual value.

Best regards,
lho

nikunjp26, 2023-12-01:
Dear Iho,
We didn't have car now. will update you as testing complete with car.

Best Regards,
Nikunj Patel

closed 2024-01-05 by github-actions[bot]