Sevenstax/FreeV2G #256: Even Number of session is not working with Hundai Kona Car

nikunjp26, 2023-11-01:
Platform: EVSE
Firmware Version: i.e. V02_01_00
Host Controller Interface: Ethernet
Host: Own Implementation

We are using whitebeet module with 02_01_00 in our charger. We are trying to charge the Hundai Kona car but even(2nd/3rd/6th) session was always goes in error and odd number of session is working fine. 

**FYI :** We tested with hard reset plc module on every session so it's working fine with Kona car so there was no any issue with CAR Firmware.

Here we attached the logs of session in which we got the error so please help us to know why the even number of session is not working properly? based on that we can improved our software

[KONA_Consucutive_session_011123.zip](https://github.com/Sevenstax/FreeV2G/files/13227886/KONA_Consucutive_session_011123.zip)

let us know if need more details.

nikunjp26, 2023-11-03:
Dear Team,

Any update on this as we are waiting for this as this issue we are facing on public charger?

best regards,
Nikunj Patel

nikunjp26, 2023-11-08:
Dear Iho,

We modify the logic as per suggestion given in #255 and everytime we are stopping the SLAC. We did again testing with Kona Car but still issue is facing so plesae check the logs and provide us the quick solution.

Please find the logs.

[continous_session_errorLogs_kona.zip](https://github.com/Sevenstax/FreeV2G/files/13295107/continous_session_errorLogs_kona.zip)

Best Regards,
Nikunj Patel

lho-stx, 2023-11-09:
Hey @nikunjp26,

I checked the logs and it seems there are two successful charging sessions in there. Did you unplug and re-plug the PEV or did you leave the PEV plugged in? I am interested in how you recognize the failed charging session, because in the log I do not see a failed SLAC attemped. 

Best regards,
lho

nikunjp26, 2023-11-09:
**Dear Iho,**

With Kona car we are not facing the SLAC matching issue. SLAC matching issue we are facing only with the XUV400 CAR. Here we are facing issue during the Precharge to charging state transaction. We got session stop request from CAR in every EVEN session. Odd session works fine.

**Best Regards,
Nikunj Patel**

nikunjp26, 2023-11-15:
Dear Iho,

any update on this as we are waiting for your support.

Best Regards,
Nikunj Patel

lho-stx, 2023-11-15:
Hey @nikunjp26,

I am sorry, I misunderstood the issue. I inspected the logs again, and I have two findings:

1. In all PreChargeRequests the PEV reports  "DC EV Error Code: Failed - RESS Temperature Inhibit (1)", even in the working sessions
2. In the PowerDeliveryRequests of the failing sessions the PEV reports "Ready To Charge State: False" and "EV Ready: False". This will trigger a "Response Code: Failed - Sequence error (5)" in the EVSE.

Is it possible that the PEV has a defect, or the battery is in its physical limits for operation? If you try the charging session with the PEV sitting for a time, so the battery can cool down do you also see this issue?

Best regards,
lho

nikunjp26, 2023-11-17:
Dear Iho,

The car does not have any issue because alternate session (even number of sessions) works fine. 
If we restart our service every time so it work fine. But in our application we are hard resetting plc on starting of application.

Best regards
Nikunj Patel

nikunjp26, 2023-11-27:
**Dear Iho,**

1) Our application is working fine with another car like xuv400 in this case so i think something happening inside the module. still we are again reviewing the application code and will tell you if found anything.

2) **Apart from this issue**,  We have license version of wireshark but still we didn't see the **framing protocol payload** parcing of frame data in our system(laptop) which we found in your above snapshot as per attached image. do we need to install any extra plugin? let us know if need to follow any extra installation steps.

[image]

**Best Regards,
Nikunj Patel**

lho-stx, 2023-11-27:
Hey @nikunjp26,

1. You are applying real power to the PEV, right? I would like to know if the reported values match the actual voltage applied to the outlet. In the pictures I posted you see, that the PEV charges only if you exceed the requested target voltage (request 328V, reported by EVSE 340V). If EVSE reports the request target voltage to the PEV (second picture, request 327 V, reported 327V) it does not charge. Either there is a mismatch with what the PEV sees on it s connectors and the reported voltage,  the voltage delta is too high for the PEV or the PEV does not follow ISO 11518.
2. This is a custom dissector that is not part of the greenshark package. Let us move this topic to the e-Mail communication.

Best regards,
lho

nikunjp26, 2023-12-01:
Dear Iho,

We checked the our application and found the glitch in "maxevseconnectorcurrent" parameter. after refining that all the session are working fine with the Kona car.

Thanks for your support.
Nikunj Patel

closed 2023-12-01 by nikunjp26