Sevenstax/FreeV2G #258: Question: Any SPI STM32 HAL implementation?

pxidr, 2023-11-06:
Hello,

I'm using the White-Beet EI 1.1 (with FW version 2.1.0.0) and a STM32H750 as a host controller.

I do succeed to perform a full charge session, but sometimes SPI (configured at 625Kbit/s with the right timings, as stated in the datasheet) gets jerky and i got some desynchronizations of the WB.

Do Sevenstax can provide their SPI implementation on the STM32 microcontroller?

Best regards, Damien

lho-stx, 2023-11-08:
Hey @pxidr / Damien,

Do you ask for a STM32 based host application? Unfortunately we do not have a STM32 based port of FreeV2G.

Or are you asking for the SPI implementation in the Whitebeet itself? This is not part of FreeV2G and confidential information that will not be shared.

Please let me know if I misunderstood your question!

Best regards,
lho

pxidr, 2023-11-09:
> Hey @pxidr / Damien,
>
> Do you ask for a STM32 based host application? Unfortunately we do not have a STM32 based port of FreeV2G.
>
> Or are you asking for the SPI implementation in the Whitebeet [...]

Since the WB uses a STM32F7, yes it would be great if the SPI comm part was shared (but only the SPI implementation, nothing else, but I can understand that Sevenstax doesn't want to share this).

Here's is a Saleae SPI logic analyser log with a failed communication (at +33s 511ms)
[Session 0.zip](https://github.com/Sevenstax/FreeV2G/files/13307840/Session.0.zip)

lho-stx, 2024-01-11:
Hey @pxidr,

Please apologize for the long delay.

At +33s 511ms the Whitebeet had a transmission pending. You read this message, but the TX pin keeps high. The host application then continues with the application flow and transmits another 8 byte to the Whitebeet, altough the Whitebeet tells the host application in the size exchange frame that there are 9 byte pending. I think this is the reason for the desynchronization. Please try to either read pending transmissions before sending frames to the Whitebeet or pad your transmission with 0x00 when necessary.

Best regards,
lho

closed 2024-02-03 by github-actions[bot]