Sevenstax/FreeV2G #259: Question:

Artom-Kondrat, 2023-11-08:
Hello

Platform: EVSE
Firmware Version: V01_01_07
Host Controller Interface: SPI

We are trying to charge VW id.4
After "RequestStartCharging" WB starts to send "RequestChargeLoopParameters". First "TargetCurrent" is equale to 0A. Second "TargetCurrent" is equale to 1A. We can't change charging current from 0 to 1A in 0.15sec, because the CHARGER driven by a MODBUS and a result of change returns in 0.24-0.36sec
After that "TargetCurrent" in request becomes "0xFF 0xFF 0xFF" (or -0.1A???) and we can reciew it for a 60 seconds. Aftrer that the WB stops the charge session.

If we make a charge about 1A to 5A  and send results of measure, a session continues as long as we send the current.

Question: How we can interpret the  "TargetCurrent" is "0xFF 0xFF 0xFF" in "RequestChargeLoopParameters"

[issue.txt](https://github.com/Sevenstax/FreeV2G/files/13294694/issue.txt)

lho-stx, 2023-11-08:
Hey @Artom-Kondrat,

the Whitebeet is just reporting the target current that is send by the EV to the host.  Apparently there is something off. To further investigate this issue please create logs accordingly to [the guide in the wiki](https://github.com/Sevenstax/FreeV2G/wiki/guide-creating-log-files). If you do not have the possibility to create SPI logs right now, please send just the wireshark logs.

Besides this, you are using an outdated firmware version. Please consider updating.

Best regards,
lho

Artom-Kondrat, 2023-11-08:
Thank you for your answer. We will take a time to update a FW.

Artom-Kondrat, 2023-11-15:
Problem was solved after update.

closed 2023-11-15 by Artom-Kondrat