Sevenstax/FreeV2G #260: FW Upgrade on EVSE Module

pranavsparalikar-eaton, 2023-11-09:
Hi,

I have a WB EVSE module with FW version V01_01_07 (EVSE Dev. Kit) and I want to upgrade its FW to version V02_01_00. I initially followed the steps given in the White Beet - User Manual v2.16. I have used ‘WHITEBEET_ISO15118_EVSE_V02_01_00.fwu’ available on CODICO site.

Based on the procedure, I have encrypted the FW file. Following is the screenshot of the config script. The default signing key shared along with the binaries, is used for signing the FW.

[image]

The following commands were used for updating the FW to the EVSE side CODICO module:
1.	$./StxFwGen.py -k SigningCert.key -c fwu_config_01.py -o WHITEBEET_ISO15118_EVSE_V02_01_00.fwu
2.	$sudo ./FWUtool.py -f WHITEBEET_ISO15118_EVSE_V02_01_00.fwu -t (Dev. Kit MAC Add.) -i eno1
Below was the response received from the module once the update was completed (pls. note, the file used for update was signed and encrypted only):

[image]

After this, to check whether the FW update of the CODICO module is successful or not, I connected the UART port to my Laptop and tried to read the FW version through the debug messages. Pls. refer the screenshot below:

[image]

It can be seen that, the FW version V01_01_07 is not updated to V02_01_00 and some error is also encountered at the time stamp of 5.612s.

Here, I realized the mistake I had done and then tried uploading the  WHITEbeet module with the original FW (WHITEBEET_ISO15118_EVSE_V02_01_00.fwu), which I had downloaded from the CODICO site. Pls. note, this is the same module on which, I had updated the FW with default signing key. The FW was updated successfully without any error, it can be seen in the debug messages. But the same error at 5.8s is encountered. The debug messages are stopped at this point. Pls. refer the screenshot below for your ref.:

[image]

If I connect an ethernet cable to the Dev. kit., messages shown in screenshot below are seen:

[image]

Request you to have a look and suggest a way ahead.

--
Thanks & Regards,
Pranav

lho-stx, 2023-11-09:
Hey @pranavsparalikar-eaton,

I need to discuss this issue with a colleague. I will come back to you as soon as I have news.

Best regards,
lho

jpo-stx, 2023-11-13:
Hello @pranavsparalikar-eaton ,

you created a firmware update file, which overwrites the QCA firmware with the Whitebeet firmware. Have a look at your fwu_config_01.py at source and destination.

You should take the WHITEBEET_ISO15118_EVSE_V02_01_00.fwu as it is and upload it.

Now you would need to create a fwu file with the QCA firmware.

pranavsparalikar-eaton, 2023-11-30:
Hello @jpo-stx @lho-stx ,

Thanks for your reply. 

I followed the procedure given in the WB user manual and was able to upload the QCA FW successfully. Now the VFS ERROR seen in earlier screenshots is not there, but there is still a continuous message "IF not linked" seen on the terminal. Pls. refer screenshots below in case of SPI as well as Ethernet:
[image]

[image]

Pls. advise on this. Where or what have I missed...?

--
Thanks & Regards,
Pranav

pranavsparalikar-eaton, 2023-12-12:
Hello @jpo-stx @lho-stx,

I have shipped this module to CODICO for reprogramming. So, I am closing this issue.

--
Thanks & Regards,
Pranav

closed 2023-12-12 by pranavsparalikar-eaton