Sevenstax/FreeV2G #261: Session Stopped after some minutes with EKA Electric Bus

nikunjp26, 2023-11-23:
Hello,
We have facing problem of session stopped after around every 4mins of time with EKA Electric Bus.
Here I attached Wireshark log for same.
Please provide some quick solution.
[Session_Stop4Min.zip](https://github.com/Sevenstax/FreeV2G/files/13445667/Session_Stop4Min.zip)

Best Regards,
Nikunj Patel

jpo-stx, 2023-11-23:
Please use the template when raising an issue!

Which firmware version are you running?

nikunjp26, 2023-11-23:
Hello,
We are using following version.

Platform: EVSE
Firmware Version: i.e. V02_01_00
Host Controller Interface: Ethernet
Host: Own Implementation

Best Regards,
Nikunj Patel

lho-stx, 2023-11-27:
Hey @nikunjp26,

we had a look into this log and need some more time for investigation.

Best regards,
lho

nikunjp26, 2023-11-28:
Dear Iho,

We did analysis at our side as per below which may help you. Can you please quick helps to solve this issue as it's urgent?

Whenever a bus is going into charging mode, there are "Current Demand" (initiated by Bus) and "Current Response" (initiated by Charger) packets that are transmitted between Charger and Bus. So, here our observation is that for "Current Demand" packets, it is successfully acknowledged, and for "Current Response" which is sent by the charger on behalf of the "Current Demand" response, acknowledgement is not received. Hence, after a timeout is reached, charging session was stopped.

Also find the attched snapshot of frames difference between eka cars and Our Kona cars.
[EKA_Bus_Packets.zip](https://github.com/Sevenstax/FreeV2G/files/13487499/EKA_Bus_Packets.zip)

Thanks & Best Regards,
Nikunj Patel

lho-stx, 2023-11-28:
Hey @nikunjp26,

the ack is in the next CurrentDemandRequest, so no problem here. 

Do you have log files of the UART of this session? We may have an idea what's going on, but we need to check the UART log.

Best regards,
lho

nikunjp26, 2023-11-30:
Dear lho,
Here I attached UART Log files with EKA Bus. Please support us for a quick solution.

[EKA_Bus_UartLog.zip](https://github.com/Sevenstax/FreeV2G/files/13511966/EKA_Bus_UartLog.zip)

Best regards,
Nikunj Patel

lho-stx, 2023-11-30:
Hey @nikunjp26,

we need some more time to investigate, because reproducing the problem on our side is quite complicated. I hope to give you more details middle of next week!

Thank you for being patient,
lho

nikunjp26, 2023-11-30:
Dear Iho,

Please find the uart and wireshark logs with v2g  logs of error session only .  we also attached the our board logs just for reference.
[eka_bus_issue_30112023.zip](https://github.com/Sevenstax/FreeV2G/files/13516068/eka_bus_issue_30112023.zip)

you can ignore the previous attached logs because it does not contain the v2g logs. Please refer this logs only. right now we are in very critical situation so please help us as soon as possible. i am dropping email to you for some points which need to consider now. let you tell me if need more details

Best Regards,
Nikunj Patel

lho-stx, 2023-12-11:
Closing this issue, because we worked on this with the customer directly.

closed 2023-12-11 by lho-stx