Sevenstax/FreeV2G #265: Question: How to use the White Beet EVSE to control power supply to work as an EV charger?

ChaiyotMix, 2023-12-12:
**Your Question**
How can I use the White Beet EVSE to control power supply to work as an EV charger?

My setup is:
Platform: EVSE
Firmware Version: i.e. V02_01_00
Host Controller Interface: Ethernet
Host: FreeV2G

Platform: PEV
Firmware Version: i.e. V01_00_07
Host Controller Interface: Ethernet
Host: FreeV2G

I want to use White Beet for both EVSE and PEV to control two power supplies to work as an EV charger and as a PEV. The power supplies used are:
1. N8957APV Photovoltaic Array Simulator works as EV charger.
2. ITECH IT6036C Regenerative Bidirectional Programmable DC Power Supply works as PEV.

Now I have experimented with communicating with the power supply using SCPI communication in Python. My controller is a Raspberry pi 4 model B and it can use SCPI communication with both of the as above power supplies. And it is expected that the Raspberry pi will be used to control the white beet and power supply at the same time.

I would like to know if it would be possible for me to use White Beet to control these two power supplies under these conditions. And if so, what should I do? Please give me some advice.

lho-stx, 2023-12-14:
Hey @ChaiyotMix,

I understood you connected the two power supplies to a raspberry Pi and you also connected the Whitebeets to the raspberry Pi. You need to create your own user application. You can use FreeV2G as a starting point for your user application and to see how the V2G communication works in general. You need to adapt or rewrite FreeV2G to support your specific use case, e.g. by setting the voltage and current of your EVSE power supply to the values the EV requests in the callback functions Keep in mind Whitebeet is only for communication about the charging session, the user application is in control of the actual values to transmit. FreeV2G is only a demonstration and is using a EVSE/PEV simulation. You need to replace this simulation with the handling of your actual hardware.

Best regards,
lho

closed 2023-12-14 by lho-stx

ChaiyotMix, 2023-12-18:
Hi @lho-stx 

Thank you for your respond. I think I can understand that I should modify FreeV2G code and adapt for my work. In my understanding that I can add or rewrite FreeV2G with my code right? And I still use White beet EVSE and White beet PEV for communication in charging session work as a EV charger's socket right? So, White beet is available to use but the code will need to be adapted to my hardware. Did I understand correctly?