Sevenstax/FreeV2G #266: EV stops the Charge without any error messages.

Artom-Kondrat, 2023-12-13:
[output.log](https://github.com/Sevenstax/FreeV2G/files/13659406/output.log)
[3.zip](https://github.com/Sevenstax/FreeV2G/files/13659422/3.zip)

Hi!

My setup is:
Platform: EVSE
Firmware Version: i.e. V02_01_00
Host Controller Interface: SPI

We are trying to charge VW Id.4
1.WB-EVSE sends "handleRequestStartCharging" and requests the current (0-1-4-6-8-13-16 Amps)
The current has not yet begun to increase.
2.Аfter one second from "handleRequestStartCharging"  the EV stops sending V2G packets. Despite the fact that there must be a request for 18 and 20 amp. 
3. The current begins to increase. 
WB-EVSE does not sends V2G packets too.  Despite the fact that there must be "Update DC Charging Parameters" packets sended to the "EV"
4.Аfter 1.5 seconds from "handleRequestStartCharging"  the EV switches from state "C" to state "B"
5.WB-EVSE does not close the charging session for another 25 seconds.
6.WB-EVSE stops the charging session without any errors.

It can't be a connection failure. 
We connected WB-EVSE to WB-EV with a CCS cable and applied current.
V2G packets continued to be transmitted without errors.

lho-stx, 2023-12-14:
Hey @Artom-Kondrat,

Do you have also the UART log of the Whitebeet available? In the "Update DC charging parameters" frames you are sending to the Whitebeet the checksum is set to 0x00 for every message. I want to see if there is an error message on the Whitebeet. I think the Whitebeet ignores this messages due to the checksum being wrong.

Best regards,
lho

Artom-Kondrat, 2023-12-18:
Hi!

[20_no_bod.txt](https://github.com/Sevenstax/FreeV2G/files/13707283/20_no_bod.txt)
[20_no_bod2.txt](https://github.com/Sevenstax/FreeV2G/files/13707285/20_no_bod2.txt)
[20_wbo.txt](https://github.com/Sevenstax/FreeV2G/files/13707286/20_wbo.txt)

I am sending you some examples of logs.  They were taken earlier that day. With the same result.
The log corresponding to the first files was not saved.

Artom-Kondrat, 2024-01-12:
Hi!

Can you answer my question as soon as possible?

lho-stx, 2024-01-12:
Hey @Artom-Kondrat ,

I rechecked the logs again. I've found several problems:

The UART logs seem to be ok.

In the pre charging sequence the PEV initially asks for 200V. The Evse reports present voltage of 500V. The PEV sets the target voltage to ~345V. The EVSE still reports 500V. After some time the EVSE reports 247V.

In the charging loop the PEV asks for ~420V. The EVSE reports present voltage of 247V and present current of 0A. The PEV ramps up the target current, the EVSE sets the present voltage to 344V but still 0A present current. It may be that the PEV does an emergency shutdown, therefore there is no V2G communication anymore. In the framing communication the target current is reported correctly from the Whitebeet to the host.

For me it seems this a problem of the host application or with the power delivery hardware. Can you please recheck your implementation?

Best regards and thank you,
lho