Sevenstax/FreeV2G #269: Firmware Update Failed on Ethernet

nikunjp26, 2023-12-14:
latform: EVSE
Firmware Version: i.e. V02_01_00
Host Controller Interface: Ethernet
Host: Own Implementation

Dear Team

We are trying to update the firmware using our application as we must need for the remote update inwhich we are following the chaptor 14 of "Sevenstax_WHITEBEET_Software_usermanual_r2.17.pdf" document. 
We are getting the failure response of the "**set firmware mode**" request so can you please quick help so we can go ahead as we are facing the multiple charger on different location?

please find the wireshark logs which we got during testing. let us know if need more details.
[Fwu_log.zip](https://github.com/Sevenstax/FreeV2G/files/13672004/Fwu_log.zip)

Best Regards,
Nikunj Patel

lho-stx, 2023-12-14:
Hey @nikunjp26,

the beginning of the transmission looks good, but after sending 2 FWU packets, the Whitebeet does not ACK the third packet send. The Whitebeet does not ACK this packets, but you keep sending other packets. Then the Whitebeet requests to resend packet with index 2, this is ignored and more FWU data is send. The Whitebeet requests packet with index 2 multiple times, but this is ignored by your host. When your host finished sending the FWU file, the Whitebeet requests the same packet once again, nothing happens anymore and the FWU module in the Whitebeet times out.

Please wait until the Whitebeet ACK the FWU packets before sending an other one, or retransmit the last packet if the Whitebeet requests it. This should fix your issue.

Best regards,
lho

PS: Sometimes the framing dissector is confused, so take the error messages in the dissector too seriously. Sometimes it tries to parse an ACK message with the message dissector. When in doubt check the packet bytes!

nikunjp26, 2023-12-19:
Dear Iho,
Thanks for your support.

We got success as followed by your suggestion.

Best Regards,
Nikunj Patel