Sevenstax/FreeV2G #273: SLAC Process Falied and C to B State abnoramely changed with latest model of cars

nikunjp26, 2023-12-19:
Platform: EVSE
Firmware Version: i.e. V02_01_00
Host Controller Interface: Ethernet
Host: Own Implementation

Dear Team,

We observed the below issues with the latest car model "TATA Nexon Fearless LR, Electric" and "BYD E6(Rs. 29.15 Lakh)".
Older version of the same cars working fine with our charger.

**1. Module stop to detecting the state B when User plug in the GUN**. In our application code we are always be in one software state inwhich we are waiting fot **state B** or **Authorise request** from the module but didn't received any data. User had tried multiple time but not getting any data. We restart our application inwhich we hard reset the module using it's reset pin and then able to detect the GUN. FYI. Same software works well with older models of same car.
Can you please help to solve this issue? We observed this scenarios in different chargers. We didn't have it's log as those are much far to our location.

**2. Getting the SLAC Process Failed** responce as gun plugedd and init process start. We didn't have log of the session as charger are much far from our location. We observed this scenarios in different chargers. this issue observed with the above latest models only.

[image]

**3. GUN State change from C to B during Cable check request.** We found this issue with most of the tata's car and same gun is working fine with our Hyundai Kona car. Please find the logs of this session. it will be good if you can provide quick solution for that so we can move with new production and installation.

[TataNexon_GunTest.zip](https://github.com/Sevenstax/FreeV2G/files/13714461/TataNexon_GunTest.zip)

let us know if need more details.

Best Regards,
Nikunj Patel

lho-stx, 2024-01-02:
Hey @nikunjp26,

Regarding 1. and 2.:
I apologize, but without log files I can not do anything for you. 

Regarding 3:
There are basically to scenarios in this log file:

1. Scenario: Charging session proceeding, but the host controller sets the target current to 0A, so there is no actual power transfer. This session is closed by the PEV without any error (besides no power transfer has taken place).
2. Scenario: The PEV sets the state to C, before even the SLAC process has finished. Your host is setting the duty cycle to 5% and the PEV sets the state to state C. PEV shall change to state C or D only when either power is drawn when doing AC or basic charging (dutyt cycle between 10% and 96%), or starting the cable check when doing DC charging. Maybe the PEV tries to do basic charging, but this is a DC charger, so this will fail. As the same version of the DC charger works with an older version of the car, it looks like an issue with the PEV software.

Best regards,
lho

nikunjp26, 2024-01-05:
**Dear Iho,**

Q3: Same New Nextion car is charging with the other compititor's charger easily. Do you feel any CP line/signaling issue on charger side? 

best regards,

lho-stx, 2024-01-08:
Hey @nikunjp26,

the session failing is due to the PEV changing into state C before SLAC even starts. I am confident, this is an issues in the PEV and not of the Whitebeet/your charger. If it is an issue with the transmission power or calibration, the SLAC process would fail.

Does the PEV work reliably on chargers from other manufacturers? If the PEV has issues on other chargers too it is clearly the fault of the PEV.

Best regards,
lho

adirschl, 2024-01-09:
Dear Nikunj,

I sent you an email directly with more details.

But I want to clarify here as well, that there should be no seriel resistance implemented on CP, as this would attenuate the PLC signal and as well damage the CP state detection, as it would influence voltage on CP.

BR,
Andreas Dirschl
CODICO

closed 2024-01-09 by lho-stx