Sevenstax/FreeV2G #277: Charging Issue: EV Target Current Fluctuate a lot in the first 60 seconds

nigaru05, 2024-01-11:
**Platform**: WB-CARRIER-BOARD-EI as EVSE
**Firmware** Version: V02_00_01 PnC
**Host** Controller Interface: SPI
**Host**: FreeV2G (EVSE_v2.0.1_4)

I am working with a WB-CARRIER-BOARD-EI, the FreeV2G implementation is running on a Compute Module 3, which is connected to the WB by an SPI interface. The Compute Module control a 40kW Power Module by an CAN interface. Car is a VW eGolf.

**Describe the bug**
I have been able to upload, but I am concerned about the initial behavior of the upload. In the first seconds of charging (45 to 50) the "EV target current" has fluctuations or drop peaks of a lot of current.
I attach the screenshot of wireshark (port mirror) and a graph where you can see:
Vset = EV target voltage
Iset = EV target current
Vout = output voltage of the power module.
Iout = output current of the power module.
(The Iset and Iout label was wrong, the units say V instead of A, but the values are correct)
As can be seen in the graph and in the log, the EV asks for a value close to 120A and then drops to 105A, rises again to 120A and then falls to 95A, rises again to 120A and then falls to 82A, continues at this cycle until it drops with the peak to 2A (approximately 45 seconds) and then rises to 120A for the rest of the charge, decreasing slowly as the battery charges. In this case we have stopped the loading almost after 60 seconds to have logs that are easy to manage.

**Questions**:
I have done several loads and they always have the same behavior.
My question is if this is normal? Does it depend on the car? or is it something misconfigured?

**Logs and graph:**
[40kW_eGolf.txt](https://github.com/Sevenstax/FreeV2G/files/13903880/40kW_eGolf.txt)
[40kW_eGolf_pcap.zip](https://github.com/Sevenstax/FreeV2G/files/13903896/40kW_eGolf_pcap.zip)
[image]
[curve_V_I_png.zip](https://github.com/Sevenstax/FreeV2G/files/13903906/curve_V_I_png.zip)

I hope you can guide me to continue with my development.

Best regards.

lho-stx, 2024-01-11:
Hey @nigaru05,

I checked your log files, unfortunately the pcap does not contain any V2G or framing messages (besides SLAC and two SeccDP requests). Can you double check the connection? I see SSH traffic, maybe you just sniffed the wrong interface?

If you upload a log containing the V2G and framing communication I can check if there is a difference in what the PEV asks for and what the Whitebeet reports.

Best regards,
lho

nigaru05, 2024-01-11:
Thank you very much for your quick reply.
I have the following line before the loop:
            evse.setSchedule(schedule)
            evse.whitebeet.networkConfigSetPortMirrorState(1)   # Port Mirroring
            # Start the EVSE loop
            evse.loop()

I don't know what's wrong.
The ssh messages are the connection between my notebook and the Compute module, which controls the withebeet via SPI.

I will charge the car again and capture the communication again.

lho-stx, 2024-01-11:
Hey,

ah I misread your setup, I am sorry. When using SPI you still need to connect to the ethernet port of the whitebeet in order to get the V2G messages. So just rerun your charging session but this time also connect to the ethernet port (maybe via ethernet dongle and your development PC) and create a wireshark log. You also can refer to [the wiki](https://github.com/Sevenstax/FreeV2G/wiki/guide-creating-log-files).

I am looking forward to the log files!

Best regards,
lho

nigaru05, 2024-01-12:
I haven't been able to capture it yet. On Monday or Tuesday I will do it and upload the logs.

Best regards

nigaru05, 2024-01-15:
Hey,

I have been able to take the captures, I attach the captures.
I am looking forward to your response.
[image]
[eGolf_iso_picos_debug.txt](https://github.com/Sevenstax/FreeV2G/files/13940785/eGolf_iso_picos_debug.txt)
[eGolf_iso_picos.txt](https://github.com/Sevenstax/FreeV2G/files/13940786/eGolf_iso_picos.txt)
[eGolf_iso_picos.zip](https://github.com/Sevenstax/FreeV2G/files/13940808/eGolf_iso_picos.zip)

Best regards.

lho-stx, 2024-01-16:
Hey @nigaru05,

thank you for the log files!

In the V2G logs the PEV ramps up the target current as expected. Although the PEV asks for 360V target voltage, the EVSE sets ~320V present voltage. The PEV asks for ~105A and your charger delivers ~105A.

[image]

(Please ignore the wrong EVSE present current reading in the legend, the plot it is indeed correct I double checked with the messages).

What else can be seen in the log files is, that the EVSE present voltage fluctuates and the PEV asks for more current in response. I assume the PEV tries keep the power transfer as high as possible and therefore asks for more current when the voltage drops. 

What I don't understand why FreeV2G sees the drop in target currents sometimes, because they are not in the V2G communication, also it is not seen in the EVSE present current. Did you make big changes to the FreeV2G code? I also will double check for conversion problems in FreeV2G.

To really sort out the problem I need a log of the SPI communication. I kindly ask you to create a log of the SPI communication, please have a look [at this wiki page](https://github.com/Sevenstax/FreeV2G/wiki/guide-creating-log-files#spi) how to create a SPI log.

Thank you and best regards,
lho

nigaru05, 2024-01-16:
**Thank you for your quick response.**
I am going to try to comment on each paragraph that you have sent me, since I do not fully understand your conclusions or I have not correctly explained the problem I have.
For this reason, I first return to the context:

### **The graph has:**
Vset = EV target voltage (FreeV2G API)
Iset = EV target current (FreeV2G API) - orange
Vout = output voltage of the power module. (Vout is reported to the FreeV2G API as PresentVoltage for reporting to the PEV).
Iout = output current of the power module. (Iout is reported to the FreeV2G API as PresentCurrent for reporting to the PEV). -yellow

[image]

My concern is the behavior of the Iset (EV target current FreeV2G API), you can see that during the first 45 or 50 seconds the Iset that I get from the 0x85 messages (self._handleDCChargeParametersChanged(data)) varies its value a lot. In the graph you can see 11 descending peaks.

### **Questions about your response:**
**_"In the V2G logs the PEV ramps up the target current as expected. Although the PEV asks for 360V target voltage, the EVSE sets ~320V present voltage. The PEV asks for ~105A and your charger delivers ~105A."_**
Please correct me if I'm wrong. The PEV asks for an EV target voltage of 360V, I obtain the 360V value with the API (EV target voltage), I send it to the Power Module, but since the PEV battery is at 30% with 320V, the voltages are equal, when going charging, the output voltage of the Power Module and the PEV battery voltage rise. Is this correct? In my experience, this happens with any lithium battery charger.

**_"What else can be seen in the log files is, that the EVSE present voltage fluctuates and the PEV asks for more current in response. I assume the PEV tries to keep the power transfer as high as possible and therefore asks for more current when the voltage drops."_**
I can't see the voltage fluctuations, the big fluctuations I see are those of Iset = EV target current (FreeV2G API) and due to these, the output configuration fluctuations of the Iout of the Power Module.
We also have a log of the status of the Power Module and these fluctuations in the Iout are real.

**_"What I don't understand why FreeV2G sees the drop in target currents sometimes, because they are not in the V2G communication, also it is not seen in the EVSE present current. Did you make big changes to the FreeV2G code? I also will double check for conversion problems in FreeV2G."_**
You do not see fluctuations in the plc communication (wireshark), I cannot decode the messages since I do not have the dissertor. But, what you tell me, I don't understand, since I use FreeV2G API and in the log you can see the fluctuation of EV target current.
The modification I have made to FreeV2G (discussed with you in another issue) is to obtain:
'present_voltage': int(self.charger.getEvsePresentVoltage()),
'present_current': int(self.charger.getEvsePresentCurrent()),
In the Evse.py module, lines 141,142 and 362,363.
I have replaced the getEvsePresentVoltage() and getEvsePresentCurrent() methods with 2 methods of my own that obtain the actual current voltage and current values of the Power Module. In this way I inform the PEV of the real value that is being delivered.

**_"To really sort out the problem I need a log of the SPI communication. I kindly ask you to create a log of the SPI communication, please have a look at this wiki page how to create a SPI log."_**
I'm going to try to make this log, I don't know if I can do it.

It is important to mention that if I continue with the load the PEV reaches 100% without giving an error, but those 11 initial peaks do not seem correct to me.

Please tell me if there is something that I am not explaining correctly.
I look forward to your response

Best regards.

lho-stx, 2024-01-16:
Hey @nigaru05,

please apologize if my explanation was not detailed enough! I will try to answer all your questions. I added numbers to your questions, so I can answer them better.

**Regarding 1:**
You are completly right. In the pre-charge phase the voltage of the EVSE shall be raised to the battery voltage, so the PEV can close it's connectors. When charging starts the PEV raises it's target voltage.

This can be seen in the graph I added to my comment. The graph shows the values send via V2G communication between PEV and EVSE over time. The data is from the wireshark log file you send to me.

In the beginning the PEV has it's target voltage set to 320V (labeled as _EV Target Voltage_  in the graph) and the EVSE slowly rises the present voltage to 320V (labeled as _EVSE Present Voltage_  in the graph) . At around 20s the PEV starts actual charging and rises it's target voltage to 360V (this is also seen in the FreeV2G logs, the PEV target current is set to 360V). The EVSE also slowly rises it's present voltage, but it is fluctuating between 320V and 330V. This is not the correct behavior in this situation, the EVSE shall rise it's present voltage to the PEV target voltage.

**Regarding 2:**
Maybe we need to agree  on the term "fluctuating". I mean the "ripple" in the _EVSE Present voltage_ or the _Vout_ of your graph.

**Regarding 3 + 4:**
I need to correct myself: I actually see the drop in current in the V2G communication, but only in the messages send from the EVSE to the PEV. This problem I want to sort out, therefore I need the SPI logs. If it is not possible for you to create such a log file we may workaround it by printing out the received bytes, but I need time for checking where add some code to print the data.

The changes you made are ok for me, I see no problem here.

I agree, that these initial 11 peaks are not normal and we should find the reason for this behavior.

Thank you and best regards,
lho

nigaru05, 2024-01-16:
Hey @lho-stx 

Thank you very much again for your quick response.

**Regarding 1:**
I will keep in mind the voltage variation from 320V to 330V to solve it later.

**Regarding 2:**
Perfect, it's clear to me. I was instead referring to the Iset = EV target current, obtained from the FreeV2G API.
I hope you assign a label to the Iset problem, since this is the one I need to solve at this moment.

**Regarding 3 + 4:**
In the eGolf_iso_picos.txt file you can see the arrival of each 0x85 message by setting the EV target current in the API, then, being inside the charging loop we send the getPresentCurrent to the PEV, which obtains the output current of the Power Module established by the previous EV target current command.
Due to my difficulty in obtaining the SPI log, I would like to be able to give you more information with current tools.
The EV target current value of the log is the result of line 346 of Evse.py. I have highlighted a sharp drop in EV target current from 105A to 88A, like this there are 10 more drops that each time reach a lower level as you can see in the graph (approximate values 105 to 98, 105 to 88, 105 to 82 , 105 to 72, 105 to 62, 105 to 52, 105 to 45, 105 to 30, 105 to 22, 105 to 6 and 105 to 2) and then it is approximately 105A.
If you want, I can add more information to the log, tell me in which line and with what tags you want it to be printed in the txt.
I am attaching a pdf with the highlighted log.
[EvTargetVoltage.pdf](https://github.com/Sevenstax/FreeV2G/files/13952238/EvTargetVoltage.pdf)
My biggest concern is that in the pcap you do not see any messages from the PEV to lower the EV target current and I do see them from the FreeV2G API.

I look forward to your response

Best regards.

lho-stx, 2024-01-19:
Hey @nigaru05,

I created a patch file for enabling printing the received SPI bytes. Please apply the patch and create another log of the session and send it to me. I could not test the patch in and out, but it should work.

[enable_spi_packet_dump.txt](https://github.com/Sevenstax/FreeV2G/files/13986477/enable_spi_packet_dump.txt)

To apply the patch file go to your local copy of FreeV2G, place the patch file in it and run:

`git apply enable_spi_packet_dump.txt`

Thank you and best regards,
lho

nigaru05, 2024-01-19:
Hey @lho-stx,

Thanks for sending me the modification.
I attach 3 files:

- **eGolf_iso_debug_spi.txt** _(file has the complete log)_
- **eGolf_iso_debug_only_spi.txt** _(file with only the output of the modification that you have sent me)_
- **eGolf_iso_debug_spi.pcapng** _(wireshark capture)_

Let me know if what I am sending you is correct.

Thank you and best regards,
Ruben.

[eGolf_iso_debug_spi.txt](https://github.com/Sevenstax/FreeV2G/files/13991021/eGolf_iso_debug_spi.txt)
[eGolf_iso_debug_only_spi.txt](https://github.com/Sevenstax/FreeV2G/files/13991022/eGolf_iso_debug_only_spi.txt)
[eGolf_iso_debug_spi.zip](https://github.com/Sevenstax/FreeV2G/files/13991053/eGolf_iso_debug_spi.zip)

lho-stx, 2024-01-25:
Hey @nigaru05,

I inspected the log files and I did not see the drop in current in the framing messages between Whitebeet and host. But I found something different that may be the problem:

The Whitebeet sends the notifications with the reqID 0x00. This is a bug in the old firmware of the Whitebeet. Please update to the latest firmware (you are using v2.0.1, the latest is 2.1.0) and see if the issue still persists.

Thank you,
lho

nigaru05, 2024-01-29:
Hey @lho-stx,

Sorry for the delay.
I updated the FW to version WHITEBEET_ISO15118_EVSE_V02_01_00.fwu. But it doesn't work with the following configuration:
```
        self.evse_config = {
            "evse_id_DIN": '+49*123*456*789',
            "evse_id_ISO": 'DE*A23*E45B*78C',
            "protocol": [0, 1], 
            "payment_method": [0],
            "energy_transfer_mode": [0, 1, 2, 3],
            "certificate_installation_support": False,
            "certificate_update_support": False,
        }
```
The only way it works is in DIN, with the following configuration:
```
        self.evse_config = {
            "evse_id_DIN": '+49*123*456*789',
            "evse_id_ISO": 'DE*A23*E45B*78C',
            "protocol": [0], 
            "payment_method": [0],
            "energy_transfer_mode": [1],
            "certificate_installation_support": False,
            "certificate_update_support": False,
        }
```
I could not load the eGolf with the ISO protocol, **however, with the version WHITEBEET_ISO15118_EVSE_V02_00_01.fwu it does load in the ISO protocol.**

In order to continue solving the problem that caused this ISSUE and to be able to take the captures, I configured the APP in DIN. I attach the captures made. The file captured from the console contains the entire log, from which I generated 2 other files separating the output of the original Sevenstax APP and the output added with the modification of the SPI log. The txt files are organized as follows:

- **eGolf_2.1_din_picos_debug.txt** _contains all the console output (APP print + SPI)._
- **eGolf_2.1_din_picos_debug_only_spi.txt** _contains only the SPI output._
- **eGolf_2.1_din_picos_debug_only_app.txt** _contains only the APP output._
[eGolf_2.1_din_picos_debug.txt](https://github.com/Sevenstax/FreeV2G/files/14084077/eGolf_2.1_din_picos_debug.txt)
[eGolf_2.1_din_picos_debug_only_spi.txt](https://github.com/Sevenstax/FreeV2G/files/14084078/eGolf_2.1_din_picos_debug_only_spi.txt)
[eGolf_2.1_din_picos_debug_only_app.txt](https://github.com/Sevenstax/FreeV2G/files/14084082/eGolf_2.1_din_picos_debug_only_app.txt)

**I also attach the pcappng and the graph obtained.**
[eGolf_v2.1_din_picos.zip](https://github.com/Sevenstax/FreeV2G/files/14084088/eGolf_v2.1_din_picos.zip)

[image]

If you like, then let me know and we will see the problem why with the WHITEBEET_ISO15118_EVSE_V02_01_00.fwu version the eGolf no longer loads with ISO15118, I receive the Start V2G message, wait 30 seconds and then EVSE loop finished. We talk about it in this ISSUE or do I open a new one?

I look forward to your response

Best regards.

lho-stx, 2024-01-30:
Hey @nigaru05,

regarding the new issue: not all energy transfer modes are supported in each protocol. Please try to set up a ISO only config. If the issues still persists, please open a new issue!

Regarding the other issue. I still do not see the drop in current in the V2G logs nor in the `DC Charge Parameters Changed` notification (mod ID: 0x27, sub ID: 0x85) send from Whitebeet to host. What I do see is that the current drops in the `Update DC Charging Parameters` messages (mod ID: 0x27, sub ID: 0x85) send from Host to the Whitebeet. Please double check how you create the plot of `Iset` in your image above. Also please double check your modified FreeV2G version to use the `DC Charge Parameters Changed` notification for setting the target current of your power electronics.

Best regards,
lho

nigaru05, 2024-01-30:
Hey @lho-stx,

Thank you very much for your quick reply.

Regarding the new problem, I have tried it only in ISO and it has not worked. I will open a new case, I don't know if right now or if I will wait to resolve the current case. Thank you.

Regarding our original case, **I will double-check everything you asked of me**. But, you can verify that the console output of the Sevenstax APP (eGolf_2.1_din_picos_debug_only_app.txt) shows each 0x85 message with the values requested by the vehicle.
Below you can see a summary of the eGolf_2.1_din_picos_debug_only_app.txt file:

> **_linea 510_**
> 2024-01-24 11:43:27.617014  -  El id de mensaje recibido es **85**
> "DC Charge Parameters Changed" received
> EV maximum current: 125.0A
> EV maximum voltage: 360.0V
> EV ready: True
> Error code: 0
> SOC: 66%
> EV target voltage: 360.0V
> EV target current: 88.7A
> Charging complete: 0
> Remaining time to full SOC: 1200s
> 2024-01-24 11:43:27.831930  -  El id de mensaje recibido es **85**
> "DC Charge Parameters Changed" received
> EV maximum current: 125.0A
> EV maximum voltage: 360.0V
> EV ready: True
> Error code: 0
> SOC: 66%
> EV target voltage: 360.0V
> EV target current: 83.9A
> Charging complete: 0
> Remaining time to full SOC: 1200s
> 2024-01-24 11:43:28.049885  -  El id de mensaje recibido es **85**
> "DC Charge Parameters Changed" received
> EV maximum current: 125.0A
> EV maximum voltage: 360.0V
> EV ready: True
> Error code: 0
> SOC: 66%
> EV target voltage: 360.0V
> EV target current: 119.5A
> Charging complete: 0
> Remaining time to full SOC: 1200s
> 2024-01-24 11:43:28.267208  -  El id de mensaje recibido es **85**
> "DC Charge Parameters Changed" received
> EV maximum current: 125.0A
> EV maximum voltage: 360.0V
> EV ready: True
> Error code: 0
> SOC: 66%
> EV target voltage: 360.0V
> EV target current: 119.60000000000001A
> Charging complete: 0
> Remaining time to full SOC: 1200s

**Other**

> **_linea 697:_**
> 2024-01-24 11:43:31.274375  -  El id de mensaje recibido es **85**
> "DC Charge Parameters Changed" received
> EV maximum current: 125.0A
> EV maximum voltage: 360.0V
> EV ready: True
> Error code: 0
> SOC: 66%
> EV target voltage: 360.0V
> EV target current: 119.2A
> Charging complete: 0
> Remaining time to full SOC: 1200s
> 2024-01-24 11:43:31.491208  -  El id de mensaje recibido es **85**
> "DC Charge Parameters Changed" received
> EV maximum current: 125.0A
> EV maximum voltage: 360.0V
> EV ready: True
> Error code: 0
> SOC: 66%
> EV target voltage: 360.0V
> EV target current: 67.10000000000001A
> Charging complete: 0
> Remaining time to full SOC: 1200s
> 2024-01-24 11:43:31.708182  -  El id de mensaje recibido es **85**
> "DC Charge Parameters Changed" received
> EV maximum current: 125.0A
> EV maximum voltage: 360.0V
> EV ready: True
> Error code: 0
> SOC: 66%
> EV target voltage: 360.0V
> EV target current: 118.9A
> Charging complete: 0
> Remaining time to full SOC: 1200s
> 2024-01-24 11:43:31.925443  -  El id de mensaje recibido es **85**
> "DC Charge Parameters Changed" received
> EV maximum current: 125.0A
> EV maximum voltage: 360.0V
> EV ready: True
> Error code: 0
> SOC: 66%
> EV target voltage: 360.0V
> EV target current: 119.80000000000001A
> Charging complete: 0
> Remaining time to full SOC: 1200s
> 2024-01-24 11:43:32.143589  -  El id de mensaje recibido es **85**
> "DC Charge Parameters Changed" received
> EV maximum current: 125.0A
> EV maximum voltage: 360.0V
> EV ready: True
> Error code: 0
> SOC: 66%
> EV target voltage: 360.0V
> EV target current: 119.7A
> Charging complete: 0
> Remaining time to full SOC: 1200s
> 

These are just 2 samples of the 10 or 12.
Regarding this part of the code I have not made any changes.

The code is 0x85 that I am using to obtain this output and is the original:
**Evse.py**

> elif id == 0x85:
>                  self._handleDCChargeParametersChanged(data)

Within this method, the output is made through the console. Original.

I look forward to your response
Best regards.

lho-stx, 2024-01-30:
Hey @nigaru05,

I need to apologize, I double check also and it seems there is an issue with my tool. I indeed see the drop in target current in the raw SPI bytes.

So we have now two options where the problem is:
1. Whitebeet firmware. We do not see this behavior with other setups. Did you try with other cars too? Is this only occurring on this specific model?
2. SPI transmission. Maybe there are glitches on the SPI bus. What is the SPI clock frequency you are using, the default 12 MHz? You can try a lower value by changing `self.spi.max_speed_hz = 12000000` in `SpiAdapter.py`

I will also continue my investigation, but answering my questions will help alot!

Thank you,
lho

nigaru05, 2024-02-05:
Hey @lho-stx,

First of all, I want to apologize for the delay in responding. I did not have the eGolf to test.

**Regarding the 2 possible options:**

1. I have tried the eGolf, a Tesla and an MG4. The only thing that doesn't happen to me is the MG.
2. The SPI CLK frequency is 12Mhz. But I don't see any error messages with SPI.

**To put it in context:**
After your response on my ISSUE Charging problem with battery level less than 76%
#182 (https://github.com/Sevenstax/FreeV2G/issues/182), where you told me:
_"Please keep in mind that the FreeV2G application is a demonstration. If you want to use FreeV2G for actual charging you need to make changes to the code. FreeV2G "simulates" a charging controller, therefore there may differences in the values calculated by FreeV2G and current physical values."_

At that time add these 2 methods that ask the POWER MODULE for the real voltage and current at its output. Actually, for ISSUE #182, I first added the getPresentVoltage and when reporting the real value to the WHITEBEET, the vehicles charged with any percentage of battery. Then bring in the getPresentVoltage. The 2 methods consult the real output value through a second SPI BUS. I am attaching the modified Evse.py that I have called Evse_gets.py
[Evse_gets.txt](https://github.com/Sevenstax/FreeV2G/files/14165966/Evse_gets.txt)

But when the current problem appeared, I started testing regarding my modification of getPresentVoltage and getPresentCurrent. Inside the def _handleDCChargeParametersChanged(self, data) method, I did the following:
self.presentCurrent = self.charger.ev_target_current
And I didn't use getPresentCurrent() anymore.
I am attaching the Evse_getVoltage.py
This way I no longer see the oscillations in the EV target current.
[Evse_getVoltage.txt](https://github.com/Sevenstax/FreeV2G/files/14165968/Evse_getVoltage.txt)

I don't understand what's wrong. While it is charging cyclically it performs self.whitebeet.v2gEvseUpdateDcChargingParameters(charging_parameters), I thought that before this function it was important to know the real current delivered to the vehicle, but this does not work well, did I introduce some delay?
I also tried adding the 2 gets before:
         
```
        self.presentVoltage = self.getPresentVoltage()
        self.presentCurrent = self.getPresentCurrent()
        
        charging_parameters = {
            'isolation_level': 0,
            'present_voltage': int(self.presentVoltage),
            'present_current': int(self.presentCurrent),
            'max_voltage': int(self.charger.getEvseMaxVoltage()),
            'max_current': int(self.charger.getEvseMaxCurrent()),
            'max_power': int(self.charger.getEvseMaxPower()),
            'status': 0,
        }

        try:
            self.whitebeet.v2gEvseUpdateDcChargingParameters(charging_parameters)
        except Warning as e:
            print("Warning: {}".format(e))
        except ConnectionError as e:
            print("ConnectionError: {}".format(e))
```

It didn't work either, in fact, it seemed to me that it made the load worse, the current oscillations of msg 0x85 lasted more than 45 seconds and then it stabilized.

I will do more tests but I am a little disoriented, since theoretically it was necessary to update the WHITEBEET with the real values. I'll be doing something wrong.

I hope you can help me.

Thank you so much
Best regards.

lho-stx, 2024-02-07:
Hey @nigaru05,

thank you for the message. It is a lot of information I need to sort and review the logs again. Please allow some more time so I can find a solution for your problem!

Best regards,
lho

PS: If you are charging other/new vehicles and it is not a great hassle to create logs it would be really great if you share these logs with us.

lho-stx, 2024-02-13:
Hey @nigaru05,

It seems there is a race condition in the _FramingInterface.py_. Therefore some notifications are not received by the application (_Evse.py_) in the correct order. The drop in target current is due to this race condition, the packets are received in the wrong order and it looks like the target current drops. I already created a patch. I attach it here, but I did not test it by myself yet, so **please be aware that it might not work**:

[enable_spi_packet_dump_modify_queue_handling.txt](https://github.com/Sevenstax/FreeV2G/files/14268727/enable_spi_packet_dump_modify_queue_handling.txt)

Best regards,
lho

P.S.: I recommend you to modify _Charger.py_ in order to interface with you power electronics. In this way you don't need to change anything in Evse.py.

lho-stx, 2024-02-22:
Hey @nigaru05,

we need some more time to fix the issue, but I am confident that this is the cause for your problems.

Best regards,
lho

nigaru05, 2024-02-22:
Hey @lho-stx ,

Thank you very much for your quick response.

I understand clearly. I await your response.

Best regards.

nigaru05, 2024-04-23:
Hey @lho-stx ,

I wanted to know if there is any progress or update.

Thanks in advanced.
Best regards.