Sevenstax/FreeV2G #278: Question:Do not work sequence (PLC Datatranfer mode 4 (AC 1Phase Charger))

ByunJongHyun, 2024-01-16:
# Request for answer about WHITE best Carrier boad EV ISO15118 (EV)

Dear Codico PLC Members

hello.
I am a developer (jong hyun) who sent a question related to PLC some time ago.
I will send you an email with the answers to the last reply and any additional questions.

---
**First, this is a reply to the last email.**
#### Q. Can I please ask you the question what ISO 15118-3 solution - means - what PLC solution - you intend to use in your new AC charger that applies ISO-15118 2 and ISO-15118 20 ?

#### A. Yes, that's right. Previously, in Korea, ISO-15118 was used as a protocol for DC chargers, and IEC-61851 was used as a protocol for AC chargers. Starting in 2024, AC chargers are also scheduled to release models that apply ISO-15118 using PLC Modem. Our company is developing a PLC modem for AC chargers to be used at this time.
---

And let me ask you a question.
I looked for a similar case on the Internet (https://github.com/Sevenstax/FreeV2G/issues/116).

As you can see here, I set the energy transfer mode to 4 (AC Single Phase) and want to read the SOC information, but an error occurs.

I'm in a similar situation.

In my settings, the energy transfer mode is set to 4 (AC Single Phase).
And with the voltage of CP and CE being 9V, I considered giving 5% PWM as the starting trigger and configured it to enable such a setting.

[image]

Looking at the order of the image, after ChargeParameterDiscoveryReq/Res is made, when the PowerDeliveryReq signal is requested from the vehicle side -> charger side, PowerDeliveryRes is sent in the order of sending from charger to vehicle.

If you look at the log attached at the bottom,
On the charger side, it times out while waiting for PowerDeliveryReq after ChargeParameterDiscoveryRes and turns off.

On the vehicle side, Ready for charging (0xC7)
Reference: (White Beet - User Manual p122)
After waiting for 40 seconds, it says that the session will proceed to Stop.
Among them, there is a corresponding WARN message.
[ 6074.027] [WARN ] APL-ETH: # Unhandled Ethernet notifycode 365

I don't know if my understanding is correct, but
APL-ETH: # Unhandled Ethernet notifycode 365 This part appears to be the part where there is no response to ChargingParameterDiscoveryRes, and the session seems to end 40 seconds after that.

This is what I have figured out and my guesses so far, but which parts should I modify before proceeding? In the last email, I received a response that the ISO15118-AC charger mode was also working normally, but since there are posts on the Internet saying that there is an error, I am curious about how to support it, if it is in fact supported correctly. If you just do it, it will proceed with DC Charger.

### Now let me ask you a question

#### 1. When analyzing the log, if it is a problem on the charger side, it would be nice if you could tell me in detail and the direction of the larger picture as to which part needs to be modified in order to receive PowerDeleveryReq.

#### 2. If the problem is on the vehicle side, please let me know if there is a solution.

Thanks in advance for your reply!

Best,

<< log below >>

```
Charger Log (applied EVerest Program)

2024-01-15 05:56:19.929792 [INFO] manager          ::   ________      __                _
2024-01-15 05:56:19.930396 [INFO] manager          ::  |  ____\ \    / /               | |
2024-01-15 05:56:19.930457 [INFO] manager          ::  | |__   \ \  / /__ _ __ ___  ___| |_
2024-01-15 05:56:19.930505 [INFO] manager          ::  |  __|   \ \/ / _ \ '__/ _ \/ __| __|
2024-01-15 05:56:19.930550 [INFO] manager          ::  | |____   \  /  __/ | |  __/\__ \ |_
2024-01-15 05:56:19.930592 [INFO] manager          ::  |______|   \/ \___|_|  \___||___/\__|
2024-01-15 05:56:19.930633 [INFO] manager          ::
2024-01-15 05:56:19.930669 [INFO] manager          :: Using MQTT broker localhost:1883
2024-01-15 05:56:19.981393 [INFO] manager          :: Loading config file at: /home/koolsign/checkout/everest-workspace/everest-core/config/config-sil.yaml
2024-01-15 05:56:20.893562 [INFO] manager          :: - Types loaded in [55ms]
2024-01-15 05:56:20.893697 [INFO] manager          :: - Types validated [826ms]
2024-01-15 05:56:20.895747 [INFO] manager          :: - Errors loaded in [0ms]
[... 999 lines cut ...]
[ 6066.019] [NOTICE] APL-MAIN:
[ 6066.019] [NOTICE] APL-MAIN: =============================
[ 6066.020] [NOTICE] APL-MAIN: NEW LAN CONFIGURATION (IPv6):
[ 6066.020] [NOTICE] APL-MAIN: for IP Config handle [fe00]
[ 6066.021] [NOTICE] APL-MAIN: IPv6 Addresses:
[ 6066.022] [NOTICE] APL-MAIN:   Link-Local:  fe80:0:0:0 : c693:ff:fe47:ce36
[ 6066.023] [NOTICE] APL-MAIN:   Local: 0:0:0:0 : 0:0:0:0
[ 6066.023] [NOTICE] APL-MAIN:   DNS1:  2001:4860:4860:0 : 0:0:0:8888
[ 6066.024] [NOTICE] APL-MAIN:   DNS2:  fd00:0:0:0 : be05:43ff:fe44:5280
[ 6066.025] [NOTICE] APL-MAIN: =============================
[ 6066.026] [NOTICE] APL-MAIN:
[ 6066.758] [WARN  ] V2GSVC: Security options satisfied, connect!
[ 6066.772] [NOTICE] V2GSVC: TCP: Connected to server without TLS
[ 6067.621] [NOTICE] V2GAPP: Received event 'Session started'
[ 6068.280] [NOTICE] V2GAPP: Received event 'Schedule received'
[ 6068.281] [NOTICE] V2GAPP: Received event 'Ready for charging'
[ 6068.282] [NOTICE] V2GAPP: stxXBAR_V2G_SendXBARStatus_EmptyMsg() - SubAdd = 0xC7
[ 6074.027] [WARN  ] APL-ETH: # Unhandled Ethernet notifycode 365
[ 6108.274] [NOTICE] V2GISO13: Sequence performance time of 40000ms reached.
[ 6112.172] [NOTICE] V2GSVC: TCP: We were disconnected from the server
[ 6112.173] [NOTICE] V2GAPP: Received event 'Session stopped'
[ 6112.173] [NOTICE] V2GAPP: stxXBAR_V2G_SendXBARStatus_EmptyMsg() - SubAdd = 0xCB
[ 6113.182] [NOTICE] SLAC: stxSLAC_Stop()

```

lho-stx, 2024-01-16:
Hey @ByunJongHyun,

first of all thank you for all the detailed information and such a nice and tight issue ticket!

Regarding the issue #116 you mentioned:
I think there is a misconception here! In this issue the author asked if it is possible to get the state of charge (SoC) in charging mode 4 (AC single phase). The SoC is not transferred form the PEV to the EVSE in charging mode 4, this is a limitation of the ISO15118-2. The author asks if it is ok to start with DC charging to get the SoC, let the session fail and continue with basic charging (setting the maximum current available via the PWM duty cycle). This is indeed a valid workaround for the limitation of ISO15118-2.

The ISO15118-2 AC charging modes indeed are working as expected. Unfortunately I can not determine why the message is not received by your charger. I kindly ask you to create a .pcapng log file. Please refer to [this wiki page](https://github.com/Sevenstax/FreeV2G/wiki/guide-creating-log-files) for more information about how to create such a logfile. Please make sure to start FreeV2G with the '-p' parameter, so the Whitebeet mirrors the V2G messages on its 
ethernet port.

Thank you,
lho

ByunJongHyun, 2024-01-16:
The progressed test log is the same, and I created a pcapng file with wireshark as you mentioned and sent it. Please analyze.
[wb-pi-test.zip](https://github.com/Sevenstax/FreeV2G/files/13948154/wb-pi-test.zip)

lho-stx, 2024-01-16:
Thank you for the log. Unfortunately there is no V2G communication in the wireshark log. Is the Whitebeet conected to a network switch? If yes please remove it and connect to your Whitebeet directly.

Nevertheless I may have a clue what is going on here:

From the logs I see, the PEV received a charging schedule from the EVSE where the power is set to 0.

[image]

In Ev.py from FreeV2G [there is a check](https://github.com/Sevenstax/FreeV2G/blob/a81d3bc181e13fa1c2e5ba0f7773312ae196bba1/Ev.py#L266) if the in voltage is greater than the target voltage +/- a tolerance. If this is not the case, the charging will not start. I assume, because the charger does not want to deliver power, it just sets the nominal voltage or maximum current to 0V / 0A, therefore the PEV does not start the charging process. Please double check that your EVSE is setting the nominal voltage and maximum current value to a non-zero value.

Best regards,
lho

closed 2024-02-07 by github-actions[bot]