Sevenstax/FreeV2G #280: Question: how can I use WHITE beet EV ISO15118 to connect to as many different chargers as possible?

Kristanpivk123, 2024-01-25:
Hi,

I am intending to use WHITE beet PI 1.1 to handle the connection with an EVSE.

My goal is to be able to connect to as many different EVSE as possible and fully execute the charging session.

For now I was making a driver for STM32 which would control the WHITE beet over SPI to execute the whole process of charging but I recently started thinking that it might not be able to handle the communication with EVSE whose communication might not be totally up to the standards.

So these are my questions:
- is WHITE beet able to make connection and execute charging with EVSE whose communication is not totally up to the standars?
- is it my job to handle all different exceptions from the standards or will WHITE beet do it for me (automatically adjust its execution to be able to connect to EVSE and start charging)?
- if I used python scripts, will these exceptions (connection to EVSE that might differ from the standards) be handled by the python script? 

Best regards,
Kristan

lho-stx, 2024-01-25:
Hey @Kristanpivk123,

this is a complicated question.

In general I can tell you: the EVSE violates the protocol that leads to termination of the charging session (per standard) the Whitebeet will terminate the session. If it is a mild violation, that not leads to a failure of the charging session (per standard) the Whitebeet will try to continue the charging process. The Whitebeet will also select the correct parameters, that fits its configuration and are in scope of session negotiation (according to ISO-15118). But there is no mechanism in the Whitebeet nor in FreeV2G that tries to find a suitable configuration that work with a specific charger when a charging session fails. As the user application is in charge of setting parameters it also can try different combinations to get the charging session to work, e.g. looking at the energy transfer modes transmitted from the EVSE and setting Whitebeets supported energy transfer modes accordingly and restart the charging session.

As CP handling is included in the Whitebeet informative only (so no functional safety here), you may decide to turn off CP service of Whitebeet and let your user application handle CP lines. You then can decide to ignore violations related to CP.

We also try to adapt Whitebeet firmware if we identify common problems across charger, but until now most of the time we had issues with vehicles violating standard.

If hope this information is enough. If yes, please close this issue ticket, else let me know what you need more information about.

Thank you,
lho

Kristanpivk123, 2024-01-25:
Hi, Iho,

thank you for the answer, you answered to what I was wondering and more.

I have a follow-up question regarding FreeV2G (just to confirm for sure); so FreeV2G does not support a functionality which would try different combinations to start the charging session and if I wished to have that functionality, I would have to make my own user application, right?

Apart from this question I have no more questions on this topic. Thank you, again.

Best regards,
Kristan

lho-stx, 2024-01-25:
Hey @Kristanpivk123,

yes! You would need to modify FreeV2G to enable such a feature. You may also try to enable all parameters in the configuration (e.g. energy transfer modes, authorization methods etc.) the EVSE should select a working configuration.

Keep in mind the configuration still needs to match your used power electronics (if applicable).

Best regards,
lho