Sevenstax/FreeV2G #283: Question: How to use raspberry pi 4 and plc module with SPI communication?

alabora33, 2024-01-30:
Hi,
We are using FreeV2G V02_00_01 version.  I provided a SPI connection between raspberry pi 4 and plc module. 
[image]
I set up the spi connection as shown in the image, checked it many times, and ran Application.py as follows.

**sudo python3 Application.py spi -i spidev5.0 -m "my own mac address" -r EVSE**

I also made the necessary additions to the raspberry /boot/config.txt file.
**dtoverlay=spi5-1cs**

**ls -l /dev/spidev*
crw-rw---- 1 root spi 153, 4 Oca 30 12:17 /dev/spidev5.0**

I can also see spi 5. Also the spidev and rpi.GPIO libraries are installed correctly.

However, I get the errors in the image.
[image]

Why does this error occur and how do I solve it?

Thank you.
Best regards

lho-stx, 2024-01-30:
Hey @alabora33,

did you configure the Whitebeet to use SPI as host controller interface? If not please have a look into the _Whitebeet user manual chapter 9.0.1 Selection of the host controller interface (HCI)_. You can confirm the selected host controller interface by looking at the message on the UART when powercycling the Whitebeet. It tells you if it is using SPI or ETH as host controller interface near the start of the log.

Best regards,
lho

alabora33, 2024-01-30:
[image]
I set it to SPI. You can find the UART output in the image.

lho-stx, 2024-01-31:
Hey @alabora33,

I double checked and in my setup I have no problem with the SPI connection. I used the command:

`sudo -E .venv/bin/python3 Application.py -i spidev0.1 -m 11:22:33:44:55:66 spi -r EVSE`

The main difference here is, that I am using spidev0.1 and not spidev5.0. I enabled the SPI0 connection by running `sudo raspi-config`, then go to '3 Interface options' > 'I4 SPI' > 'Yes'. If you are using another SPI bus you need to change the pins you connect your Whitebeet too. Quickly checking the [RaspberryPi documentation](https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/raspberry-pi/spi-bus-on-raspberry-pi.adoc) it seems SPI5 does use other pins for communication.

Please make sure to connect to the correct SPI pins or use the spidev0.1.

Best regards,
lho

alabora33, 2024-01-31:
https://datasheets.raspberrypi.com/rpi4/raspberry-pi-4-datasheet.pdf
I set the spi5 as GPIO 12-13-14-15 in the datasheet and adjusted the connection accordingly. Spi is also on in raspi-config.
Afterwards, I made the connections in spi0 as you showed and defined spi0 in /boot/config.txt. However, I encountered the same error.
[image]
[image]

lho-stx, 2024-01-31:
Hey @alabora33,

I used '/dev/spidev0.1' (although /dev/spidev0.0 should also work) and did not enable anything in the '/boot/config.txt'. Are you aware to not use the SPI CS Pin, but physical pin 18 for chip select?

Best regards,
lho

alabora33, 2024-01-31:
[image]
As you said, I connected the spi cs pin to the 18th pin, that is, gpio24. There is no need to make adjustments for spi0, but it was necessary for spi5.

lho-stx, 2024-01-31:
If you are using SPI5 please also connect to GPIO24. This Pin is toggled via software. There is a bug with the SPI implementation on Raspberry Pi and I do not know if SPI5 is also affected.

closed 2024-02-22 by github-actions[bot]