Sevenstax/FreeV2G #284: Question: Message performance Timeout reached

pranavsparalikar-eaton, 2024-02-05:
EV Setup:
An EV Simulator

EVSE Setup:
Platform: EVSE (WB-CARRIER-BOARD-EI-1.1)
Firmware version: V02_01_00
Host Controller Interface: SPI
Host: Own HW

I am trying to complete a whole charging cycle with Codico WB EVSE Module and an EV Simulator. All the required steps, viz. cable check, precharge, actual charging (with 400V, 7A) are executed properly. But the charging cycle is terminated abruptly with an error "Message performance Timeout of 4550 ms reached". Pls. refer the screenshot of the serial log below:
[image]

Could you pls. let me know all the possible reasons of this error..? Pls. let me know if any additional info. is required from myside.

Awaiting for your reply.

Pls. note.: I have tested the EV Simulator separately and found to be working well the DIN as well as ISO.

--
Thanks & Regards,
Pranav

lho-stx, 2024-02-06:
Hey @pranavsparalikar-eaton,

please use the other issue template for such kind of issues:

https://github.com/Sevenstax/FreeV2G/issues/new?assignees=&labels=&projects=&template=bug_report.md&title=

Please create a new issue and add the information that is asked for in the issue template. There are multiple reasons why  charging can fail, to really tell you anything about it we need the log files.

Thank you and best regards,
lho

closed 2024-02-06 by lho-stx

pranavsparalikar-eaton, 2024-02-08:
Hi @lho-stx ,

Thanks for your reply. Actually, there was some sync issue in my FW due to which, the the stop charging command (0x6F) was sent before the status message Stop Charging Requested by the EV.

[image]

I realized it and then fixed it. Now everything is working well.

[image]

Thank you for your support.

--
Best Regards,
Pranav