Sevenstax/FreeV2G #286: ConnectionError: Problem with send/receive

varshilgandhi, 2024-02-20:
Setup: We have two board which is connected to PLC. 
First Board is White Beet Carrier Board EVSE ISO15118 and Second Board is White Beet Carrier Board EV ISO15118.
Mac Address of EVSE Board: c4:93:00:48:ac:47
Mac Address of EV Board: c4:93:00:47:cc:30

Now we start the implementation and complete the steps which is given in the ReadMe file, and I complete all that steps.
In the Visual Studio code, I open the two terminal and then in the first terminal I give the EVSE command and in the Second Terminal I give the EV Command but gives me some errors so can you please help me to solve it.

EVSE Command: python Application.py eth -i Ethernet -m c4:93:00:48:ac:45 -r EVSE
EV Command: python Application.py eth -i Ethernet -m C4:93:00:47:CC:2e -r EV

[EV Logs.txt](https://github.com/Sevenstax/FreeV2G/files/14340050/EV.Logs.txt)
[EVSE LOGS.txt](https://github.com/Sevenstax/FreeV2G/files/14340051/EVSE.LOGS.txt)

Below I am giving the WireSharks Logs as well.
[WireShark Logs.zip](https://github.com/Sevenstax/FreeV2G/files/14340142/WireShark.Logs.zip)

lho-stx, 2024-02-20:
Hey @varshilgandhi,

In the log files you send me I see the following message:

> WARNING: WinPcap is now deprecated (not maintained). Please use Npcap instead

Please check your pcap driver. I also give you the recommendation to switch to a Linux based host if possible. We are dropping Windows support soon, because of such problems.

Best regards,
lho

varshilgandhi, 2024-02-20:
I try it on the Linux OS.

I got below logs

[EV.txt](https://github.com/Sevenstax/FreeV2G/files/14343837/EV.txt)
[EVSE.txt](https://github.com/Sevenstax/FreeV2G/files/14343845/EVSE.txt)

lho-stx, 2024-02-20:
Hey @varshilgandhi,

you need to run the FreeV2G as root or with sudo. Please have a look at the [quick start guide in the wiki](https://github.com/Sevenstax/FreeV2G/wiki/guide-quick-start#usage-freev2g) for further information.

Best regards,
lho

varshilgandhi, 2024-02-20:
Yes I did the dual boot and Install the Ubuntu on my system.
Then Again complete all the steps but again when I give the command it gives me below errors which I give in the log files.

[EV.txt](https://github.com/Sevenstax/FreeV2G/files/14343909/EV.txt)
[EVSE.txt](https://github.com/Sevenstax/FreeV2G/files/14343911/EVSE.txt)

lho-stx, 2024-02-20:
Hey,

there are multiple reasons why the communication can fail. To investigate this issue I need wireshark (or tcpdump) logs  and UART logs for the session.

Please create the log files and send it to me.

Best regards,
lho

varshilgandhi, 2024-02-20:
Then I change Ethernet to wifi and then I got below kind of logs.

[EVSE.txt](https://github.com/Sevenstax/FreeV2G/files/14344006/EVSE.txt)
[EV.txt](https://github.com/Sevenstax/FreeV2G/files/14344008/EV.txt)

varshilgandhi, 2024-02-20:
Wait I am sending you the wireshark logs and UART Logs

varshilgandhi, 2024-02-20:
UART LOGS File:

[EVBOARD UARTLOGS.txt](https://github.com/Sevenstax/FreeV2G/files/14344284/EVBOARD.UARTLOGS.txt)
[EVSEBOARD UARTLOGS.txt](https://github.com/Sevenstax/FreeV2G/files/14344285/EVSEBOARD.UARTLOGS.txt)

WireShark Logs FIle:

1. Using Ethenet
[WireShark Logs - Ethenet.pcapng.gz](https://github.com/Sevenstax/FreeV2G/files/14344289/WireShark.Logs.-.Ethenet.pcapng.gz)

2. Using Wifi
[WireShark LOgs - Wifi.pcapng.gz](https://github.com/Sevenstax/FreeV2G/files/14344290/WireShark.LOgs.-.Wifi.pcapng.gz)

lho-stx, 2024-02-20:
Hey @varshilgandhi,

these logs are not matching the output of FreeV2G you send me earlier. In the latest FreeV2G logs it seems there is a connection between Whitebeet and your host.

In the wireshark logs there are only two framing messages. These are send from the host to two different Whitebeets. It seems they are send over the same interface. Are you using a network switch between the Whitebeets and your host PC and a single network interface? This setup is not supported, please connect each Whitebeet directly to a separate network interface and try again. You also can read more about the setup in the quick [start guide](https://github.com/Sevenstax/FreeV2G/wiki/guide-quick-start).

Best regards,
lho

varshilgandhi, 2024-02-20:
Hi,
In my setup I have one EVSE WHitebeet board, one EV WhiteBeet BOard and one PLC and both the boards I connected via PLC.
And ALL three connected in the same Network.
And right now what happens is when I run the commands first time then it gives me below logs till start V2G.

EVSE Logs:

(.venv) varshil@varshil-Vostro-3520:~/FreeV2G$ sudo .venv/bin/python3 Application.py eth -i enp1s0 -m c4:93:00:48:ac:45 -r EVSE
Welcome to Codico Whitebeet EVSE reference implementation
Initiating framing interface
iface: ETH, name: Wi-Fi, mac: c4:93:00:48:ac:45
WHITE-beet-EI firmware version: V02_01_00
Set the CP mode to EVSE
Set the CP duty cycle to 100%
Start the CP service
Start SLAC in EVSE mode
EV already connected
Start SLAC matching
Set duty cycle to 5%
SLAC matching successful
Set V2G mode to EVSE
Start V2G
We did not receive a frame before the timeout of 30s
EVSE loop finished
Goodbye!

EV logs:

(.venv) varshil@varshil-Vostro-3520:~/FreeV2G$ sudo .venv/bin/python3 Application.py eth -i enp1s0 -m c4:93:00:47:cc:2e -r EV
Welcome to Codico Whitebeet EV reference implementation
Initiating framing interface
iface: ETH, name: Wi-Fi, mac: c4:93:00:47:cc:2e
WHITE-beet-PI firmware version: V01_00_06
Set the CP mode to EV
Start the CP service
Set the CP state to State B
Set SLAC to EV mode
Start SLAC
EVSE connected
Start SLAC matching
SLAC matching successful
Set V2G mode to EV
Set V2G configuration
Set DC charging parameters
Set AC charging parameters
Create new charging session
ControlPilot duty cycle: 0.0

varshilgandhi, 2024-02-20:
I have one question that as you told me connect each Whitebeet to a separate ethernet interface (without switch/router etc.)
But what I am doing is I am connect that both WhiteBeet Ethernet cable to router and then that router ethernet cable to the My System.
Is it Correct?

I think it should be right because first time when I give the command then it works till the Start V2G and Create New Charging Session and after the SEcond time when I give same commands then I did not get anything just directly show me the connection kind of errors.

lho-stx, 2024-02-20:
Hey @varshilgandhi,

You have this setup:

```mermaid
graph LR;
Whitebeet_PEV <---> Router
Whitebeet_EVSE <---> Router
Router <---> enp0s1
Whitebeet_PEV <-- PLC ---> Whitebeet_EVSE

subgraph Host
    enp0s1 <---> FreeV2G_EV
    enp0s1 <---> FreeV2G_EVSE
end
```

This setup is not supported. You may be lucky if this setup works, but to set it up correctly you would need to setup VLANs in your router/switch and then add different interfaces on your host, one for each VLAN. 

The setup you should use is this one:

```mermaid
graph LR;
Whitebeet_PEV <---> eth0
Whitebeet_EVSE <---> eth1
Whitebeet_PEV <-- PLC ---> Whitebeet_EVSE

subgraph Host
    eth0 <---> FreeV2G_EV
    eth1 <---> FreeV2G_EVSE
end
```

I advise you to use ethernet-usb dongles for eth0 and eth1.

Please put each Whitebeet in an separate network and try again.

Best regards,
lho

lho-stx, 2024-02-21:
Hey @varshilgandhi,

I am happy to hear that you made some progress. Does it work when you do a reset of the Whitebeets before starting a new session?

I always need logs of wireshark, UART and FreeV2G output for inspecting an issue.

Best regards,
lho

varshilgandhi, 2024-02-21:
I need one help from you.

I added the reset command in my code.

In the Whitebeet python file I added below function,

def systemRestart(self):
        response = self._sendReceiveAck(self.sys_mod_id, self.sys_sub_restart_system, None)
        return response
        
and also added that command like below,

# System sub IDs
        self.sys_mod_id = 0x10
        self.sys_sub_get_firmware_version = 0x41
        self.sys_sub_restart_system = 0x48
        
And then IN the EV file added below code,

def _initialize(self):
        """
        Initializes the whitebeet by setting the control pilot mode and setting the duty cycle
        to 100%. The SLAC module is also started. This one needs ~1s to 2s to be ready.
        Therefore we delay the initialization by 2s.
        """

        print("Restarting the device...")
        self.whitebeet.systemRestart()
        print("Set the CP mode to EV")
        self.whitebeet.controlPilotSetMode(0)
        print("Start the CP service")
        self.whitebeet.controlPilotStart()
        print("Set the CP state to State B")
        self.whitebeet.controlPilotSetResistorValue(0)
        print("Set SLAC to EV mode")
        self.whitebeet.slacSetValidationConfiguration(0)
        print("Start SLAC")
        self.whitebeet.slacStart(0)
        time.sleep(2)

And then IN the EVSE.py file added below code,

    def _initialize(self):
        """
        Initializes the whitebeet by setting the control pilot mode and setting the duty cycle
        to 100%. The SLAC module is also started. This one needs ~1s to 2s to be ready.
        Therefore we delay the initialization by 2s.
        """
        print("Restarting the device...")
        self.whitebeet.systemRestart()
        print("Set the CP mode to EVSE")
        self.whitebeet.controlPilotSetMode(1)
        print("Set the CP duty cycle to 100%")
        self.whitebeet.controlPilotSetDutyCycle(100)
        print("Start the CP service")
        self.whitebeet.controlPilotStart()
        print("Start SLAC in EVSE mode")
        self.whitebeet.slacStart(1)
        time.sleep(2)

Then I give the EVSE Command and I got below logs at EVSE:

Terminal Logs of EVSE:
[EVSE Terminal Logs.txt](https://github.com/Sevenstax/FreeV2G/files/14359557/EVSE.Terminal.Logs.txt)

UART Logs of EVSE:
[EVSE UART Logs.txt](https://github.com/Sevenstax/FreeV2G/files/14359560/EVSE.UART.Logs.txt)

WireShark Logs of EVSE:
[EVSE WireShark Logs.pcapng.gz](https://github.com/Sevenstax/FreeV2G/files/14360165/EVSE.WireShark.Logs.pcapng.gz)

At that same time I give the EV command and I got below logs at EV:

Terminal Logs of EV:
[EV Terminal Logs.txt](https://github.com/Sevenstax/FreeV2G/files/14359563/EV.Terminal.Logs.txt)

UART Logs of EV:
[EV UART Logs.txt](https://github.com/Sevenstax/FreeV2G/files/14359564/EV.UART.Logs.txt)

WireShark Logs of EV:
[EV WireShark Logs.pcapng.gz](https://github.com/Sevenstax/FreeV2G/files/14359566/EV.WireShark.Logs.pcapng.gz)

varshilgandhi, 2024-02-21:
Hey, 
I successfully found the solution and fully connection worked now and every time also it reset automatically with the command.
I give me changed code for the community,

  # System sub IDs
        self.sys_mod_id = 0x10
        self.sys_sub_get_firmware_version = 0x41
        self.sys_sub_restart_system = 0x48

EV.PY FILE:

def _initialize(self):
        """
        Initializes the whitebeet by setting the control pilot mode and setting the duty cycle
        to 100%. The SLAC module is also started. This one needs ~1s to 2s to be ready.
        Therefore we delay the initialization by 2s.
        """

        print("Restarting the device...")
        self.whitebeet.systemRestart()
        time.sleep(10)
        print("Set the CP mode to EV")
        self.whitebeet.controlPilotSetMode(0)
        print("Start the CP service")
        self.whitebeet.controlPilotStart()
        print("Set the CP state to State B")
        self.whitebeet.controlPilotSetResistorValue(0)
        print("Set SLAC to EV mode")
        self.whitebeet.slacSetValidationConfiguration(0)
        print("Start SLAC")
        self.whitebeet.slacStart(0)
        time.sleep(2)

EVSE.PY FILE:

   def _initialize(self):
        """
        Initializes the whitebeet by setting the control pilot mode and setting the duty cycle
        to 100%. The SLAC module is also started. This one needs ~1s to 2s to be ready.
        Therefore we delay the initialization by 2s.
        """
        print("Restarting the device...")
        self.whitebeet.systemRestart()
        time.sleep(10)
        print("Set the CP mode to EVSE")
        self.whitebeet.controlPilotSetMode(1)
        print("Set the CP duty cycle to 100%")
        self.whitebeet.controlPilotSetDutyCycle(100)
        print("Start the CP service")
        self.whitebeet.controlPilotStart()
        print("Start SLAC in EVSE mode")
        self.whitebeet.slacStart(1)
        time.sleep(2)

lho-stx, 2024-02-22:
Hey @varshilgandhi,

I am happy to hear your solved the issue by yourself. Thank you for contributing, we will consider adding your changes to the FreeV2G code.

Best regards,
lho

closed 2024-02-22 by lho-stx