Sevenstax/FreeV2G #288: Issue with F-150 Lightning initiating charging

ahuster, 2024-03-07:
**Your Setup**
Platform: EVSE
Firmware Version: i.e. V02_01_00
Host Controller Interface: Ethernet
Host: FreeV2G-based implementation

**Describe the bug**
We're having trouble getting an F-150 Lightning to connect and charge. We've talked to folks at Codico for some initial investigation but it seems like this is more likely a software issue. We fail at the evseStartListen step, timing out after 30 seconds with no response.

We've been able to successfully initiate charging with other vehicles, like a Kia EV6 and Chevrolet Bolt, so we know our software works at least some of the time.

**Logs**
- Attached
[eth0_f150_2.pcap.zip](https://github.com/Sevenstax/FreeV2G/files/14524680/eth0_f150_2.pcap.zip)

lho-stx, 2024-03-08:
Hey @ahuster,

Thank you for the ethernet log files. In the log files I see the host is polling the CP state from the Whitebeet. The Whitebeet tells the host, that the CP is in state B. This is going on for ~80s. Then the Whitebeet tells the host that the CP state is state A, the host starts configuring the Whitebeet and starts SLAC. The last message in the log is the Whitebeet sending the notification, that the CP state has changed to state A. For me it seems the host application has a bug, because I expect the host to start SLAC when it reads state B.

You are using a custom implementation? Please double check your host application to actually start configuration and SLAC when it reads state B.

Best regards,
lho

ahuster, 2024-03-08:
To be clear, our implementation is almost identical to the FreeV2G implementation, with just some additional code added to pass information to another Python script that allows us to start our inverter at the right time and things like that. So from this perspective, I don't think we've changed anything about the CP handling from what's in this repo.

lho-stx, 2024-03-08:
Hey @ahuster,

then I need to some more information. What are you actually doing and in which order when the error occurs? I also need the log files of the UART output from the Whitebeet and the output of the FreeV2G script. I'll then try to reproduce the issue.

Thank you and best regards,
lho

ahuster, 2024-03-08:
We don't have UART logs as we're connected to the Whitebeet over ethernet. Here are the V2G logs.
[V2Glog.txt](https://github.com/Sevenstax/FreeV2G/files/14538824/V2Glog.txt)

The sequence here is:
- The charger is initialized and waits for an EV to be connected
- The EV is connected, SLAC matching starts
- SLAC matching is completed successfully, so the Set DC Charging Parameters message is sent
- We try to start V2G by doing EvseStartListen
- This times out after 30s

It's the same flow as in the Evse.py script.

lho-stx, 2024-03-08:
Hey @ahuster,

I double checked. I missed the first charging attempt, my bad. It seems the EVSE does not answer the SECCDP requests. Unfortunately I can not see the reason for this behaviour from the log files you provided, the UART logs would help alot here. Are you using a Whitbeeet Carrier Board or a plain Whitebeet? If you are using the Carrier Board I kindly ask you to connect the USB-C connector at J11 to your RaspberryPi and create a log. You can use the built in Linux tools for this:

```console
stty -F /dev/ttyACM0 1000000
cat /dev/ttyACM0 | tee uart-log.txt
```

Replace /dev/ttyACM0 with the actual serial device connected to the Whitebeet.

In the meantime I'll try to reproduce the issue.

Best regards,
lho

ahuster, 2024-03-13:
Hi lho,

Sorry for the delay, here's a log from the UART for the F-150. I suspect the issue is the SLAC parameter requests that are coming in after we do V2GStartListen. 

[uart-log-f150.txt](https://github.com/Sevenstax/FreeV2G/files/14592968/uart-log-f150.txt)

lho-stx, 2024-03-15:
Hey @ahuster,

please allow some more time as I am still looking into this

Best regards,
lho

lho-stx, 2024-03-18:
Hey @ahuster,

I reviewed your uart logs and I am not sure how to interpret them. 

wireshark logs:

After SLAC has finished successfully the V2G service is started and the Whitebeet listens for incoming SECCDP messages. In wireshark log frame 439 the PEV sends a IPv6 neighbour solicitation. In frame 451 the PEV sends the first SECCDP request. This fails, we do not see any SECCDP response messages. 

uart logs:

Same picture as above. SLAC successful, V2G service started, EVSE is listening, but no SECCDP requests seen. Then the PEV tries to start SLAC again, but the EVSE is still in matched stated.

My theory is:
I expect the PEV to send a IPv6 neighbor advertisement after the solicitation message. This is not done, so the EVSE does not know the IPv6 address of the PEV and therefore the SECCDP requests are not answered. The PEV does not see any SECCDP responses and thinks it is not connected to an EVSE. In addition to the missing IPv6 neighbor advertisement the PEV does not send 50 SECCDP per ISO 15118-2.

I also have another question:

Is the uart log from the same session as the wireshark log?

I am a bit confused, because I do not see any SLAC messages in the wireshark log. Did you connect to your Whitebeet directly, or are you using e.g. a network switch? If yes, please remove the network switch in your setup and create new log files.

Best regards,
lho

ahuster, 2024-03-19:
They're not from the same session. I can work on capturing both logs at the same time if that would help you diagnose.

lho-stx, 2024-03-20:
Yes, this is necessary so I can align the events in the log files! Thank you!

ahuster, 2024-03-25:
Unfortunately our customer didn't have the F-150 available today, but they do have one of their battery-powered bucket loaders (which is what our charger is actually meant for) that exhibits the same behavior. I've attached the ethernet and UART logs for that.

[Loader_Logs.zip](https://github.com/Sevenstax/FreeV2G/files/14744092/Loader_Logs.zip)

lho-stx, 2024-03-25:
Hey @ahuster,

thank you for the logs. Altough V2G traffic is missing, it is the exactly the same behavior here too. The PEV tries to restart SLAC again after some time, but the EVSE is not responding, because it is still waiting for V2G traffic. Is the battery-powered bucket loader known to work with other chargers? Do you have log files of a working setup, so I can compare the two logfiles?

I will also discuss this issue with a colleague.

Best regards,
lho

PS:
please always run FreeV2G with the `-p` argument when creating log files. Then I am able to also see the V2G traffic.

ahuster, 2024-03-27:
It does work with other chargers, but none that we have the capability of logging unfortunately.

I saw in the other issue thread (about the Mach-E) that this might be an issue with this latest firmware? Is it possible to flash old firmware onto the board and try that? How would we go about that?

Also, my company is willing to pay to expedite any diagnosis/fixes for this, if that is possible. For that, you can reach out to me directly at andrew.huster@acculonenergy.com to discuss.

lho-stx, 2024-03-28:
Hey @ahuster,

yes the customer charged the vehicle successully with v1.1.7. You can download the firmware from CODICO website and downgrade, but this is a not officially supported use case. Also the framing API was changed from v1.1.7 to v2.1.0.

Please apologize for not being clear enough:
I meant can you create a log file of your EVSE running a successful session (e.g. with the KIA 6).

Regarding the direct support: I'll discuss this internally and will reach out via e-Mail if necessary.

Thank you in advance,
lho

ahuster, 2024-04-03:
Here are logs from the Kia EV6.
[eth0_kia2.pcap.zip](https://github.com/Sevenstax/FreeV2G/files/14855776/eth0_kia2.pcap.zip)
[uart-log-kia.txt](https://github.com/Sevenstax/FreeV2G/files/14855779/uart-log-kia.txt)

lho-stx, 2024-04-05:
Hey @ahuster,

I checked the logs again and there are multiple findings. I'll send you an e-Mail, because it might be necessary to run a debug session, that is out of scope of this issue tracker.

I'll leave the ticket open until we have found a final solution. I'll update the ticket in this case.

Best regards,
lho

closed 2024-04-26 by github-actions[bot]