Sevenstax/FreeV2G #289: Request for SLAC attenuation profile retrieval from whitebeet in FreeV2G

giguerator, 2024-03-08:
**Your Setup**
Platform: PEV 
Firmware Version: i.e. V01_00_07
Host Controller Interface: Ethernet 
Host: FreeV2G (EV_v1.0.7_0) + some minor changes

**Describe the bug**
I want to be able to retrieve the SLAC profile in freeV2G. An array of values or the maximum/minimum over given range could also do.

**To Reproduce**
N/A

**Expected behavior**
Following SLAC successful or not, I would like to option for FreeV2G to provide SLAC attenuation profile metrics (or complete profile) to the console.

**Logs**
N/A

**Additional context**
SLAC calibration will likely be required for production tests of EVSEs in the very near future.

lho-stx, 2024-03-11:
Hey @giguerator,

actually there is a 'Get AttnTxRef Values' message in the SLAC module (chapter 16.4.10). You can modify FreeV2G to send this message and e.g. print the received attenuation profile. You can also use the 'Set AttnTxRef Values' message to set the attenuation profile.

I hope this is the feature you are looking for.

Best regards,
lho

giguerator, 2024-03-11:
@lho-stx Thanks, for the info! For some reason, I had missed that in the documentation.

I will have a look and get back to this thread for questions.

giguerator, 2024-03-11:
@lho-stx , I am a bit confused here in how the whitebeet works in that context. When I request the AttnTxRef Values (16.4.10), I get all zeros. Based on the explanation in 5.5, it seems this is an attenuation profile that the HLE sets for future usage. I am really looking the get the content of the CM_ATTEN_CHAR.IND message from the EVSE I am connected to. 

Can you enlighten me a bit?

lho-stx, 2024-03-20:
Hey @giguerator,

I think I understand the confusion now! I misunderstood your intenteion, you actually want to read the values measured during SLAC process. Indeed the messages I told you to look into are used to calibrate the Whitebeet, after you conducted the channel estimation routine. This is not suitable for getting the actual measurements of the attenuation.

Unfortunately there is no possibility to get the measurements of CM_ATTEN_CHAR.IND at the moment.

Best regards.
lho

closed 2024-03-20 by lho-stx

giguerator, 2024-03-20:
@lho-stx, thanks for the feedback. Can this be added to the feature backlog?

Also, are you aware of a limitation that would prevent from implementing via FW/Python (ie. requiring hw change) ?