Sevenstax/FreeV2G #290: Connection Issue with PLC and EV

varshilgandhi, 2024-03-12:
**Setup**
Platform: PEV 
Firmware Version: i.e. Master Branch Code
Host Controller Interface: Ethernet 
Host: Own Implementation

Platform: EVSE
Firmware Version: i.e. Master Branch Code
Host Controller Interface: WiFi
Host: Own Implementation

**Describe the bug**
When I start the communication between the EVSE and EV and in between the PLC.
Then what happens is my execution is at EVSE Side terminal it shows me that **Start V2G** and at EV Side terminal it shows me that **Start SLAC matching**. 
Then I disconnect CP Line of EV Board which connect to PLC and again I connect to PLC then it again continue to the communication.

**Logs**

**EVSE Terminal Logs:**

(myenv) PS C:\Tirex\Latest Code 11-03-2024\FreeV2G-Latest Code> python Application.py eth -i Wi-Fi -m c4:93:00:48:AC:45 -r EVSE  
Welcome to Codico Whitebeet EVSE reference implementation
Initiating framing interface
iface: ETH, name: Wi-Fi, mac: c4:93:00:48:AC:45
WHITE-beet-EI firmware version: V02_01_00
Restarting the device...
Set the CP mode to EVSE
Set the CP duty cycle to 100%
Start the CP service
Start SLAC in EVSE mode
EV already connected
Start SLAC matching
Set duty cycle to 5%
SLAC matching successful
Set V2G mode to EVSE
Start V2G

**EV Terminal Logs:**

(myenv) PS C:\Tirex\Latest Code 11-03-2024\FreeV2G-Latest Code> python Application.py eth -i Ethernet -m c4:93:00:47:CC:2e -r EV
Welcome to Codico Whitebeet EV reference implementation
Initiating framing interface
iface: ETH, name: Ethernet, mac: c4:93:00:47:CC:2e
WHITE-beet-PI firmware version: V01_00_06
Restarting the device...
Set the CP mode to EV
Start the CP service
Set the CP state to State B
Set SLAC to EV mode
Start SLAC
EVSE connected
Start SLAC matching

Now I disconnect the CP line of the PLC and Again Connect:

**EVSE Side Logs:**

"Session started" received
Protocol: 1
Session ID: a0c2e9dd466e0030
EVCC ID: c4930047cc2e
"Payment selcted" received
Selected payment method: 0
"Request Authorization" received
54999
Authorize the vehicle? Type "yes" or "no" in the next 53s: yes
Vehicle was authorized by user!
"Energy transfer mode selected" received
Energy request: 1000
Maximum voltage: 300
[... 506 lines cut ...]
        Payment method: 0
        Energy transfer mode: 0
"DC Charge Parameters Changed" received
        EVSE min voltage: 0
        EVSE min current: 0
        EVSE min power: 0
        EVSE max voltage: 400
        EVSE max current: 100
        EVSE max power: 400
        EVSE present voltage: 0
        EVSE present current: 0
        EVSE status: 0
        EVSE isolation status: 0
        EVSE voltage limit achieved: 0
        EVSE current limit achieved: 0
        EVSE power limit achieved: 0
        EVSE peak current ripple: 0

Here below I am attach my setup photo so that you will get better understanding:

[image]
[image]
[image]

https://github.com/Sevenstax/FreeV2G/assets/76485639/db4871ef-8f09-4914-9f9f-182c145e52df

lho-stx, 2024-03-12:
Hey @varshilgandhi,

do I understand it correctly: you are starting FreeV2G on both sides, after SLAC you are disconnectin CP, then reconnect and the communication continues?

This use case is not handled by FreeV2G. FreeV2G is just a demonstration to get an impression how the communication works. It is not designed to run in an actual charger and therefore does not comply to all ISO15118 requirements. If you want FreeV2G to support this use case, you need to implement it by your own. The Whitebeet sends a status message when the control pilot state or PWM duty cycle changes (e.g. when you disconnect the CP line), please have a look into chapter 15.5 in the Whitebeet manual. You need to listen for this status message and stop the charging.

Please let me know if I misunderstood your issue!

Best regards,
lho

varshilgandhi, 2024-03-12:
Hi,
Yes I start the FREEV2G on both the sides and after Start V2G in EVSE and SLAC matching in EV I did not get any logs after that and at EVSE Side it shows me message that it did not receive the frame in particular time.
But when it reach to Start V2G and SLAC At that time when I disconnect and then connect CP line to PLC then my logs continue to come at both sides.

So can please explain me that what kind problem it is?

lho-stx, 2024-03-12:
Hey @varshilgandhi,

ok then I misunderstood your question. Can you please create wireshark, uart and FreeV2G logs of the sessions? 

Please have a look into [the wiki for more information](https://github.com/Sevenstax/FreeV2G/wiki/guide-creating-log-files).

Best regards,
lho

varshilgandhi, 2024-03-13:
Heym @lho-stx,
I will send you all the logs.
First I want to tell you that I modify the code part and in that modifications like I put reset command in the code itself so it reset both the EVSE and EV board automatically.

Now I want to explain you about My SetUp.
I connect EV Board to direct with my laptop using Ethenet Cable and CP and PE Line of EV Board into the PLC.
Then I connect My EVSE board to the Router using Ethenet and then I connect My Laptop with that Router Network.

Now I start the communication between both the boards using this below two commands:

python Application.py eth -i Wi-Fi -m c4:93:00:48:AC:45 -r EVSE
python Application.py eth -i Ethernet -m c4:93:00:47:CC:2e -r EV

Scenario 1: When I did not make any changes in the PLC like nothing to disconnect and connect I am giving you the logs which I found,

UART Logs:

lho-stx, 2024-03-15:
Hey @varshilgandhi,

I am still looking into this. Please allow some more time for investigation!

Best regards,
lho

varshilgandhi, 2024-03-16:
Hey @lho-stx ,

No rush at all! Take all the time you need for your investigation. I appreciate your efforts and I'll patiently await your update.

Best regards,
@varshilgandhi

varshilgandhi, 2024-03-18:
Hey @lho-stx,
Is there any update you get?

Best regards,
@varshilgandhi

lho-stx, 2024-03-20:
Hey @varshilgandhi,

Thank you for the detailed log files and information.

I checked the logs, but unfortunately I can not tell why it is working when you reconnect the CP yet...

But I found some other caveats and issues:
1. You are using Windows. Windows is known to be unreliable and we will drop Windows support soon.
2. Using any network gear in between the Whitebeet and your host PC is problematic. Try to connect to both Whitebeets via ethernet **directly**, no switches or routers in between (especially WiFi)

Please alter your setup, at least bullet point number 2.

If the issue still persists, please create new log files, but know pass the -p parameter to Application.py too!

Thank you,
lho

varshilgandhi, 2024-03-21:
Hey @lho-stx,

Thank you for your insights and recommendations.

I've taken note of the issues highlighted and will make the necessary adjustments. Firstly, I'll switch to using Linux OS to address the reliability concerns with Windows. Additionally, I'll directly connect both WhiteBeets to my host PC via Ethernet, bypassing any intermediary network gear as suggested.

I'll keep you updated on the progress after implementing these changes.

Best Regards,
@varshilgandhi

varshilgandhi, 2024-03-22:
Hey @lho-stx,
First of all I switch to Linux OS.
Then I connect both EVSE and EV boards via the EtherNnet.
After that I give below commands into separate terminals.

sudo .venv/bin/python3 Application.py eth -i enp1s0 -m c4:93:00:48:AC:45 -r EVSE
sudo .venv/bin/python3 Application.py eth -i enp1so -m c4:93:00:47:cc:2e -r EV

Then my communication starts and then it working perfectly but during the that when I disconnect the CP line from the PLC and then also that communication continuous on.

So I want to know that during the communication If I disconnect the CP Line from the PLC then also why it continues receives the frame and then it SOC also goes to 100% and still CP line is disconnect from the PLC.

Best Regards,
@varshilgandhi

varshilgandhi, 2024-03-22:
I want to send the WireShark Logs But In Linux When I Install the WireShark and after that when I open it I can see the Ethernet                                                Option in that.

[image]

lho-stx, 2024-03-22:
Hey @varshilgandhi,

first of all I am happy that you made progress!

As your implementation is based on FreeV2G it will most likely not handle CP disconnect during an active charging session (because FreeV2G does not handle such a case). You need to implement this handling by your own.

PLC is known for high crosstalk. Depending on your setup you may be able to receive the signal on the other side, even if the CP is disconnected. I also had such experiences.

This two conditions combined may result in the effect, that the communication is still taking place, even after disconnecting CP.

Regarding wireshark:
Most likely your user does not have the right permission to access the raw sockets. Is your user in the wireshark group? You also can start wireshark as root. 

Best regards,
lho

varshilgandhi, 2024-03-23:
Hey @lho-stx,

lho-stx, 2024-03-25:
Hey @varshilgandhi,

the Whitebeets are a slave device in the sense, that it will not do anything without configuration through a host controller. It has a set of default values, but I will only react to V2G messages in respect of it's configuration.

This also means there is no integrated "simulator mode". The V2G communication WILL break after the ISO15118-2 related timeouts are not met (e.g. 250 ms in a DC charge loop). This is also the reason why the communication can continue when changing the CP state, the user application is in charge of handling these events.

As a experiment you can try to initiate communication, but skip SLAC and CP handling and keep CP disconnected. The Whitebeets should be in the same PLC network after reset and if not, you can tell them to join the same network (please see 17.4.1 Join HPGP network for more details). If it is still possible to run the full charging session I highly suspect it is due crosstalk / EMV.

I am looking forward to your logfiles!

Best regards,
lho

Another thing that came to my mind is, that your terminal is just too slow. Maybe it is still printing it's buffer, and the printing does not keep up with actual communication. But the session runs for several minutes after disconnecting CP, right?

varshilgandhi, 2024-03-26:
Hey @lho-stx ,

Here First I disconnect the CP Line and start the communication:

At that time this below is my terminal logs:

EVSE SIDE:

PS C:\Tirex\Latest Code 11-03-2024\FreeV2G-Latest Code> .\myenv\Scripts\activate
(myenv) PS C:\Tirex\Latest Code 11-03-2024\FreeV2G-Latest Code> python Application.py eth -i Ethernet -m c4:93:00:48:AC:45 -r EVSE
Welcome to Codico Whitebeet EVSE reference implementation
Initiating framing interface
WARNING: Wireshark is installed, but cannot read manuf !
WARNING: Wireshark is installed, but cannot read manuf !
iface: ETH, name: Ethernet, mac: c4:93:00:48:AC:45
WHITE-beet-EI firmware version: V02_01_00
Restarting the device...
Set the CP mode to EVSE
Set the CP duty cycle to 100%
Start the CP service
Start SLAC in EVSE mode
Restarting the device...
Set the CP mode to EVSE
Set the CP duty cycle to 100%
Start the CP service
Start SLAC in EVSE mode
Wait until an EV connects

EV SIDE:

PS C:\Tirex\Latest Code 11-03-2024\FreeV2G-Latest Code> .\myenv\Scripts\activate
(myenv) PS C:\Tirex\Latest Code 11-03-2024\FreeV2G-Latest Code> python Application.py eth -i Ethernet -m c4:93:00:48:AC:45 -r EVSE
Welcome to Codico Whitebeet EVSE reference implementation
Initiating framing interface
WARNING: Wireshark is installed, but cannot read manuf !
WARNING: Wireshark is installed, but cannot read manuf !
iface: ETH, name: Ethernet, mac: c4:93:00:48:AC:45
WHITE-beet-EI firmware version: V02_01_00
Restarting the device...
Set the CP mode to EVSE
Set the CP duty cycle to 100%
Start the CP service
Start SLAC in EVSE mode
Restarting the device...
Set the CP mode to EVSE
Set the CP duty cycle to 100%
Start the CP service
Start SLAC in EVSE mode
Wait until an EV connects

Then I wait around 30 min then also both are in the same situation waiting for EVSE and EV to connect.

Then I connect the CP and PE line and after that both side connection start and SOC goes 100% and charging complete.

Here Below I capture the Wireshark Logs as well and in that I disconnect the CP line and start the connection and then wait for some time and during that time both shows like waiting for EVSE and EV to connect and after that I connect the CP line and it works and complete the whole charging.

[WireShark Update Latest Logs.zip](https://github.com/Sevenstax/FreeV2G/files/14756431/WireShark.Update.Latest.Logs.zip)

I also want to know as you told me above that The Whitebeets should be in the same PLC network after reset and if not, you can tell them to join the same network (please see 17.4.1 Join HPGP network for more details). But honestly, I don't understand how to do it. So can you please tell me little bit about it.

Best Regards,
@varshilgandhi

varshilgandhi, 2024-03-26:
You refer me to implement this Join HPGP Network right:

[image]

lho-stx, 2024-03-28:
@varshilgandhi,

yes!

Regarding the log file:
I expect to get the 'Duty cycle changed' notification from the SLAC service when disconnecting the CP line during an active charging session (basically simulating pulling out the charging gun). I do not see this notification in the log file. For me it means the device thinks the CP is still connected. Can you also provide the log file of the EVSE side?

Just for my sanity:

you are actually pulling out the CP connector? You are going from this:

```mermaid
graph LR;

subgraph  CONNECTED
direction LR
ev_cp[CP] <--> evse_cp[CP];
ev_pe[PE] <--> evse_pe[PE];

subgraph PEV
ev_cp
ev_pe
end

subgraph EVSE
evse_cp
evse_pe
end

end
```

to this:

```mermaid
graph LR;

subgraph  DISCONNECTED
direction LR
evse_cp[CP];
ev_pe[PE] <--> evse_pe[PE];
ev_cp[CP];

subgraph PEV
ev_cp
ev_pe
end

subgraph EVSE
evse_cp
evse_pe
end

end
```

Thank you and best regards,
lho

varshilgandhi, 2024-03-28:
1. HPGP Network:
-> I implemented HPGP Command in my code and after that when I execute then also not any such changes in the previous issue. and the same thing again happens like suppose from the start I disconnect the CP line and I start the communication and after that at EVSE side it shows me that waiting for EV to connect and at EV side it shows me that waiting for EVSE to connect and after some time when I connect the CP line again then it starts the communication and SOC also increasing  and then when I disconnect the CP line now it does not make any difference and communication continues and Full Charging done and SOC goes to 100%.

2.you are actually pulling out the CP connector? You are going from this:
-> As you ask me this, So I want to tell you that yes, I am pulling out the CP connector. But Interestingly I tried that also during the communication I remove the first PE line from the EV and then also communication continues giving frames but at that time when I remove the CP line as well then at EV side session stopped command trigger. and this same thing happens vice-versa as well. And this same happens when during the communication when I remove both the CP and PE line from the EV boards then at both the EVSE and EV get the session stop command.

Log File OF EVSE:

[WireShark Update Latest Today Logs.zip](https://github.com/Sevenstax/FreeV2G/files/14789597/WireShark.Update.Latest.Today.Logs.zip)

I implemented HPGP Network in the source code and then Again when I start the communication and also it starts the charging and soc with 98% and after that When I disconnect the CP line from the EV board then after around 12 to 15 second session stop command was triggered and stop the charging. 

But then I again start the communication and at this time now I disconnect CP line from the PLC not from the EV board and this time my charging did not stop and goes to 100% and charging is full.

Here below I attach that logs of EVSE:
[WireShark Update Latest Today Logs 2.zip](https://github.com/Sevenstax/FreeV2G/files/14790032/WireShark.Update.Latest.Today.Logs.2.zip)

Regards,
@varshilgandhi

varshilgandhi, 2024-04-02:
Hey @lho-stx ,
Is there any update you got?

Regards,
@varshilgandhi

varshilgandhi, 2024-04-02:
Right now during the communication when I disconnect the CP line from the EV then after 12 second it receive the session stop command  it stop the charging and similarly during the communication when I disconnect the CP line from the EVSE board in that also after 10-12 seconds it receive the session stop command and it stops the charging. 

But during the communication when I disconnect the CP line from the PLC then also charging continuous on and SOC also goes to 100% and charging is full.

lho-stx, 2024-04-02:
Hey @varshilgandhi,

we had holiday season here in germany and therefore no one has worked on the issue yet. Please allow some more time for investigation.

What do you mean by

> disconnect the CP line from the PLC

Please answer the two  questions:

1. When you disconnect the CP line from either the PEV and EVSE board, the communication is stopping as expected? 
2. When disconnecting the CP line in middle of the wire the communication still continues?

If both answers are yes, I think the two disconnected wires are acting as antenna and the PLC signal is still transmitted over these antennas. This is expected, because this behavior is inherent in the way PLC is working. It is not observed when disconnecting the PLC lines from the boards, because the coupling is to low for the receivers.

Therefore you actually need to handle the CP disconnecting by monitoring the CP states.

Best regards,
lho

varshilgandhi, 2024-04-02:
Hey @lho-stx ,

First of all Sorry to interrupt you holiday season and thanks for giving your time right now,

Disconnect the CP Line from the PLC means like below image you can see:

[image]

I below answer your 2 questions:

1. Yes, when I disconnect the CP line from either the PEV or EVSE board then the communication is stopping as expected.
2. Yes, when disconnecting the CP line in middle of the wire the communication still is continues.

I understand your point that I have to handle the CP Connecting by monitoring the CP States.
But I confuse in below point:
As we know when we disconnect the CP Line from the PEV or EV board then it gives the session stop command to the system and after that charging is stop. But when I disconnect the CP line from the PLC then at that time system did not receive any session stop command right. At that time how system know that CP line is disconnected from the PLC.

Best Regards,
@varshilgandhi

lho-stx, 2024-04-02:
Hey @varshilgandhi,

no problem holiday season is already over, I just told you for you understanding.

You see the session stop command in case 1. because the PLC signal is not seen in the EVSE and EV anymore and it will run into a V2G related timeout.

In case 2. there is still coupling between the two PLC lines, because of the nature of the PLC signal, therefore there will be no V2G timeout and the charging session continues. 

This high crosstalk of PLC signals is also the reason why SLAC exists. The EV is doing a channel estimation and based on the attenuation of the seen signals it is determining to which EVSE it is connected. If the EV is not able to tell EVSEs apart it will start a so called BCB toggle, where it is periodically switching between state B and state C. If an EVSE sees these state changes it will tell the EV and they know that they are connected to each other.

To test this further you may disconnect the GND lines too and move the charging gun further away from the Whitebeet board.

Best regards,
lho

varshilgandhi, 2024-04-03:
Hey @lho-stx,

You're absolutely right. When I disconnected the power line (CP) and GND line from the PLC and moved the charging gun further away from the charging station control panel, it received the session stop command every time (tested 10 times).

Thank you so much for your time and helpful reply!

Best Regards,
@varshilgandhi

lho-stx, 2024-04-03:
Glad I was able to help you!

closed 2024-04-03 by lho-stx