Sevenstax/FreeV2G #291: Ford E-mach no communication starts after V2G2 Start listen command

Ivangel1972, 2024-03-13:
Platform: EVSE
Firmware Version: i.e. V02_01_00
Host Controller Interface: SPI
Host: Own Implementation 

**Describe the bug**
As in the title: Ford E-mach, does not start communication after V2G2 Start listen command. I tried a solution with the addition of the Set SDP Config command, but this also does not bring a positive resulatu.
When the car is plugged in and next to the charging socket, the charging initiation indicator lights flash white, then, after calling the V2G2 Start listen command on the car, these lights turn red and the car does not respond in any way. I tried changing the protocols to DIN alone or ISO alone - no change. I tried setting only one energy trasnfer mode (Extended) but even that does not help.
I can provide the configuration for all the commands BEFORE the V2G2 Start listen command, but I based it on your FreeV2G program and basically we only differ in the other IDs (ISO and DIN EVSE ID)

**To Reproduce**

**Expected behavior**
In the older FW, i.e. 01_01_07 the same car starts charging correctly, it looks like the problem lies in the FW of the WhiteBeet module - please verify this and solve the problem.

**Logs**
Of course, I can post the SPI log between the CPU and the WhiteBeet module, but I think it's pointless - all the other cars we've tested so far, as well as those tested on stations of our production, charge without problems.
Unfortunately I have NO way to download the log from the port mirror - only from the SPI.

Regards
Marcin Matusiak

lho-stx, 2024-03-13:
Hey @Ivangel1972,

creating the log files is not pointless, else I do not have any clue what is going on. I may not see the V2G traffic but the framing messages are a first indication what is wrong.

Please create the UART and SPI logfiles accordingly to [the wiki](https://github.com/Sevenstax/FreeV2G/wiki/guide-creating-log-files#spi). If you do not have have a Logic2 or KingstVIS based logic analyzer I can tell you in which format you can print out the SPI traffic from your CPU, so I am able to extract the framing messages from it.

Best regards,
lho

Ivangel1972, 2024-03-13:
Hej @lho-stx 
This will not be a simple and quick action, I have to organize a customer who has and will be willing to provide such a car for further tests.
I remind you additionally that this is another time when you expect from me SPI log in which you find nothing wrong - but ok. 
UART log also involves the presence of a car - I will attach as soon as I can.
Regards
Marcin

Ivangel1972, 2024-03-13:
Hey @lho-stx 
He is just working to get everything together logistically. I'll try to do it as soon as possible.
Greetings
Marcin

fjdicker, 2024-03-13:
I have exactly the same problem.
When I connect a TESLA Model 3 everything works fine.

lho-stx, 2024-03-13:
@fjdicker,
do you have the possibility to create logs of the whitebeet port mirror? This would really help alot!

fjdicker, 2024-03-13:
Since I use the SPI Interface that is all what I have.

[20240313_MME_SPI_Log.txt](https://github.com/Sevenstax/FreeV2G/files/14590353/20240313_MME_SPI_Log.txt)
[20240313_MME_UART_Log.txt](https://github.com/Sevenstax/FreeV2G/files/14590354/20240313_MME_UART_Log.txt)

fjdicker, 2024-03-13:
I will try to use a different IPv6 address.
Doing this I found this TYPO.
[image]

fjdicker, 2024-03-13:
The command 0x43 (Get IPv6 Configuration) works only with Index = 0, otherwise I get 0xFF Module Internal error.
I think there is only one entry in the IPv6 address, the example should be adopted.

fjdicker, 2024-03-14:
What steps are necessary to change the IPv6 address of the access point?
I use 0x42 (Set IPv6 Configuration) to add a new address, that works.
But I can't delete the original address with this command.
Do I have to stop the interface before changing IPv6 address and restart it again after changes are applied?

fjdicker, 2024-03-18:
Hello @lho-stx 

Today I created new logs.
The only difference is, that I inserted a delay of 500 ms into my main loop.
You can comprehend that if you have a look at the timestamps of the telegramms.
Of course that can't lead to a working communication after the Set Schedules command is sent.
But, as you can see in the log, I get as Session started Message from the Whitebeet Modul.

Are there any timing restrictions in the high level communication?
Do I have to insert a delay after a specific command?

I also wonder why the MME and also the TESLA always request 0 Schedules.
But we can clarify that later.

[20240318_MME_SPI_Log.txt](https://github.com/Sevenstax/FreeV2G/files/14637635/20240318_MME_SPI_Log.txt)
[20240318_MME_UART_Log.txt](https://github.com/Sevenstax/FreeV2G/files/14637637/20240318_MME_UART_Log.txt)

Kind Regards
F.J.

fjdicker, 2024-03-21:
@lho-stx 

Do already have an idea what happens here?

Kind Regards
F.J.

lho-stx, 2024-03-21:
Hey,

I am still working on it, missing the V2G logs makes it a bit more complicated then usual.

I will have a look into the different FW versions, thank you for pointing it out again!

Best regards,
lho

ahuster, 2024-03-21:
Would the hardware "old firmware"/"new firmware" DIP switch on the board allow rolling back in this case?

lho-stx, 2024-03-22:
Hey everyone,

I checked the SPI logs and the session is closed due to "Charge parameter timeout reached". You mentioned Fords where charging with FW version v1.1.7. Are you aware, that in v2.1.0 the host application needs to send "Update DC Charging Parameters" periodically (default 500 ms interval, please have a look into chapter 18.10.4 of the Whitebeet manual). In the SPI logs I do not see that this message is send periodically.

What I find strange, is that it is only happening for Fords...

My questions to you are:
- Did you also updated the FreeV2G version?
- Do other cars charge with this exact same setup?
- Are you 100% sure your are sending "Update DC Charging Parameters" periodically.

Best regards,
lho

Ivangel1972, 2024-03-24:
Hey @Iho-stx
Re 1: I rely on the corresponding FreeV2G but the code in the production controllers is written in C
Re 2: I ALWAYS use the same settings
Re 3: Here may lie the solution, because I attach this loop a little later - right after the car responds. I understand that it should be after receiving a confirmation that the V2G2 Start listen command has been run correctly?
The change will be simple and I will test it as soon as possible.
Greetings

Marcin

lho-stx, 2024-03-25:
Hey,

we checked the logs again. From the UART and SPI logs it seems the host application is not sending "Update DC Charging Parameters" in time. You need to start to send it as soon as you send the schedules to the Whitebeet, don't need to wait for the PEV to send anything. The default time window is 500 ms, so maybe send the message every 300 ms. You can also increase the window size, but this is only intended for debug purposes.

Please see "18.10.21 Set Session Parameter Timeout" in the Whitebeet manual for more details.

Please double check the handling of "Update DC Charging Parameters" in your host application again.

Best regards,
lho

fjdicker, 2024-03-26:
@Ivangel1972 
I have a question.
When you started this issue you had the problem that the MME didn't answer with a "Session started (0x80) after your command "Start listen (0x6A)".
Am I right?

Kind Regards
F.J.

Ivangel1972, 2024-03-30:
@fjdicker 
Sorry for the long delay in responding, very busy - yes, exactly, any activity in communication stops after receiving confirmation of a properly started Start Listen command
Best regards
Marcin

Ivangel1972, 2024-05-20:
Hey everyone,

I have confirmed information from a customer that the Ford E-Mach continues to fail to charge. As I said I have no way to get the logs through the mirror port.
Greetings
Ivangel

fjdicker, 2024-05-21:
Hello together,
That's exactly the problem.
Using the SPI interface we cant record the logs with port mirror.
What else can we do to fix the problem?

On 18.03.2024 I provided logs which solve the problem.
I just added a delay of 500 ms between all the commands before "Start Listen" command.
Then I get a answer after sending the "Start Listen" command.

However, the session ends soon because of other issues.

Why do I need this delay to start the V2G session ???

Kind regards
F.J.

Ivangel1972, 2024-05-21:
Hey  @lho-stx 
Where can I download this experimental FW to see if it works ?

Ivangel

lho-stx, 2024-05-27:
Hey,

the firmware is shared with CODICO, you will get it from their website.

Best regards,
lho

fjdicker, 2024-05-31:
Hello @lho-stx
I can't find the experimental FW at CODICO website!
The latest Version there is V02_01_00.

Kind Regards
F.J.

Ivangel1972, 2024-06-03:
Hey @lho-stx
Unfortunately, our Whitebeet modules with FW version 02_01_00 do not allow other FW to update.
Sincerely
Ivangel

Ivangel1972, 2024-06-03:
Hey @lho-stx 
I am having trouble uploading a new FW to the Whitebeet module. Every time after the 3rd packet (each packet is confirmed) right after confirming its receipt comes the error message 0x82. I am not able to update any module with FW 02_01_00 to another FW, while modules with FW 01_07_00 update without any problem.
Regards
Ivangel

lho-stx, 2024-06-04:
Hey @Ivangel1972,

are you updating to the experimental firmware? Are you using the FWUTool on Windows or on Linux?

Best regards,
lho

Ivangel1972, 2024-06-04:
Hey @lho-stx 
No, we have production controllers where I use SPI for communication, and therefore for FW updates. As I mentioned all Whitebeet modules with FW version 01_01_07 or lower I can upgrade to version 02_01_00 without any problem, including the experimental version. But already with FW 02_01_00 I can't do anything, and the vast majority of our stations have modules with FW 02_01_00
Greetings
Ivangel

lho-stx, 2024-06-04:
Hey,

I apologize I think I need to rephrase my question. FW 02_01_00 is the latest official FW, which one is the desired target FW to updated to? The experimental FW? So you want to do _FW 01_01_07 -> FW 02_01_00_experimental_ and _FW 02_01_00 -> FW 02_01_00_experimental_ and the later one is not workling?

Best regards,
lho

Ivangel1972, 2024-06-04:
Hey @lho-stx 
We have more than 300 chargers built based on the WhiteBeet module, most of them are 2 x CCS, so there is one or mostly two modules in each - and almost all of them have FW 02_01_00
Regards
Ivangel

lho-stx, 2024-06-04:
Understood,

I have an idea what's wrong. Please allow me to analyze the issue, most likely there is an error in the update file, not the WHITE Beet itself.

Best regards,
lho

Ivangel1972, 2024-06-04:
Ok, I understand, that's why I attached the SPI log where you can clearly see what is happening. I'll also hint that when I change the payload from the default in my case of 512Bytes to say 250Bytes, the error comes out after the fifth packet.
Greetings
Ivangel

Ivangel1972, 2024-06-05:
I have information that Mustang is charging
Ivangel

lho-stx, 2024-06-05:
Hey @Ivangel1972,

can you share some more information? Did you use the experimental firmware? I was still looking into the issue with the update file.

I try to understand what is now different then before....

Best regards,
lho

Ivangel1972, 2024-06-05:
Hey @lho-stx 
Yes, I have information that Mustang on experimental FW is charging, but all the time we can't update FW on modules with FW 02_01_00
Regards
Ivangel

Ivangel1972, 2024-06-06:
Hay @lho-stx 
A bit of information: I have a WB module about 2 years ago updated to the first version of FW 02_xx_xx - I connected it today and no FW update can be done on it either. I have the impression that the problem is in the whole FW 02_xx_xx series when trying to update via SPI.
Greetings
Ivangel

lho-stx, 2024-06-07:
Hey @Ivangel1972,

can you share the UART and SPI logs with me?

I tried to reproduce your issue, but for me it is working...

Best regards,
lho

Ivangel1972, 2024-06-09:
Hey @lho-stx 
Here is SPI log
https://github.com/Sevenstax/FreeV2G/issues/291#issuecomment-2146815717
UART - i try
Regards
ivangel

Ivangel1972, 2024-06-12:
Hey @lho-stx 
There is UART log when im trying to update FW:

`

▒TION:ENCE▒P▒%
      ▒r▒▒▒▒  ▒▒ ISO15118▒▒YM▒
                              ▒▒▒▒ͥ▒▒鐂V02▒▒▒a0
                                              ▒▒▒ɝ▒▒鐂▒*▒U▒BEET
                                                                ▒(U▒▒  ▒▒▒*Uj▒2▒                                                                                                                                                                                                                                             45
  ▒
   ▒▒kэ▒▒▒ɽ▒▒▒▒AInterface:▒*▒▒
                              ▒
▒TION RUNNING▒▒▒▒              ▒▒1%
▒ stxSL▒▒▒▒▒ѡ)   [▒▒▒r▒09] [NOTICE]▒**
uAAPL-ETH:▒▒MAC of▒*jma]:ꊕ     c4:93:00:34:ae:f1
uAAPL-ETH:  ▒SP(z▒ANIC[1]:      c4:93:00:34:ae:f0[  ▒▒r▒▒▒]▒+ꊕ
uA▒ŵM▒RJ▒BOOT-CFG:▒▒▒AQC▒▒▒▒firmware▒▒▒▒Ahost    [▒▒ 0.520]▒+ꊕ
                                             [▒▒▒r▒▒▒▒A[NOTICE] APL-MAIN:▒▒▒▒▒▒U                                                                                                                                                                                                                                             ~▒[ ▒▒0.522] [NOTICE] ▒ŵM▒RJ▒BOOT-CFG:▒S▒ձ▒Aconfigured to▒Q▒U▒
                                                              [▒▒▒r▒▒▒▒A[NOTICE]                                                                                                                                                                                                                                             Y▒HL:  Boot▒▒▒▒Ahost▒▒▒▒ѕ▒▒
▒▒▒▒▒▒▒▒)S▒▒▒Vkѕ▒          [▒▒▒r▒▒▒▒A[NOTICE]▒▒:
         [▒▒▒r▒▒▒▒A[NOTICE] ▒ŵETH:▒ԋAlink:
                                          [  ▒▒r▒▒▒▒A[NOTICE] APL-ETH: ▒▒ Interf                                                                                                                                                                                                                                             uAAPL-ETH:  ▒▒MAC:▒▒▒▒ C4:93:00:34:AE:F0
                                        [▒▒▒r�▒▒A[NOTICE] PLCV-HL: Reset▒T▒▒00x
                                                                               [                                                                                                                                                                                                                                             uAAPL-M▒RJ▒ꊕ
           [▒▒▒&▒▒▒▒A[NOTICE]▒PŵMAIN:▒OOOOOOOOOOOOOOOOOOOOOOOOOOOO▒
                                                                   [  ▒▒r▒▒▒▒A[N                                                                                                                                                                                                                                             OTICE]▒PŵMAIN:▒▒]ALAN CONFIGUR▒U▒▒
                                  BJA▒▒):▒
                                          [▒▒▒&▒▒▒▒A[NOTICE] APL-MAIN:▒▒KAIP Con                                                                                                                                                                                                                                             fig▒+▒▒▒▒A[fe01]
uAAPL-MAIN:▒ReِP▒▒▒͕▒▒ 2.464]▒+ꊕ
uAAPL-M▒RJ▒▒b▒▒▒▒Local: ▒V▒҂▒▒:0 :▒,S▒▒2▒▒fe34:aef0
                                                   [▒▒▒&▒▒▒▒A[NOTICE]▒PŵMAIN:  ▒                                                                                                                                                                                                                                             ▒k▒▒▒   0:0:0:0▒'▒:0:0:0
                        [  ▒▒r▒▒▒▒A[NOTICE] APL-MAIN:▒▒DNS1:    2001:4860:4860:0                                                                                                                                                                                                                                             ▒'▒▒▒:0:8888
uA▒ŵMAIN:▒▒DNS2:2.467]▒+fd00:0:0:0 :▒▒▒Ң▒2▒▒fe44:5280
                                                     [▒▒▒&▒▒▒▒A[NOTICE] APL-MAIN                                                                                                                                                                                                                                             :▒OOOOOOOOOOOOOOOOOOOOOOOOOOOO▒
uAAPL-M▒RJ▒                    [ ▒▒2.469]▒+ꊕ
           [ ▒▒3.742] [NOTICE] PLCV-HL:  =======================================                                                                                                                                                                                                                                             ▒
Y▒HL:▒▒▒▒A[NOTICE]▒▒
         ▒▒APIB▒W▒Aloaded▒▒׍▒▒͙ձA
Y▒HL:▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒OTICE]▒▒
Y▒HL:▒▒ BOOT FROM HOST RE▒T▒                 [ ▒▒4.258] [NOTICE]▒▒
▒HL:▒zݹAPLC MAC▒O▒:93:00:34:▒Q'▒▒i▒▒▒A[NOTICE]▒▒
                                 [ ▒▒5.161]▒P▒▒ [BB] FWU▒j▒0 %
                                                              [▒▒▒M)ʚ▒A▒▒▒ [BB]                                                                                                                                                                                                                                              FWU X▒K▒0▒▒
           [  ▒▒r▒▒▒▒A▒▒▒ [BB] FWU X▒K▒0▒▒
                                          [▒▒ 5.228]▒P▒▒▒U▒QQUM▒Q*▒J▒
[... cut ...]          "9M▒:   fd00:0:0:0▒'▒a5:43ff:fe44:5280
uA▒ŵMAIN:▒OOOOOOOOOOOOOOOOOOOOOOOOOOOO▒ ▒▒6.681]▒+ꊕ
uAAPL-MAIN:▒                           [ ▒▒6.682]▒+ꊕ
            [ ▒▒7.721] [NOTICE] PLCV-HL:▒▒      AFile▒W▒Aupdated▒O▒Download▒▒  A                                                                                                                                                                                                                                             without reset
`
Regards
Marcin

Ivangel1972, 2024-06-12:
Hey @lho-stx 
Could the reason be that the time is not set?
Regards
Marcin

Ivangel1972, 2024-06-14:
Hey @lho-stx 
Update will working, the reason was that the time was not set. If I set it before starting the update, the update starts.
Regards
Marcin

lho-stx, 2024-06-14:
Hey @Ivangel1972,

please apologize my late reply, high activity in this issue tracker at the moment.

Indeed, setting the time is required, so the certficates can be validated. Happy you resolved the updating issue.

Best regards,
lho

PS:
Thank you for the UART log. I kindly request to create a .txt file and place the UART logs in there, use a dedicated <pre></pre> tag or the markdown cite option, because github sometimes messes with the format of stuff pasted in the here, because it interprets it as markdown syntax.

Ivangel1972, 2024-06-26:
Hey @lho-stx 
The problem with charging Ford cars seems to have been solved. As of now, I have no confirmed information about charging problems with any other car. It seems to me that the current experimental FW could already become the next official version and be published and you can probably close this thread.
Greetings
Ivangel

Ivangel1972, 2024-08-24:
Hey @lho-stx 
are you planning a final firmware version for WB? Because you could probably close the thread already?
Greetings
Ivangel

jptalledo10, 2024-12-04:
Hi. I am having issues with freeV2G talking with a Ford Mach-E without success.  What firmware version should address the communication properly with the Ford Mach-E potentially?

lho-stx, 2024-12-09:
Hey,

please send an e-Mail to support@whitebeet.sevenstax.de to obtain the experimental firmware.

Best regards,
lho

Ivangel1972, 2025-01-25:
Hey @lho-stx 
I think it's high time to inform me when will be the final version of FW for WB?
Regards
Ivangel

fri-stx, 2025-02-07:
Hey @Ivangel1972,

a official release is planned for next week. It will include the changes of the experimental FW.

BR,
fri