Sevenstax/FreeV2G #293: Question: How can I change the IPv6 address

fjdicker, 2024-03-15:
What steps are necessary to change the IPv6 address of the access point?
I use 0x42 (Set IPv6 Configuration) to add a new address, that works.
But I can't delete the original address with this command.
Do I have to stop the interface before changing IPv6 address and restart it again after changes are applied?

lho-stx, 2024-03-18:
Hey @fjdicker,

normally you do not need to stop the interface. 

I set a custom IPv6 on a PEV (station) successfully with the following commands:

1. delete the existing MAC address by sending modID = 0x05 subID = 0x42 interface = 0x01 command = 0x00 IP = ip of my device
2. double check if the deletion was successful by sending modID = 0x05 subID = 0x43 interface = 0x01 index = 0x00. This leads to response code 0xff (internal error). This is expected, because no IP address is set at the moment.
3. send a new MAC address by sending modID = 0x05 subID = 0x42 interface = 0x01 command = 0x01 IP = ip of my device.
4. double check if adding the IP address was successful by sending modID = 0x05 subID = 0x43 interface = 0x01 index = 0x00.

If you follow these steps, does it work for you?

If not, can you tell me which IP address you want to set?

Best regards,
lho

fjdicker, 2024-03-18:
Hi @lho-stx 
I will check that, thanks for your prompt reply.

Kind Regards
F.J.

closed 2024-04-11 by github-actions[bot]