Sevenstax/FreeV2G #295: Question: ReSync from SPI communication Error

xuanlanghuang, 2024-03-25:
**Your Question**
Ask your questions here.
Is it possible to resync from a spi communication error?
From WB user manual 9.2.3, It mentions " If the WHITE Beet Module SPI communication gets out of sync the WHITE Beet module needs to be reset."

But there is also a 10.2 Recovering from synchronization Error session talks about how to resync from a framing error.

Are these two talking about the same synchronization error? If they are, does sending Ping message on SPI require the Size exchange frame?

xuanlanghuang, 2024-04-02:
Hi Iho,

1. How do we identify a framing sync error?
2. Is it recommend sending and receiving to/from the WB at the same "size-exchange-->data exchange" sequence? We have some rare occasion that we have a non-zero size-exchange from both sides. 9.2.4 mentions we should use the larger values as the size for the next SPI transfer. If we padded the shorter size, will that create a SPI error?

lho-stx, 2024-04-02:
Hey @xuanlanghuang,

I assume you are working on a custom host application?

1. If you e.g. read from the Whitebeet and the framing header is no present
2. It is possible to to a combined read/write, but there is no real benefit in this scenario. I would advise to do read and write separated. Yes you need to use the larger size, else you will get a framing sync error. Padding the data with shorter size should not be a problem

Do you have a specific use case which is failing? Are you able to create SPI log files following [this guide](https://github.com/Sevenstax/FreeV2G/wiki/guide-creating-log-files#spi)?

Best regards,
lho

closed 2024-04-30 by github-actions[bot]