Sevenstax/FreeV2G #296: Question: SPI communication failure due to re-sync

FivaTagin, 2024-03-27:
Hi There, 

Currently we are using STM32G4 as our controller to communicate with Whitebeat 

The system generally operates as intended; however, we frequently encounter synchronization issues where the TX_PENDING ping cannot be reset by the WB site.

Upon examining the data packet with a logic analyzer, it appears that the host site (STM32) has received the full length of the payload. Nonetheless, the WB site fails to reset the TX, preventing the host site from proceeding with subsequent processes.

The digital signal log, recorded by DSView, is attached for reference. 
To pinpoint the issue I've highlighted, please refer to the last data package in the log.
[Double header.zip](https://github.com/Sevenstax/FreeV2G/files/14766370/Double.header.zip)

lho-stx, 2024-03-28:
Hey @FivaTagin,

thank you for creating log files, but unfortunately we do not offer support for DSView. Please create the log files with Saleae Logic2 or KingstVIS if possible, or create a `.csv` dump of the SPI traffic with the following header:

`Time [s],Packet ID,MOSI,MISO`

Where MOSI and MISO are the hex values of the bytes seen on the SPI lines.

Your understanding is valued,
lho

FivaTagin, 2024-04-03:
Hi @lho-stx 
Thanks for your reply, 

The attachment is the csv file if it helps. 
You may find the problematic timestamp from this file.
Feel free to advise any questions if you may have.
[image]

[virtual-session-la-240403-141159.csv](https://github.com/Sevenstax/FreeV2G/files/14846382/virtual-session-la-240403-141159.csv)

Best Regards,

Josh.

lho-stx, 2024-04-09:
Hey @FivaTagin,

unfortunately you logfile does not comply to the asked format. I need the packet ID in order to see which bytes belong to which packet. Is it possible to add this information to the logfile?

Best regards,
lho

FivaTagin, 2024-04-15:
Hi @lho-stx , 

Sorry to hear the file does not work. 

We have just upgraded our communication driver with Whitebeet, we will get back to you if this issue still occur under this update. 

Much appreciated 

Josh.

FivaTagin, 2024-04-30:
Hi @lho-stx 
We have tried to enhance our communication with whitebeet, it still works most of the moments. 
But we found there is a situation that is confusing to be handled. 
As the following picture shows in the middle of our command sent via MOSI. 
The TxPending event was triggered by Whitebeet. 

What does the host system need to handle this situation? 
Because as you can see although we could receive the ack message from MISO. 
But the TX pending has never gone off. 

I would appreciate it if you could help us review the following pictures because you are not using the same digital analyser. 
These are the screen shot if it helps. 

[image]

[image]

[image]

[image]

FivaTagin, 2024-04-30:
and we also found a non-response issue on whitebeet during the charging. 

For example, in Image 1, the timing shows when WB is finishing the charge process. At this time, the host was delivering the end charge event mod: 0x27 cmd: 0x63, but a TX pending event also came in at the same moment. Following up in Image 2, the WB should deliver the confirmation event by 0x27 0x63 instead of delivering a full data package with all 0xAA. Then, the WB triggered another TX pending event, but this time there was no further valid header or subsequent data package.

I wonder if you can advise the concern about this behaviour, I am wondering if it a wrong operating timing to catch the TX pending event. 

image 1. 
[image]

image 2
[image]

lho-stx, 2024-05-08:
Hey @FivaTagin,

I want to focus on the first set of pictures for now. Just for my understanding:
1. you send a size exchange frame, during this frame the tx pending line went high
2. you are sending your data exchange frame. It is hard for me to understand the next flow, because somehow github removed [this image](https://private-user-images.githubusercontent.com/11451881/326667302-7e639799-c0fb-4029-8eb8-c2840b597198.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTUxNTIzNzUsIm5iZiI6MTcxNTE1MjA3NSwicGF0aCI6Ii8xMTQ1MTg4MS8zMjY2NjczMDItN2U2Mzk3OTktYzBmYi00MDI5LThlYjgtYzI4NDBiNTk3MTk4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTA4VDA3MDc1NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBkMjdiYWYzNjRhZjYzM2JjYWJlYjYyNTc2ZmRiYjAwZWVjMjJhZGZkMGY3MjhlY2M4ZTQ1MTkzMWYyOTg2M2YmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.De72EhvJiJmhq_uB5qCrNjOCwYKNFWSIKDJAcRxcSTw), I can only see the preview. But for me it seems you are sending your data.
3. Then you try to read back the pending TX frame shown in the last picture.

For me this looks ok. Did you try to read back more TX frames? Because 

>When the SPI_TX_PENDING pin is high there is at least
one full framing message ready to be reveived by the host.

(taken from Whitebeet manual p. 40).

Now to the last set of pictures:

With the first size exchange frame the Whitebeet tells you that it has 44 pending bytes (0x2c), but you only send out 28 bytes of data (0x1c). Therefore there is more data in the Whitebeet TX buffer. You need to transfer the maximum of the two values, so the Whitebeet can send out its data completely. Just pad your data with 0x00 if the Whitebeet has more data to send then the host.

Please try the suggestions and let me know if this resolves your issues!

Best regards,
lho

closed 2024-05-30 by github-actions[bot]