Sevenstax/FreeV2G #299: Status Message Schedules Requested

fjdicker, 2024-04-01:
Platform: EVSE
Firmware Version: V02_01_00
Host Controller Interface: SPI
Host: Own Implementation

When I get the Schedules Requested Status Message Max entries is always 0.
That's the same wit FORD MUSTANG and Tesla.

WhiteBeet Back: 0x27 | 0x84 | 0xFF | 6d = 0x00, 0x00, 0xD6, 0xD3, 0x00, 0x00, 
V2G Service - 0 Schedules requested within 54995 ms

How should I respond?
The first byte in the answer Set Schedules can be:

Code uint8 Result code:
0: OK
1: Unable to deliver schedules

Is it possible to work without Schedules?

[image]

jpo-stx, 2024-04-04:
There is the possibility defined in the standard where the PEV can request the maximum number of entries the EVSE shall use in it's pamx schedule and sales tariff. However, the standard does not define limits on the max entries parameter, but the pmax schedule and sales tariff shall have at least 1 element each for the message to be valid.

Working without a schedule is not possible. Have you tried setting only one entry in the pmax schedule?

fjdicker, 2024-04-04:
No, I haven't tried that.
In the moment I use the example from the Whitebeet Manual:

uint8_t DefaultSchedule[] =
{

 // Revision No.: 2.17 Page 150 / 247 State: Released
 // White Beet - User Manual

 0x00,                       // (Result code: OK)
 0x01,                       // (1 schedule tuple)
 0x00, 0x01,                 // (Tuple ID of the 1st schedule)
 0x00, 0x02,                 // (2 schedule entries)
 0x00, 0x00, 0x00, 0x00,     // (1st entry starts now)
 0x00, 0x00, 0x0E, 0x10,     // (Duration of 1st entry: 3600s (1h))
 0x01, 0x18, 0x03,           // (Available power: 280kW)
 0x00, 0x00, 0x0E, 0x10,     // (2nd entry starts after 3600s (1h))
 0x00, 0x01, 0x43, 0x70,     // (Duration of 2nd entry: 82800s (23h))
 0x00, 0x32, 0x03,           // (Available power: 50kW)
 0x00,                       // (No energy to be delivered present)
 0x00                        // (No sales tariffs)
};

Do you have a example with only one entry for me?

Kind Regards

jpo-stx, 2024-04-04:
Just leave out the 2nd entry and set the number of entries to 1:
```
0x00, // (Result code: OK)
0x01, // (1 schedule tuple)
0x00, 0x01, // (Tuple ID of the 1st schedule)
0x00, 0x01, // (1 schedule entry)
0x00, 0x00, 0x00, 0x00, // (1st entry starts now)
0x00, 0x00, 0x0E, 0x10, // (Duration of 1st entry: 3600s (1h))
0x01, 0x18, 0x03, // (Available power: 280kW)
0x00, // (No energy to be delivered present)
0x00 // (No sales tariffs)
```

fjdicker, 2024-04-04:
Thank you,
I will test that.

Kind Regards

fjdicker, 2024-04-30:
Test was OK, you can close this issue.

Kind Regards

closed 2024-05-02 by lho-stx