Sevenstax/FreeV2G #300: Error: Did not receive frames after "DC Charge Parameters Changed Received"

varshilgandhi, 2024-04-04:
Platform: EVSE
Firmware Version: i.e. V02_01_00
Host Controller Interface: Ethernet
Host: FreeV2G (EV_v1.0.4_0) 

Platform: EV
Firmware Version: i.e. V02_01_00
Host Controller Interface: WiFi
Host: FreeV2G (EV_v1.0.4_0) 

**Describe the bug**
I start the communication and when at EVSE Terminal when it DC Charge Parameters Changed Received then it did not receive any frames.

**Logs**
Terminal Logs:

EVSE Side: 
[EVSE Terminal Logs.txt](https://github.com/Sevenstax/FreeV2G/files/14865196/EVSE.Terminal.Logs.txt)

EV Side:
[EV Terminal Logs.txt](https://github.com/Sevenstax/FreeV2G/files/14865199/EV.Terminal.Logs.txt)

Wireshark Logs:

EVSE Side:
[EVSE_Error_DC_Charge_Parameter_Received.zip](https://github.com/Sevenstax/FreeV2G/files/14865205/EVSE_Error_DC_Charge_Parameter_Received.zip)

EV Side:
[EV_Error_DC_Charge_Parameter_Received.zip](https://github.com/Sevenstax/FreeV2G/files/14865210/EV_Error_DC_Charge_Parameter_Received.zip)

**Additional context**
I also enable the port mirror (put evse.whitebeet.networkConfigSetPortMirrorState(1) in the line above evse.loop() in Application.py)

varshilgandhi, 2024-04-07:
Hey,
Can you please guide me for this issue?
I read some of yours closed issues on this topic but still I did not clear about it. And also Somewhere I read that this issue is known and it is the bug from your side. Is it True?

Thanks and Regards
@varshilgandhi

lho-stx, 2024-04-09:
Hey @varshilgandhi,

which ticket you are referring to?

I checked the logs and when I compare them with the logs created by a session run by latest FreeV2G  there are differences.

Also your version information is a bit confusing. There is no EV firmware version V02_01_00, from the logs I see you are using EV firmware V01_00_06 and EVSE firmware V02_01_00. 

Did you run the session with the latest FreeV2G stable branch? You are stating to use

> Host: FreeV2G (EV_v1.0.4_0)

This version is outdated, please try the session with the latest stable branch or the tags corresponding to your firmware.

Best regards,
lho

varshilgandhi, 2024-04-09:
Hey @lho-stx,

My apologies for using the previous version.

Let me test with the latest version (Tag: EVSE_v2.1.0_1) and then I'll give you an update on it.

Thanks and Regards,
@varshilgandhi

varshilgandhi, 2024-04-11:
Hey @lho-stx ,
I update the firmware.
I am using this version: https://github.com/Sevenstax/FreeV2G/tree/EVSE_v2.1.0_1
Please check it is the latest version right.

After that when I run again but still I got the same issue like I mention above.

Logs:

Terminal Logs EVSE:
[EVSE Terminal Logs.txt](https://github.com/Sevenstax/FreeV2G/files/14941724/EVSE.Terminal.Logs.txt)

Terminal Logs EV:
[EV Terminal Logs.txt](https://github.com/Sevenstax/FreeV2G/files/14941726/EV.Terminal.Logs.txt)

WireShark Logs EVSE:
[EVSE WireShark Logs.zip](https://github.com/Sevenstax/FreeV2G/files/14941822/EVSE.WireShark.Logs.zip)

WireShark Logs EV:
[EV WireShark Logs.zip](https://github.com/Sevenstax/FreeV2G/files/14941828/EV.WireShark.Logs.zip)

UART Logs EVSE:
[EVSE UART Logs.zip](https://github.com/Sevenstax/FreeV2G/files/14941832/EVSE.UART.Logs.zip)

UART Logs EV:
[EV UART Logs.zip](https://github.com/Sevenstax/FreeV2G/files/14941833/EV.UART.Logs.zip)

Thanks and Regards,
@varshilgandhi

varshilgandhi, 2024-04-12:
Hey @lho-stx ,
I also wants to know that as you told me to the update the firmware.

So Is there I have to follow the steps which is given into the chapter 7 of the white beet user module?

[image]

Thanks and Regards
@varshilgandhi

varshilgandhi, 2024-04-15:
Hey @lho-stx,

Just following up on the issue we discussed earlier. Can you please provide an update on it if there's any?

Thanks and Regards,
@varshilgandhi

lho-stx, 2024-04-15:
Hey @varshilgandhi,

if you want to update the firmware of the whitebeet, you need to follow the steps given in chapter 7.2 of the whitebeet manual.

Regarding the issue:
I checked the logs. In the logs everything seems fine until the pre charging state. The EV notifies pre charging ready, but then there is no communication for 40 seconds and then there is the session stop command. What I think is strange, is that the start pre charging command is send from your host to the Whitebeet directly after the session stop notification. This also triggers the warning in the UART log, because when the session is stopped the Whitebeet can not start pre-charging anymore.

I tried to recreated the session with my reference setup, but in my case everything works as expected. My reference setup is a Debian 12 based PC, with each Whitebeet connected to it's own ethernet USB dongle. I used the latest firmware for PEV (1.0.7) and EVSE (2.1.0) and the latest FreeV2G commit on the master branch.

I see you are using Wi-Fi and Windows. Windows is known to cause issues in the low level ethernet packet handling.

My ideas of things to try are:

1. Is it possible that you try to run FreeV2G on a Linux based PC?
2. Try to connect to the Whitebeet directly via e.g. ethernet USB adapter. Do not place any network gear between Whitebeet and your Host PC. Do not use Wi-Fi.
3. Try the latest EV firmware (also I think the issue is related to your host PC / network setup)
4. Try everything above at once

Thank you and best regards,
lho

varshilgandhi, 2024-04-16:
Hey @lho-stx,

Thanks for your suggestions! I followed your ideas and:

    1. Switched to a Linux system.
    2. While I couldn't connect the Whitebeet directly via Ethernet, I continued using Wi-Fi and Ethernet.
    3. Updated to the latest firmware.

After implementing these changes, everything seems to be working smoothly now.

One question: Since my setup is working fine now, is it still necessary to update the Whitebeet firmware as you mentioned earlier (following the steps in chapter 7.2 of the Whitebeet user manual)?

Thanks and best regards,
@varshilgandhi

lho-stx, 2024-04-16:
Hey @varshilgandhi,

Good to see you made progress!

I recommend you to update, because the latest firmware has improvements and bug fixes implemented.

When using Wi-Fi you introduce more network gear into your setup. It is not necessarily true that this causes issues, but it is known that using routers / switches causes problems, especially in a multi Whitebeet setup. The reason is, that the Whitebeet e.g. sends broadcast messages and your host application needs to differentiate them. FreeV2G does not support such feature (yet). In addition to this, if you send framing messages to the broadcast MAC address, both Whitebeets will respond.

In theory you can use e.g. a switch with multiple Whitebeets, but you need to set up V-LANs and your switch needs to support it. You also need to add virtual interfaces on your host PC for each V-LAN. But setting this up is out of scope of this repository.

Best regards,
lho

closed 2024-04-16 by lho-stx