Sevenstax/FreeV2G #302: Question: Steps to Update the firmware of WhiteBeet Module

varshilgandhi, 2024-04-16:
**Your Question**
I followed the Chapter 7:2 for update the latest firmware of the WhiteBeet Module from the User-Manual of White Beet.

[image]

As mention in the above image that first I have to fulfill the requirements for using the firmware upload tool 'StxFwUpdater'. and for that I go to Chapter 22.1

[image]

I Installed all above libraries in my linux setup.

Now as per there documentation now I have to execute below command:

STxFwUpdater.exe -f mac_address.fwu -t 00:01:01:63:77:33 -i eth0

But now I am stuck that where to find this STxFwUpdater.exe file?
Also they created this mac_address.fwu file. So I want to know that how to get this files?

nststx, 2024-04-16:
Hi @varshilgandhi,

thanks for submitting your question. You can find the firmware update tool in the CODICO download area. If you're using a linux setup you don't want to use the *.exe* file as this is a windows only executable and won't run on linux. If you've already installed the required python packages there's no need to use the executable anyways as it only contains the python program prepackaged with the required packages. 
The *mac_address.fwu* file is only a sample file. It's not what you want to use to update the firmware of your board. 
To update your board please follow these steps:

1. Download the FWU tool from the CODICO download area
2. Download the FWU file for the firmware version that you want to update to
3. Install `libpcap`. For Ubuntu or Debian distributions you can do this by running `sudo apt update && sudo apt install libpcap-dev` in your terminal. Note that `libpcap` is required for the tool to work
4. Install all the required python packages (which reading from your post, you already did). Otherwise please install them by running `pip install -r requirements.txt` in the FWUTool folder in your terminal.
5. Run the tool with the following command:
`python3 FWUTool.py -f <path_to_the_fwu_file> -t <mac_address_of_your_board> -i <interface_name_where_wb_is_connected> eth`

I hope I could help you with your issue. If you have any further questions or are running into trouble while updating your board, please let me know!

Best regards,
Niklas Stoffers

varshilgandhi, 2024-04-16:
Hey Niklas,

Thanks for replying.

I followed your steps and tried to download the FWU tool from the CODICO download area, as you mentioned in the first step.

However, when I click on the file  QCA7005_FW_3.3_lite.zip (which I believe is the FWU tool based on the image), it prompts me to contact a CODICO representative.  Should I send an email to CODICO to obtain the tool?

[image]

Thanks and Regards,
@varshilgandhi

lho-stx, 2024-04-17:
Hey,

yes, please contact Codico.

Thank you,
lho

varshilgandhi, 2024-04-18:
Hey @nststx,

Thanks for the steps to update the WhiteBeet firmware on Linux. However, I encountered some difficulties creating the FWU file while following the user manual for Windows.

Since creating the FWU file seems easier on Windows, I'm currently using that approach.

I follow below steps:

First I created one python file and name of that python file is fwuconfig_macaddress.py
Here I attach that file: 
[fwu_config__example.zip](https://github.com/Sevenstax/FreeV2G/files/15020770/fwu_config__example.zip)

I am using this command to generate the FWU file:
 **  .\StxFwGen.exe -k .\SigningCert.key -c .\fwu_config__example.py -o mac_address.fwu **

I got below logs when I generate the FWU file:

PS C:\Users\varsh\Downloads\Sevenstax_WHITEBEET_tools__2023_05_10 (1)\Sevenstax_WHITEBEET_tools__2023_05_10\Windows\StxFwGen> .\StxFwGen.exe -k .\SigningCert.key -c .\fwu_config__example.py -o mac_address.fwu
SEVENSTAX Firmware Update Generator

- Creating module layer
    - Creating data module
    - Created module: ['INFO', [[16, 'CMP', 'SEVENSTAX GmbH']]] with length 28
    - Creating security module
        - Checking for certificate encoding
        - No need to convert, not a PEM file
    - Created module: ['CERTIFICATION', {'certificate': 'SigningCert.crt.der', 'signature_scheme': 'rsa-pkcs1'}] with length 1083
    - Size of module layer: 1127B

- Creating block layer
    - Adding module data blocks
    - Detected module size of 28
    - Length of module layer: 1127
    - Detected module size of 1083
    - Length of module layer: 1119
    - Creating firmware file header
    - Size of base block layer: 1233B

- Creating hash layer
    - Hashing block chain

- Signing start block fields
    - Key: 0
Found encrypted private key file. Enter passphrase:
    - Block data to be signed: b'aa55043b7733050100000431000500000201000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001032630820322308202c7a0030201020209008254c7c20f8534a8300a06082a8648ce3d040302305a310b30090603550406130244453111300f06035504070c0848616e6e6f76657231173015060355040a0c0e534556454e5354415820476d6248311f301d06035504030c1644656d6f204341205768697465204265657420504b493020170d3231303332343135313030385a180f32313030313233313233353935395a30818d310b30090603550406130244453116301406035504080c0d4e69656465727361636873656e3111300f06035504070c0848616e6e6f76657231173015060355040a0c0e534556454e5354415820476d624831143012060355040b0c0b446576656c6f706d656e743124302206035504030c1b44656d6f205369676e696e67205768697465204265657420504b4930820122300d06092a864886f70d01010105000382010f003082010a0282010100e205ac0550d32fb23c3cb0031687c7588b3ccfc3adae04eb82c8100452808f5e78c3062418d30e73864ed860885fa1995a89c41b6499bd47fea1126d17a5054304a0886e8de4e88d0ad3c90934808ab2580ae2e0fbd1e8d01d01398297df7656c189a110149c3d92812b9bf32971720a322bd6ccb8411a01565e0aab5b99afda35fb08106581adb7b69754c3fffd7cc86178869d1293567a8564fc1482862fcdbce026a0f70e838d94ae3d67ba61b843eabf52af2143dfe26c9d309510a33563434f162abb3f51ac8718d4b175bb1e1ea9228bef3a57b9cc12a3f6e11c1a9366cc5da42ef5017ec82113ff87d8556668f52827c2410805813075f9c35b4b2c350203010001a3753073301d0603551d0e0416041409e17d36373887829be2bd4be21e8b08b3bf0f7a301f0603551d230418301680144ad080d5b6781068121496d9eb70ff7b9d95b5b0300c0603551d1304053003020100300e0603551d0f0101ff04040302078030130603551d25040c300a06082b06010505070303300a06082a8648ce3d040302034900304602210086f3a56fa668ad335accb15b0b414fb6049838db516647058bff742ad5810cb0022100c1027535607b75658824785cc507a68e3760399771a31a4853602e556dfbe2ad0000000000000000000000000000000000000000000000000000000000000000'
    - Length of signed data: 1119
    - Signature size: 256B
    - Signature: b'7b2999f3d6d359b2012bdc7834f6150eb82971005234b5c639b01ecaa4f1c8273606b9fd33df727863526a2c1595d37caeb4c4becd3195dc10bba1979598afc087df80d701f3bb05f325b40cce1a1856f866abb7b21c1c91dc94cff7d41ba8bbbcd8133984e5afc2851c9249601126c9f15e6ab3c5ba7976d4e90d4a89a980023051d8867920481bf3e88ec0a082d9a112383418eb42bad7112ed06f532cd60914a1b49c6c29137ed222f7a616f898b24d3090934c1ec0ff7b2f115b910ca100788f5d0c363bd880b9b3bced2f7bafc1d315e4da610d308c9ed08da4a339b4bf874ee3a81da6667da9c0e9f8cbd71380720c511b0354de6bb1f3806a7135329f'
    - Got hash: b'78debd627f2aaafac2a2df9aa3d49973148690513d78caee8ab059a93e598531'
    - Got hash: b'f3b2b96507bb716d101f5d38687d43706c43ff79153409479e2e92e82a14ac78'

- Size of finished firmware segment (block layer): 1233B

- Size of file (complete block layer of all segments): 1233B
- Writing out file

After this I follow the steps of the Uploading the firmware via StxFwUpdater:

I am using this below command:

STxFwUpdater.exe -f mac_address.fwu -t 00:01:01:63:77:33 -i Ethernet

PS C:\Users\varsh\Downloads\Sevenstax_WHITEBEET_tools__2023_05_10 (1)\Sevenstax_WHITEBEET_tools__2023_05_10\Windows\StxFwUpdater_FWUtool> .\FWUtool.exe -f ..\StxFwGen\mac_address.fwu -t c4:93:00:48:AC:45 -i Ethernet
Welcome to the SEVENSTAX FWU Tool 2.6

Initiating framing interface
Set time on the target to 2024-04-18T06:46:57Z
Starting update process
Firmware update at 0 percent; Sent: 833 Bytes
Done.

Can you please check above steps and please tell me that is it the correct steps or not.

Thanks for your continued support!

Best regards,
@varshilgandhi

varshilgandhi, 2024-04-18:
Hey @lho-stx and @nststx ,

Can you please give an update on it?

Because What happens is When I did the above changes now again when I start the charging then every time SLAC matching is failed.

Thanks,
@varshilgandhi

varshilgandhi, 2024-04-22:
Hey @nststx and @lho-stx ,
Can you please give me update on above issue.

Thanks and regards,
@varshilgandhi

lho-stx, 2024-04-23:
Hey @varshilgandhi,

I think there is some confusion here:

There are two different types of firmware, the Whitebeet firmware and the QCA firmware (uploaded from the microcontroller in the Whitebeet to the QCA7005). To update the Whitebeet firmware you need to follow the instructions [given by @nststx ](https://github.com/Sevenstax/FreeV2G/issues/302#issuecomment-2058523603). What you now did was to overwrite the QCA firmware file in the file system of the Whitebeet with a copy of the Whitebeet firmware.

To recover you need to replace the QCA firmware with the original one and then update the Whitebeet firmware with the correct one from the Codico website, following the guide from @nststx. The fwu name should have "EVSE" and "02_01_00" in it.

When done correctly you can see if the firmware version is indeed 2.1.0 by observing the "Get firmware version" command. It then should tell you the correct firmware version. FreeV2G handles this for you.

I hope this resolves your issue.

Best regards,
lho

varshilgandhi, 2024-04-23:
Hey @lho-stx ,
As you told me to replace the QCA Firmware with the original one but how can I find the original QCA firmware.
Can you please give me steps to do it.

Thanks and regards,
@varshilgandhi

nststx, 2024-04-23:
Hey @varshilgandhi,

As @lho-stx has already told you you replaced the firmware of the QCA chip with our Whitebeet EVSE firmware. Thus the QCA does not work as expected and SLAC matching is failing. To fix this you'll need to replace the wrong QCA firmware with the original one. I'll talk to a colleague on whether and how we can provide a firmware file for you to do this. After you've replaced the QCA firmware please make sure that everything is working again as expected. After that you can update the firmware of the Whitebeet itself by following the steps that I mentioned in a previous post.

Please give me some time to clarify on whether we can provide you the QCA firmware file. I need to talk to a colleague about this as there may be license issues involved. I'll come back to you once I know how we'll proceed.

Best regards,
Niklas Stoffers

varshilgandhi, 2024-04-24:
Hey Niklas,

Thanks for replying and clarifying the issue. I apologize for the mistake of replacing the QCA chip firmware with the Whitebeet EVSE firmware.

I'm definitely waiting for your update on the availability of the original QCA firmware file.

I also have a question: Once I receive the original QCA firmware, on which board should I replace it? Is it the EVSE board or the EV board, or do I need to replace it on both boards?

Thanks,
Varshil Gandhi

varshilgandhi, 2024-04-25:
Hey @nststx,

I want to give you one more update that when I down FW VER switch in the both the EVSE and EV board like below,
[image]

And then I start the charging again and it works completely fine SLAC matching also done and charging completed 100%.

Can you please explain me how this can happens?

Best regards,
Varshil

varshilgandhi, 2024-04-29:
Hey @nststx ,

Is there update on above things?

Best regards,
@varshilgandhi

lho-stx, 2024-06-27:
Hey @varshilgandhi,

I send you an e-Mail.

Best regards,
lho

closed 2024-06-27 by lho-stx