Sevenstax/FreeV2G #310: Question: Wireshark doesn't show bridged EV messages

Zaghz, 2024-04-24:
I am using WSL on Windows 11 with an External virtual switch on my eth interface.

Wireshark normally shows me messages sent by the EV board (not whitebeet). However, when I run the Application script the messages stop showing up from the EV and only messages that are exchanged between the WHITEBEET acting as EVSE and the Host are shown.

[image]

jpo-stx, 2024-04-26:
Do you use the port mirror function of the Whitebeet?

nststx, 2024-05-02:
Hi @Zaghz,

Thanks for posting your question. Unfortunately you didn't use the provided template for creating your issue. To keep up with our issue tracker we only accept issues following our templates. Please refer to [this guide](https://github.com/Sevenstax/FreeV2G/issues/304) on how to post your issue or question.

If your issue still persists please open a new one using our provided template.

Best regards and thanks for  your understanding and patience,
Niklas Stoffers

closed 2024-05-02 by nststx