Sevenstax/FreeV2G #311: Question:

vsCodeGB, 2024-04-30:
Platform: EVSE
Firmware Version: V02_01_00
Host Controller Interface: SPI
Host: Own Implementation

Describe the bug:
I get Session stopped event right after sending Start Charging command.
When I send command "C0 27 AC 00 00 00" I get stop event.
It is tested with DIN+ISO modes, DIN mode only. Same situation with QCA7005 FW v3.0 and v1.1.0-02.
I use SPI for communication and charging to Ethernet will be complicated, maybe I send wrong command?
Python demo works perfect with this EV, but ends at pre charging, also EV charging from another station.
EV support DIN only.

Logs
Saleae Logic2 LOG files of SPI and UART debug:
https://mega.nz/file/BSN1mTKC#JwWAeRqNxsLE9M9ln9GA27-2EL3xrhbsVuHnA4UFrYY

> [   46.131] [NOTICE] V2GDIN12: EV requested to start charging.
> [   46.132] [NOTICE] V2GAPP: Received event 'Start charging requested'
> [   46.172] [NOTICE] V2GSVC: TCP: Client disconnected or connection closed by server
> [   46.173] [NOTICE] V2GAPP: Received event 'Session stopped'
> [   46.188] [NOTICE] V2GSVC: TCP: Listen on port 62672 for non-LS
> [   46.189] [NOTICE] V2GSVC: : Listen on port 62673 for TLS

UART debug from another session:
[WHITE-BEET-EI_log_host_via_spi_DC_Extended_only.txt](https://github.com/Sevenstax/FreeV2G/files/15164792/WHITE-BEET-EI_log_host_via_spi_DC_Extended_only.txt)

jpo-stx, 2024-04-30:
From the debug log I cannot see anything strange happening that would lead to the issue described by you.

Am I reading correctly that when you use the FreeV2G python application it also stops at pre charging?

To help you with the issue I kindly ask to provide a wireshark trace. How to do so is explained in [this guide](https://github.com/Sevenstax/FreeV2G/wiki/guide-creating-log-files).

If you activate port mirror this will also work when your host controller is connected via SPI. The traffic on the PLC will be mirrored to the ethernet interface when port mirror is activated.

nststx, 2024-05-02:
Hi @vsCodeGB,

Thanks for posting your question. Unfortunately you didn't use the provided template for creating your issue. To keep up with our issue tracker we only accept issues following our templates. Please refer to [this guide](https://github.com/Sevenstax/FreeV2G/issues/304) on how to post your issue or question.

If your issue still persists please open a new one using our provided template.

Best regards and thanks for  your understanding and patience,
Niklas Stoffers

closed 2024-05-02 by nststx