Sevenstax/FreeV2G #313: Discordance between EV and EVSE target current parameter

Hipowerguy, 2024-05-07:
I noticed a discordance between EV and EVSE target current parameter.

I use two WhiteBeet demonstration boards, one EV and the other one EVSE. They were bought this spring with the latest firmware as of April 2024.  Each is operated by its own Raspberry Pi 4 & 5 running separate FreeV2G.  They use isolated and direct Ethernet connection.   This setup has worked well for many weeks.

The values displayed on the two monitors during the charge by FreeV2G are 50A in the EVSE board monitor and 60A in the EV board monitor.

These should be the same and be continually exchanged on the PLC communication.

Note that the ev.json file is giving 40A as the configuration value. Why is this one also different from the other two?

Modifying the ev.json does not affect any of the charging parameters.  Why isn't it read when the FreeV2G app is launched in EV mode?

Here is the EV log for the full charging sequence from the time the 

Thanks.

[EV_Control_log.txt](https://github.com/Sevenstax/FreeV2G/files/15237044/EV_Control_log.txt)

lho-stx, 2024-05-08:
Hey @Hipowerguy,

good to hear that your setup is finally working! 

Unfortunately you did not use the issue template and the log files the template asks for are not present, a single UART log file of the EV is not sufficient to inspect this issue. Please have a look into the issue template again and provide all log files asked for.

The issue will be reopened when the information the template asks for is provided.

Best regards,
lho

closed 2024-05-08 by lho-stx

Hipowerguy, 2024-05-08:
The EV log file should be enough as it is an EV Whitebeet issue.

This application is too incomplete to be useful.

I will use a competitors' product from now on even if slightly more expensive for my line of chargers.  

I wasted toomuch time on this Whitebeete product.