Sevenstax/FreeV2G #314: Reason for Sequence Error

fjdicker, 2024-05-10:
Platform: EVSE
Firmware Version: i.e. V02_01_00
Host Controller Interface: SPI
Host: Own Implementation

[image]
[Sequence_Error.txt](https://github.com/Sevenstax/FreeV2G/files/15271193/Sequence_Error.txt)

What can be the reason for the "Sequence Error" I get when trying to charge a CUPRA?
V2G starts without problems.
But after setting the schedules I don't get the request for "Cable Check" like with TESLA.

Are there several reasons for a "Sequence Error" or is this error typical for a special condition?
Is a detailed description for the error codes available?

Kind Regards
F.J.

fjdicker, 2024-05-16:
Please answer my question.

Kind Regards
F.J.

lho-stx, 2024-05-17:
Hey @fjdicker,

there are multiple reasons why a sequence error may occur. From the log file I can not identify the problem. 

Do you have access to the ISO15118-2?

Best regards,
lho

kwoyhope, 2024-05-21:
Hello, I'm a SECC SW developer(not SevenStax), and I'd like to share some comments what I know.
I can see "CP Service - State changed to C" at line#403, and it seems weird to me during "Update DC Charging Parameters".
As I know(in general), CP State "B(9v) -> C(6v)" is occurred at starting CableCheck(ie, after ChargeParameterDiscovery Done).
Is it normal(OK) sequence in this log?

fjdicker, 2024-05-21:
Hi kwoyhope,
Thank you for your support.
I compared the shorted version of the log with a log from TESLA.
The TESLA does exactly what you expect.
From CUPRA I get a "DC Charge Parameter Changed" even before the Cable Check is requested.
Maybe the CUPRA doesn't insist on the Cable Check.

[TESLA Sequence.log](https://github.com/Sevenstax/FreeV2G/files/15386893/TESLA.Sequence.log)
[CUPRA Sequence_Error.txt](https://github.com/Sevenstax/FreeV2G/files/15386906/CUPRA.Sequence_Error.txt)

lho-stx, 2024-07-01:
Hey @fjdicker,

please apologize the long delay. You've used the "question" template for creating this issue and from your initial post I did not understand you have a specific issue.

I already checked the log files you send to @kwoyhope and it seems the CUPRA does indeed skip the cable check. To be sure it is really skipping the cable check, can you please create a wireshark log of a failing charging session with the CUPRA?

If the CUPRA is skipping the cable check, I need to discuss internally what we can do, because the cable check is mandatory.

Thank you,
lho

closed 2024-07-23 by github-actions[bot]