Sevenstax/FreeV2G #315: Question: Plans to add VAS - internet access

Dalachowsky, 2024-05-20:
Hello,
I have a question regarding one of the possible ISO15118 Value added services. Are you planning to integrate VAS - internet acces into your stack? We want our EVSE to be able to give internet access to EV for the purpose of VDV261 (querying the backend server for departure time).

lho-stx, 2024-05-24:
Hey @Dalachowsky,

this is not a trivial feature request. We might be able to modify the port mirror to achieve this. What do you need specifically? A Layer 2 bridging? Do you need separate IPv6 addresses? Also what is your setup? Are you using ethernet as host interface?

I will discuss this internally!

Best regards,
lho

Dalachowsky, 2024-05-28:
Right now we do not have a specific setup. Our device is still in the development phase so we can adjust it to your solution.
What would you propose? We want to be able to give internet access to EV. We are not aware of how it is done by other manufacturers. If you haven't considered it before we would like to hear about your propositions.

lho-stx, 2024-05-31:
Hey @Dalachowsky,

please allow some more time. I would like to test if the workaround is fully functional before I propose it to you, in order to make sure it is indeed working.

Best regards,
lho