Sevenstax/FreeV2G #317: Question: how can I (EV) know that Cable Chack had failed?

Kristanpivk123, 2024-05-25:
**Before you post**
- [x] I confirm that I've read the [guide on how to post issues](https://github.com/Sevenstax/FreeV2G/issues/304)
- [x] I confirm that I've consulted the [wiki](https://github.com/Sevenstax/FreeV2G/wiki) about my issue

**Your Question**
I am using WHITE Beet PI (EV). In case that the EVSE would determine that Cable check had failed, how would I recognize that? Is there any status message that I should receive in that case?
Best regards,
Kristan

lho-stx, 2024-05-31:
Hey @Kristanpivk123,

you might use the Isolation Status filed in the DC Charge Parameters Changed message for this. Please have a look into Whitebeet manual chapter 18.11.2 for more information on this message.

Please let me know if this soles your question.

Best regards,
lho

Kristanpivk123, 2024-06-03:
Hey Iho, 
thank you, I thought that Cable Check is supposed to be some other measurement apart from IMD, but after you pointed it out, I went and checked the IEC 61851-23 again and I see now that it's just insulation check, nothing more.
Thank you again and yes, you answered my question.
Best regards,
Kristan